chipweinberger/flutter_blue_plus #331: A problem occurred configuring project ':flutter_blue_plus'.

cadenxh, 2023-06-08:
Trying to run basic "scan for devices" code from the readme -- but there is an exception Exception: Gradle task assembleDebug failed with exit code 1 while building.

Here is the debug info I have:

<details>
Launching lib\main.dart on SM S911U1 in debug mode...
main.dart:1

FAILURE: Build failed with an exception.

* What went wrong:
A problem occurred configuring project ':flutter_blue_plus'.
> Could not get unknown property 'source' for generate-proto-generateDebugProto of type org.gradle.api.internal.file.DefaultSourceDirectorySet.

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 4s
Exception: Gradle task assembleDebug failed with exit code 1
Exited (sigterm) 
<details>

I am completely new to app development, so I am struggling to isolate the issue here. Obviously, something in either my grade config or flutter blue plus' grade config is wrong, but I have no idea what it might be. From some googling around, I think the issue may have something to do with the protobuf dependency.

brain9520, 2023-06-10:
看到有中文，我直接回复中文了。
你用Android studio直接打开flutter项目中Android的项目，然后Android studio会自动下载依赖，然后再次运行试试看，如果有报错继续发出来。

MrCsabaToth, 2023-06-13:
I wonder if this is related to #306 in any way

MrCsabaToth, 2023-06-13:
According to this https://stackoverflow.com/questions/76175068/proto-datastore-jetpack-compose-gradle-dependencies-exception version bump to protobuf 0.9.+ may help? Currently the `build.gradle` uses `com.google.protobuf:protobuf-gradle-plugin:0.8.17`

cadenxh, 2023-06-15:
> According to this https://stackoverflow.com/questions/76175068/proto-datastore-jetpack-compose-gradle-dependencies-exception version bump to protobuf 0.9.+ may help? Currently the `build.gradle` uses `com.google.protobuf:protobuf-gradle-plugin:0.8.17`

Yep, this was the issue. 

For anyone else running into this issue, you need to change what version of protobuf that flutter_blue uses. Go to wherever you have the flutter_blue plugin installed (for me, it looks like C:\Users\USER\AppData\Local\Pub\Cache\hosted\pub.dev\flutter_blue_plus-1.5.2\android) and open their build.gradle file. Under dependencies, find the line that marks protobuf as a dependency ( classpath 'com.google.protobuf:protobuf-gradle-plugin:0.8.17') and change it from version 0.8.17 (or whatever version you have) to a version higher than 0.9 (I used 0.9.1). 

@MrCsabaToth thank you for your help, I really appreciate it.

closed 2023-06-15 by cadenxh

MrCsabaToth, 2023-06-15:
> > According to this https://stackoverflow.com/questions/76175068/proto-datastore-jetpack-compose-gradle-dependencies-exception version bump to protobuf 0.9.+ may help? Currently the `build.gradle` uses `com.google.protobuf:protobuf-gradle-plugin:0.8.17`

Have you upgraded the AGP to 8.0+? The plugin still doesn't compile for me.

elliotrtd, 2023-06-26:
@cadenxh why did you close this? This should be left open as it should be an easy fix to merge since the package is still being updated.

cadenxh, 2023-06-26:
> @cadenxh why did you close this? This should be left open as it should be an easy fix to merge since the package is still being updated.

Oh, perhaps I misunderstood github etiquette... 

I thought you were supposed to close an issue once a solution had been found, regardless if the issue was on my end or the module source's end. I can reopen it.

MrCsabaToth, 2023-07-04:
@cadenxh After protobuf version bump did you regenerate the protobuf files?

hanskokx, 2023-07-07:
> Oh, perhaps I misunderstood github etiquette...
>
> I thought you were supposed to close an issue once a solution had been found, regardless if the issue was on my end or the module source's end. I can [...]

Yes, this needs to be re-opened. The issue has a workaround solution, but that solution hasn't made its way into the package itself. This solution is fine if you're building locally, but I can't expect to make these changes on my CI/CD or on every other developer's machine. This issue needs to be addressed by the package maintainer and a new version needs to be released.

hanskokx, 2023-07-07:
For anyone else facing this issue, feel free to point to my forked repo with [the fix](https://bitbucket.org/hanskokx/flutter_blue_plus/commits/35ad2cf688c9b5d11f5cdffa687d744e12c62b1c) in it.

```yaml
  flutter_blue_plus:
    git:
      url: https://bitbucket.org/hanskokx/flutter_blue_plus
      ref: protobuf_fix
```

monamoradi82, 2024-12-05:
hi guys, I have a question need your helps. I have a flutter app for Android devices, I want the app get the data from oximeter, blood pressure, or smart watch send them to firebase DB. I got this error: FAILURE: Build failed with an exception.

* What went wrong:
A problem occurred configuring project ':flutter_blue'.
> Could not get unknown property 'source' for generate-proto-generateDebugProto of type org.gradle.api.internal.file.DefaultSourceDirectorySet.  

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.
> Get more help at https://help.gradle.org/.

MrCsabaToth, 2024-12-06:
> hi guys, I have a question need your helps. I have a flutter app for Android devices, I want the app get the data from oximeter, blood pressure, or smart watch send them to firebase DB. I got this [...]

Flutter Blue and flutter blue plus are two different plugins. Flutter blue is discontinued for a long time now. Flutter blue plus on the other hand doesn't even use protobuf any more for a good while now as well. Show your yaml.