chipweinberger/flutter_blue_plus #332: PlatformException(no_permissions, flutter_blue plugin requires android.permission.ACCESS_FINE_LOCATION for scanning, null, null)

GorIvanov, 2023-06-09:
Android 13.
How to fix this?
Everything was working fine until at some point this started.

```
E/flutter (  479): [ERROR:flutter/lib/ui/ui_dart_state.cc(198)] Unhandled Exception: PlatformException(no_permissions, flutter_blue plugin requires android.permission.ACCESS_FINE_LOCATION for scanning, null, null)
E/flutter (  479): #0      StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:607:7)
E/flutter (  479): #1      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:167:18)
E/flutter (  479): <asynchronous suspension>
E/flutter (  479): #2      FlutterBluePlus.scan (package:flutter_blue_plus/src/flutter_blue_plus.dart:184:13)
E/flutter (  479): <asynchronous suspension>
E/flutter (  479): 
```

Manifest:

```
<uses-permission android:name="android.permission.BLUETOOTH" />
<uses-permission android:name="android.permission.BLUETOOTH_ADMIN" />

<uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />
<uses-permission android:name="android.permission.ACCESS_BACKGROUND_LOCATION" />
<uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" />

<uses-permission android:name="android.permission.INTERNET" />
<uses-permission android:name="android.permission.WRITE_EXTERNAL_STORAGE" />
<uses-permission android:name="android.permission.WRITE_INTERNAL_STORAGE" />
<uses-permission android:name="android.permission.FOREGROUND_SERVICE" />
<uses-permission android:name="android.permission.CAMERA" />
<uses-permission android:name="android.permission.RECORD_AUDIO" />
<uses-permission android:name="android.permission.MODIFY_AUDIO_SETTINGS" />
<uses-permission android:name="android.permission.VIDEO_CAPTURE" />
<uses-permission android:name="android.permission.AUDIO_CAPTURE" />
```

brain9520, 2023-06-10:
You just declare what permissions you need in the manifest file, but there is no dynamic application. It is recommended to use the permission_handler class library, and then apply for permissions before using the scan Bluetooth code, for example:
`var permissions = [
        Permission.location, // ACCESS_FINE_LOCATION及ACCESS_COARSE_LOCATION
        Permission.bluetooth,
        Permission.storage,
        Permission.camera,
        Permission.bluetoothConnect,
        Permission.bluetoothScan,
        Permission.bluetoothAdvertise
      ];
      await permissions.request();
      `
Then there is that the Bluetooth permission you applied for is not enough, you need to add 
`
<uses-permission android:name="android.permission.BLUETOOTH_ADVERTISE" />
    <uses-permission android:name="android.permission.BLUETOOTH_CONNECT" />
    <uses-permission android:name="android.permission.BLUETOOTH_SCAN" />
`

WangDan1989, 2023-07-12:
> Android 13. How to fix this? Everything was working fine until at some point this started.
>
> ```
> E/flutter (  479): [ERROR:flutter/lib/ui/ui_dart_state.cc(198)] Unhandled Exception: [...]

How to fix this?
I have the same problem

WangDan1989, 2023-07-12:
> You just declare what permissions you need in the manifest file, but there is no dynamic application. It is recommended to use the permission_handler class library, and then apply for permissions before using the scan Bluetooth code, for example:
> `var permissions = [ Permission.location, // ACCESS_FINE_LOCATION及ACCESS_COARSE_LOCATION Permission.bluetooth, Permission.bluetoothConnect, Permission.bluetoothScan, Permission.bluetoothAdvertise ]; await permissions.request(); `
> Then there is that the Bluetooth permission you applied for is not enough, you need to add
> `<uses-permission android:name="android.permission.BLUETOOTH_ADVERTISE" /> <uses-permission android:name="android.permission.BLUETOOTH_CONNECT" /> <uses-permission android:name="android.permission.BLUETOOTH_SCAN" />`

I run with flutter_blue_plus_example demo

brain9520, 2023-07-12:
> > 你只需在清单文件中声明你需要什么权限，但没有动态申请。建议使用permission_handler类库，然后在使用扫描蓝牙码之前先申请权限，例如：
> > `var permissions = [ Permission.location, // ACCESS_FINE_LOCATION及ACCESS_COARSE_LOCATION Permission.bluetooth, Permission.bluetoothConnect, Permission.bluetoothScan, Permission.bluetoothAdvertise ]; await permissions.request(); `
> > 然后还有就是你申请的蓝牙权限不够，需要添加
> > `<uses-permission android:name="android.permission.BLUETOOTH_ADVERTISE" /> <uses-permission android:name="android.permission.BLUETOOTH_CONNECT" /> <uses-permission android:name="android.permission.BLUETOOTH_SCAN" />`
> 
> 我用 flutter_blue_plus_example 演示运行

是否动态申请了权限？贴一下你的申请权限那里的代码

a371057600, 2023-07-17:
I had the same problem and couldn't turn on **location permissions** for the time being.

chipweinberger, 2023-07-18:
this should now be fixed in 1.7.4

closed 2023-07-20 by chipweinberger