chipweinberger/flutter_blue_plus #333: The app from Example couldn't run on Android

0xThomas3000, 2023-06-10:
I couldn't make the app run properly on Android although it can run successfully on IOS. Already turned on the Bluetooth, allowed the app to access the location. But when I clicked on the 'TURN ON' button, there seem to be nothing going to happen, the screen just stays like below.
[image]

Also have added these lines into the 'AndroidManifest.xml' to grant some permissions to the app for Android but no success.
[image]

0xThomas3000, 2023-06-11:
I fixed that problem and it now can run on Android.

closed 2023-06-11 by 0xThomas3000

kalismeras61, 2023-06-28:
@0xThomas3000 what was the issue ?