chipweinberger/flutter_blue_plus #334: BLE Scan Continuously

Kavinkumarsimple, 2023-06-11:
How do i continuously scan for BLE Devices? Im building a BLE Scanner app and it should always scan and show the changes

GrayLand119, 2023-06-27:
> How do i continuously scan for BLE Devices? Im building a BLE Scanner app and it should always scan and show the changes

Try this branch, I created a commit to support scan continuously.
Set params updateInterval, expiredInterval, and allowDuplicates=true in scan method.

https://github.com/GrayLand119/flutter_blue_plus/commit/5b5afa145cf7a19103e44e8f8f62d0a77fb346ae

R-Emagina, 2023-07-18:
Hi,

I'd love to see this in the master branch.
Does someone know when it could be merged?

Thanks!

GrayLand119, 2023-07-19:
PR Created: https://github.com/boskokg/flutter_blue_plus/pull/382

chipweinberger, 2023-07-20:
thanks for the PR, but AFAICT, it should be easy for the user to implement this feature themself. 

Just create a timer and repeatedly call `FlutterBluePlus.instance.scan` yourself, etc.

This repo already has a ton of bugs. We can't handle more complexity.

Thanks again.

closed 2023-07-20 by chipweinberger