chipweinberger/flutter_blue_plus #337: 1.5.2 Device not connecting.

kalismeras61, 2023-06-12:
While version 1.4.0 works fine, the latest version does not connect to the bluetooth device, there is no problem or warning. 

What could be the reason for this, is there anything specific we need to know?

chipweinberger, 2023-06-13:
1.5.2 works great for me on iOS.

You will need to debug the issue yourself.

Is the problem in Flutter Blue dart code? 

Is the problem in the Obj-C code?

You will need to add logging statements to determine this.

kalismeras61, 2023-06-14:
> 1.5.2 works great for me on iOS.
> 
> Is the problem in Flutter Blue dart code?
> 
> Is the problem in the Obj-C code?
> 
> You will need to add logging statements to determine this.

This issue not related only IOS even Android can't connect ble devices. Same code works fine with 1.4.0.. 

something changed on your side which we are not aware of that.

jagiro, 2023-06-14:
Same behaviour for me, upgraded the library and now I can't find any BLE device. Tried with iOS and Android

The unique change on my side to adapt it to the new library version was
`await FlutterBluePlus.instance.isScanning.first` to `FlutterBluePlus.instance.isScanningNow`

kalismeras61, 2023-06-14:
> Same behaviour for me, upgraded the library and now I can't find any BLE device. Tried with iOS and Android
> 
> The unique change on my side to adapt it to the new library version was `await FlutterBluePlus.instance.isScanning.first` to `FlutterBluePlus.instance.isScanningNow`

This is my mine connect method.  Do you think it is because of FlutterBluePlus.instance.scanResults.first ?? @boskokg fyi.

Future<bool> connect([Device? device]) async {
    try {
      await flutterBlue.stopScan();
      final devices = await FlutterBluePlus.instance.scanResults.first;
      final connectedDevice = devices.firstWhereOrNull(
          (element) => element.device.id.id == device?.uuid.toString());
      if (connectedDevice == null) {
        return false;
      }

      try {
        await connectedDevice.device
            .connect(timeout: const Duration(seconds: 20));
      } on PlatformException catch (e) {
        if (e.code != "already_connected") {
          return false;
        }
      }
      
    } catch (e, trace) {
      log.warning("DevicesBloc : Error connecting to device", e, trace);
      return false;
    }
  }

saguado69, 2023-06-15:
Same issue for me, upgrade to 1.5.2, was on 1.4.0 and same code does not detect devices and not connecting, for iOS and Android.

I run the sample code on same devices, and it scans and connect to my device. So, really confusing me.

I'm really trying very basic code:

`// SCAN AND ADD BLUETOOTH PORTS
    if (Platform.isAndroid || Platform.isIOS) {
      FlutterBluePlus ble = FlutterBluePlus.instance;
      // await ble.isOn.then((value) async {
      //   await ble.isAvailable.then((value) async {
      await ble.startScan(
          withServices: [Guid('0000ffe0-0000-1000-8000-00805f9b34fb')],
          timeout: const Duration(seconds: 4));
      var subscripcion = ble.scanResults.listen((detectedPorts) async {
        // ADD BLE PORTS
        for (var port in detectedPorts) {
          String? name = port.device.name;
          if (name != '') {
            // CHECK IF PORT IS ALREADY IN THE LIST
            bool? isAbsent =
                ports.firstWhereOrNull((p) => p['key'] == 'BLE ($name)') ==
                    null;
            if (!isAbsent) return;
            // ADD PORT TO LIST
            ports.add(
                {'key': 'BLE ($name)', 'value': name, 'type': 'bluetooth'});
          }
        }
      });
      // await Future.delayed(const Duration(seconds: 5));
      await ble.stopScan();
      subscripcion.cancel();`

chipweinberger, 2023-06-15:
@saguado69 please open a PR fixing the issue. thanks.

saguado69, 2023-06-16:
Found the issue for iOS, I just updated my Podfile with platform :ios, '15.0' (was on 12.0) and the code worked fine, so, it is scanning and connecting correctly now Same for MacOS. Let's see now Android .... Is there any change on minSdk for Android with version 1.5.2?

saguado69, 2023-06-16:
> @saguado69 please open a PR fixing the issue, or instead use v1.4.0. thanks.

Sorry @chipweinberger I don't know what do you mean with open a PR ... can you be more explicit? thanks!

chipweinberger, 2023-07-20:
please try 1.7.5+. reopen if you have issues.

kalismeras61, 2023-07-20:
@chipweinberger this seems to be work but, when i try to scan devices i am getting below error from the package. 
It is coming from utils.dart line 98

[log] Bad state: Cannot add event after closing. 
[log] #0      _StreamController.add (dart:async/stream_controller.dart:597:24)
#1      new _BufferStream.listen.<anonymous closure>

chipweinberger, 2023-07-20:
should be fixed on master: 49e99108aa6254109f976622abddf24b9be4aa98

kalismeras61, 2023-07-20:
this works fine, are you going to release new version? or should i use to continue ref ?

chipweinberger, 2023-07-20:
ill push a new version right now.

chipweinberger, 2023-07-20:
released in 1.7.7

closed 2023-07-20 by chipweinberger

kalismeras61, 2023-07-20:
@chipweinberger same code doesn't work on IOS :( it is not connecting to device....

chipweinberger, 2023-07-20:
what is happening?