chipweinberger/flutter_blue_plus #339: Remove protobuf on android

chipweinberger, 2023-06-14:
> What needs to be done for the protobuf removal in Android? If it's not complicated I might give it a try. Currently the lack of active AGP 8.0+ support, lagging Android platform dependency versions causing a great grief (effectively the Android platofm does not compile for me right now) (also see #331) I wonder if yanking the protobuf would decrease bumber of dependencies which can cause grief.

@MrCsabaToth

chipweinberger, 2023-06-14:
@MrCsabaToth that would be great.

See the latest commit on master. It gives an example.

It's pretty straight forward.

In all the places we use protobuf, we should instead use raw java maps.

The schema is already decided. See bluetooth_msgs.dart. 

see these docs: https://docs.flutter.dev/platform-integration/platform-channels

let me know if you have questions.

chipweinberger, 2023-06-14:
protobuf is an annoying dependency, which is why we are removing it.

MrCsabaToth, 2023-06-14:
Well, I poked into it and there are some questions. For example `channel.invokeMethod` (which is a Flutter API between the Android land and Dart land) was using a serialized byte array with the protobuf. So is it possible to just pass a Java Map to it? Or the map should be serialized into a jSON string and then UTF-8 encoded into a byte array and on the other side it'll be deserialized? I see "JSON" appear in the iOS code, so maybe that's why? (Because otherwise it's also a map in Objective-C).

MrCsabaToth, 2023-06-14:
Isn't there a new plugin API replacing the channel invokeMethod?

chipweinberger, 2023-06-14:
sorry json is a bad name. I should call it "map".

for your question, see the flutter documentation 

there is automatic conversion from dart maps to java maps

chipweinberger, 2023-07-18:
this work is now done in 1.7.0

closed 2023-07-18 by chipweinberger

MrCsabaToth, 2023-07-18:
Thanks for your work