chipweinberger/flutter_blue_plus #341: when the device is connected,turn off blue ,android app exit unexpected

Sindyue, 2023-06-15:
D/VRI[MainActivity](  889): onFocusEvent false
D/AndroidRuntime(  889): Shutting down VM
E/AndroidRuntime(  889): FATAL EXCEPTION: main
E/AndroidRuntime(  889): Process: com.rl.xg_admin_app, PID: 889
E/AndroidRuntime(  889): java.lang.RuntimeException: Error receiving broadcast Intent { act=android.bluetooth.adapter.action.STATE_CHANGED flg=0x15000010 (has extras) } in com.boskokg.flutter_blue_plus.FlutterBluePlusPlugin$1@d6d2a24
E/AndroidRuntime(  889): 	at android.app.LoadedApk$ReceiverDispatcher$Args.lambda$getRunnable$0$android-app-LoadedApk$ReceiverDispatcher$Args(LoadedApk.java:1942)
E/AndroidRuntime(  889): 	at android.app.LoadedApk$ReceiverDispatcher$Args$$ExternalSyntheticLambda0.run(Unknown Source:2)
E/AndroidRuntime(  889): 	at android.os.Handler.handleCallback(Handler.java:942)
E/AndroidRuntime(  889): 	at android.os.Handler.dispatchMessage(Handler.java:99)
E/AndroidRuntime(  889): 	at android.os.Looper.loopOnce(Looper.java:240)
E/AndroidRuntime(  889): 	at android.os.Looper.loop(Looper.java:351)
E/AndroidRuntime(  889): 	at android.app.ActivityThread.main(ActivityThread.java:8380)
E/AndroidRuntime(  889): 	at java.lang.reflect.Method.invoke(Native Method)
E/AndroidRuntime(  889): 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:584)
E/AndroidRuntime(  889): 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1013)
E/AndroidRuntime(  889): Caused by: java.lang.IllegalArgumentException: Unsupported value: 'com.boskokg.flutter_blue_plus.Protos$BluetoothState$Builder@ad808ee' of type 'class com.boskokg.flutter_blue_plus.Protos$BluetoothState$Builder'
E/AndroidRuntime(  889): 	at io.flutter.plugin.common.StandardMessageCodec.writeValue(StandardMessageCodec.java:297)
E/AndroidRuntime(  889): 	at io.flutter.plugin.common.StandardMethodCodec.encodeSuccessEnvelope(StandardMethodCodec.java:63)
E/AndroidRuntime(  889): 	at io.flutter.plugin.common.EventChannel$IncomingStreamRequestHandler$EventSinkImplementation.success(EventChannel.java:251)
E/AndroidRuntime(  889): 	at com.boskokg.flutter_blue_plus.FlutterBluePlusPlugin$1.onReceive(FlutterBluePlusPlugin.java:1214)
E/AndroidRuntime(  889): 	at android.app.LoadedApk$ReceiverDispatcher$Args.lambda$getRunnable$0$android-app-LoadedApk$ReceiverDispatcher$Args(LoadedApk.java:1907)
E/AndroidRuntime(  889): 	... 9 more
I/Process (  889): Sending signal. PID: 889 SIG: 9

ulscwfn, 2023-06-17:
Version 1.61 Is this issue fixed?

deekshithx, 2023-06-19:
+1, Any update? happens with the package example app too.

STR:
1) Turn off Bluetooth and Open the app.
2) Turn On Bluetooth via notification drawer 
-- App Crashes --

HiTools, 2023-06-28:
+1  me too.

deekshithx, 2023-06-28:
I was able to fix the issue by declaring the location permission in AndroidManifest.xml and also by using the latest flutter_blue_plus version 1.5.2.
Note: Remove the ^ from the **flutter_blue_plus: ^1.5.2** --> **flutter_blue_plus: 1.5.2** so that your project doesn't pull the Retracted versions of flutter_blue_plus i.e 1.6.1 and 1.6.0

chipweinberger, 2023-07-20:
try 1.7.5+. reopen if you have problems.

closed 2023-07-20 by chipweinberger