chipweinberger/flutter_blue_plus #342: Sudden Disconnect Causes App To Crash

FXschwartz, 2023-06-15:
I'm still investigating this and trying to track down the root cause, but I wanted to create an issue in case anyone else sees this.

The only change I made was updating to the latest flutter_blue_plus version. Now, I'm seeing random app crashes on iOS if the app loses connection to a paired Bluetooth device. Crashylitcs logs aren't helpful, but I'll keep updating this as I learn more.

It seems to happen every fourth lost connection. I've been unable to reproduce it when built locally, it only happens for prod builds. I also don't know if it affects Android.

Tested on iPhone 14 Pro Max

chipweinberger, 2023-06-19:
please open a pr if you fix it

18723309899, 2023-06-20:
> 如果你修好了，请打开一个pr

The reason for this error is that the Kotlin version is too high. My previous version was 1.6.10 and there were no error reports. After upgrading to 1.7.0, this issue occurred. We hope the author can fix this bug as soon as possible

chipweinberger, 2023-06-20:
please submit a PR if you want it to be fixed. thanks.

18723309899, 2023-06-20:
> please submit a PR if you want it to be fixed. thanks.
> 
> In not sure where kotlin version is specified, or how it relates to this bug.

I will try, but I'm not sure if I can fix it, it may take some time

FXschwartz, 2023-06-20:
> > 如果你修好了，请打开一个pr
> 
> The reason for this error is that the Kotlin version is too high. My previous version was 1.6.10 and there were no error reports. After upgrading to 1.7.0, this issue occurred. We hope the author can fix this bug as soon as possible

So far this is only happening to me on iOS. The Kotlin version shouldn't be affecting anything like this.

WeiKuang, 2023-07-19:
I have some issues.
It looks like a reconnection issue. But I haven't found the specific crash information.

chipweinberger, 2023-07-19:
need a backtrack

WeiKuang, 2023-07-20:
I get crash info.
Exception Type:  EXC_CRASH (SIGABRT)
Exception Codes: 0x0000000000000000, 0x0000000000000000
Triggered by Thread:  0

Application Specific Information:
abort() called

Thread 0 name:   Dispatch queue: com.apple.main-thread
Thread 0 Crashed:
0   libsystem_kernel.dylib        	       0x1e4d3e558 __pthread_kill + 8
1   libsystem_pthread.dylib       	       0x205b83118 pthread_kill + 268
2   libsystem_c.dylib             	       0x1ad37bc04 __abort + 128
3   libsystem_c.dylib             	       0x1ad323184 abort + 192
4   libc++abi.dylib               	       0x205ac1bf8 abort_message + 132
5   libc++abi.dylib               	       0x205ab1444 demangling_terminate_handler() + 348
6   libobjc.A.dylib               	       0x19eea1ea4 _objc_terminate() + 144
7   libc++abi.dylib               	       0x205ac0fbc std::__terminate(void (*)()) + 16
8   libc++abi.dylib               	       0x205ac0f60 std::terminate() + 56
9   libdispatch.dylib             	       0x1ad2c2ec0 _dispatch_client_callout + 40
10  libdispatch.dylib             	       0x1ad2d16a4 _dispatch_main_queue_drain + 928
11  libdispatch.dylib             	       0x1ad2d12f4 _dispatch_main_queue_callback_4CF + 44
12  CoreFoundation                	       0x1a5e8bc28 __CFRUNLOOP_IS_SERVICING_THE_MAIN_DISPATCH_QUEUE__ + 16
13  CoreFoundation                	       0x1a5e6d560 __CFRunLoopRun + 1992
14  CoreFoundation                	       0x1a5e723ec CFRunLoopRunSpecific + 612
15  GraphicsServices              	       0x1e130f35c GSEventRunModal + 164
16  UIKitCore                     	       0x1a81ff6e8 -[UIApplication _run] + 888
17  UIKitCore                     	       0x1a81ff34c UIApplicationMain + 340
18  Runner                        	       0x1001a9b38 main + 64
19  dyld                          	       0x1c534adec start + 2220

Thread 1 name:  com.apple.uikit.eventfetch-thread
Thread 1:
0   libsystem_kernel.dylib        	       0x1e4d37c84 mach_msg2_trap + 8
1   libsystem_kernel.dylib        	       0x1e4d4ab54 mach_msg2_internal + 80
2   libsystem_kernel.dylib        	       0x1e4d4ae2c mach_msg_overwrite + 540
3   libsystem_kernel.dylib        	       0x1e4d381c8 mach_msg + 24
4   CoreFoundation                	       0x1a5e6c024 __CFRunLoopServiceMachPort + 160
5   CoreFoundation                	       0x1a5e6d250 __CFRunLoopRun + 1208
6   CoreFoundation                	       0x1a5e723ec CFRunLoopRunSpecific + 612
7   Foundation                    	       0x1a00f2fd4 -[NSRunLoop(NSRunLoop) runMode:beforeDate:] + 212
8   Foundation                    	       0x1a00f2ebc -[NSRunLoop(NSRunLoop) runUntilDate:] + 64
9   UIKitCore                     	       0x1a83323fc -[UIEventFetcher threadMain] + 416
10  Foundation                    	       0x1a010c544 __NSThread__start__ + 716
11  libsystem_pthread.dylib       	       0x205b7d6b8 _pthread_start + 148
12  libsystem_pthread.dylib       	       0x205b7cb88 thread_start + 8

Thread 2 name:  io.flutter.1.ui
Thread 2:
0   libsystem_kernel.dylib        	       0x1e4d37c84 mach_msg2_trap + 8
1   libsystem_kernel.dylib        	       0x1e4d4ab54 mach_msg2_internal + 80
2   libsystem_kernel.dylib        	       0x1e4d4ae2c mach_msg_overwrite + 540
3   libsystem_kernel.dylib        	       0x1e4d381c8 mach_msg + 24
4   CoreFoundation                	       0x1a5e6c024 __CFRunLoopServiceMachPort + 160
5   CoreFoundation                	       0x1a5e6d250 __CFRunLoopRun + 1208
6   CoreFoundation                	       0x1a5e723ec CFRunLoopRunSpecific + 612
7   Flutter                       	       0x1009e254c 0x1006fc000 + 3040588
8   Flutter                       	       0x1009e203c 0x1006fc000 + 3039292
9   libsystem_pthread.dylib       	       0x205b7d6b8 _pthread_start + 148
10  libsystem_pthread.dylib       	       0x205b7cb88 thread_start + 8

Thread 3 name:  io.flutter.1.raster
Thread 3:
0   libsystem_kernel.dylib        	       0x1e4d37c84 mach_msg2_trap + 8
1   libsystem_kernel.dylib        	       0x1e4d4ab54 mach_msg2_internal + 80
2   libsystem_kernel.dylib        	       0x1e4d4ae2c mach_msg_overwrite + 540
3   libsystem_kernel.dylib        	       0x1e4d381c8 mach_msg + 24
4   CoreFoundation                	       0x1a5e6c024 __CFRunLoopServiceMachPort + 160
5   CoreFoundation                	       0x1a5e6d250 __CFRunLoopRun + 1208
6   CoreFoundation                	       0x1a5e723ec CFRunLoopRunSpecific + 612
[... 213 lines cut ...]
       0x100500000 -        0x10050bfff path_provider_foundation arm64  <e9373114ecb73d6f89d56a509cfb4241> /private/var/containers/Bundle/Application/958A041F-52CF-4366-9955-322B747D71BF/Runner.app/Frameworks/path_provider_foundation.framework/path_provider_foundation
       0x100520000 -        0x10052ffff qr_code_scanner arm64  <f0442bde0924345c9a253e40e0af2e06> /private/var/containers/Bundle/Application/958A041F-52CF-4366-9955-322B747D71BF/Runner.app/Frameworks/qr_code_scanner.framework/qr_code_scanner
       0x100550000 -        0x10055bfff share_plus arm64  <16b0e6cef746310ab929de94ef19797b> /private/var/containers/Bundle/Application/958A041F-52CF-4366-9955-322B747D71BF/Runner.app/Frameworks/share_plus.framework/share_plus
       0x10056c000 -        0x100577fff shared_preferences_foundation arm64  <6ad712fc0a363fb891fc57930653a78b> /private/var/containers/Bundle/Application/958A041F-52CF-4366-9955-322B747D71BF/Runner.app/Frameworks/shared_preferences_foundation.framework/shared_preferences_foundation
[... cut ...]
       0x1a7e62000 -        0x1a9686fff UIKitCore arm64e  <b3834960244b34e49ea0ca4bb44ef0f3> /System/Library/PrivateFrameworks/UIKitCore.framework/UIKitCore
       0x1c5335000 -        0x1c53baa23 dyld arm64e  <6987370ac3853135a27f6731706bcbd8> /usr/lib/dyld
       0x1a00b1000 -        0x1a098efff Foundation arm64e  <6e76dc9611af3b2eb71e215f9cc6e822> /System/Library/Frameworks/Foundation.framework/Foundation
       0x1a6e04000 -        0x1a71cffff CFNetwork arm64e  <6aafe7c4f1c43020ad1670591c86d7b0> /System/Library/Frameworks/CFNetwork.framework/CFNetwork

EOF

chipweinberger, 2023-07-20:
I dont see anything particularly specific to FBP in that. I am not able to debug it.

chipweinberger, 2023-07-25:
It would also be useful to have the logs immediately before the crash.

Currently, I think this is not a FlutterBluePlus issue. 

Perhaps the issue is in your own code, or in the Dart VM code.

WeiKuang, 2023-07-25:
@chipweinberger 
Yes, you are right. My problem is with my own code.

chipweinberger, 2023-07-25:
glad you figured it out!

closed 2023-07-25 by chipweinberger