chipweinberger/flutter_blue_plus #343: connect() crashes the app

mohamedsl22, 2023-06-18:
on first connection to device the app works fine, but if I save the scan id of the device and I close the app and reopen it to reconnect automatically the app crashes when it calls device.connect(), it was working fine with version 1.4.0 but with 1.5.2 it crashes !!

eric-taix, 2023-06-26:
Same symptom for me. Here is the stacktrace if it can help :
```
flutter: [cubit] 🐛 DeviceSynchroCubit Change { currentState: Instance of 'DeviceSynchroInitial', nextState: Instance of 'DeviceSynchroConnecting' }
*** Terminating app due to uncaught exception of class 'FlutterError'
libc++abi: terminating with uncaught exception of type FlutterError
* thread #1, queue = 'com.apple.main-thread', stop reason = signal SIGABRT
    frame #0: 0x00000001b09c97b0 libsystem_kernel.dylib`__pthread_kill + 8
libsystem_kernel.dylib`:
->  0x1b09c97b0 <+8>:  b.lo   0x1b09c97cc               ; <+36>
    0x1b09c97b4 <+12>: stp    x29, x30, [sp, #-0x10]!
    0x1b09c97b8 <+16>: mov    x29, sp
    0x1b09c97bc <+20>: bl     0x1b09a61fc               ; cerror_nocancel
Target 0: (Runner) stopped.
Lost connection to device.
```

I had to remove the application and reinstall it to be able to use the application again.

@mohamedsl22 Are you saying that v1.4.0 works fine? Did you you try to downgrade?

eric-taix, 2023-06-26:
I tried v1.4.0: my app did not crash anymore but was unable to connect to the peripheral!

It seems to be related to #349 and seems to be fixed into #350 . As the PR says that it works without scanning again, I tried to manually scan and without waiting the end of the scan, to connect to the device: it works even with latest version without crash.

Also I've noticed that the first scan did not work: 
`[CoreBluetooth] API MISUSE: <CBCentralManager: 0x281267300> can only accept this command while in the powered on state`.
But the 2nd try works.

Waiting for #350 to be merged to see if everything is solved ;-)

mohamedsl22, 2023-06-26:
hey @eric-taix , yes I does work with version 1.4.0, 
my problem with version 1.5.2 is that when I connect to device after making the scan -> it works fine, but if I closes the app and open it and tries to connect to the device ( with the saved scanId) automatically using only connect it crashes, 
here is my code hopefully it can help you:
`
connectionStatus2(String relevantScanId){
late StreamSubscription<BluetoothDeviceState> deviceState; 
    var protoBt = ProtoBluetoothDevice.BluetoothDevice(
          remoteId: relevantScanId
          );

    
    var device = BluetoothDevice.fromProto(protoBt);
    BluetoothDeviceState currEvent;
    
    try {
      deviceState = device.state.listen((event) async {
      currEvent = event;
          print('eventttt $event');
          if (event == BluetoothDeviceState.disconnected) {
            await device.connect(timeout: const Duration(seconds: 10),autoConnect: Platform.isAndroid ? false:true)
            .catchError((error) async{
              print("errrrror: ${error.toString()}");
              if(error.toString().contains('TimeoutException')) {
                await device.disconnect();
                await deviceState.cancel();
                connectionStatus2(relevantScanId);
              }
              else if(error.toString().contains('PlatformException')){// didn't find ble device
                await device.disconnect();
                await deviceState.cancel();
                FlutterBluePlus.instance
                .startScan(timeout: const Duration(seconds: 5) ,withServices: [Guid(SERVICE_DEVICE_CONTROL.toUpperCase())]).whenComplete(() => connectionStatus2(relevantScanId));
              }
              });
          }
          else if(event == BluetoothDeviceState.connected){
            print('innnnnnnnn connect');
            if(Platform.isAndroid ){
              List<BluetoothDevice> boundedDevices= await FlutterBluePlus.instance.bondedDevices;
              List<BluetoothDevice> relevantBoudedDevices = boundedDevices.where((e) => e.id.id == relevantScanId)
              .toList(); 
              print('boundedDevicess ${boundedDevices.toList()}');
              while(currEvent == BluetoothDeviceState.connected && relevantBoudedDevices.isEmpty){
                await Future.delayed(const Duration(seconds: 1));
                boundedDevices= await FlutterBluePlus.instance.bondedDevices;
                relevantBoudedDevices = boundedDevices.where((e) => e.id.id == relevantScanId)
              .toList();
              print('relevantBoudedDevices ${relevantBoudedDevices.toList()}');
              }
              if(relevantBoudedDevices.isNotEmpty){
 
              }
            }
            else{
              print('in elseeeeee');
              Future.delayed(Duration(seconds:  5), () {
                petsProvider.updateConnectionStatus(relevantScanId, true);
              });
            }

          }
        });
    } catch (e) {
      print('eeeeee $e');
    }}`

chipweinberger, 2023-07-20:
try 1.7.5+. reopen if you have problems.

closed 2023-07-20 by chipweinberger