chipweinberger/flutter_blue_plus #345: Scanning devices not working in Samsung Android version 12+ . Though all the permission needed fro Android 12 or above is allowed.

Shimanto9sqr, 2023-06-20:


Zeitgroup, 2023-07-07:
I noticed a problem with Samsung A33 and A53, with my app that uses f_b_p people can't find beacons after they update latest fw. So I bought an A33 and first of update it worked well, then i updated with latest fw A336BXXU6CWF2 and scanning failed ever.
After some trials I noticed that the problem was distance: if I move near the beacon (some cm) device finds the item.
So the problem is about latest fw of Samsung devices, and I hope Samsung fix the problem

guagnelz, 2023-07-07:
I noticed a problem with Samsung A33 and A53, with my app that uses f_b_p people can't find beacons after they update latest fw. So I bought an A33 and first of update it worked well, then i updated with latest fw A336BXXU6CWF2 and scanning failed ever.
After some trials I noticed that the problem was distance: if I move near the beacon (some cm) device finds the item.
So the problem is about latest fw of Samsung devices, and I hope Samsung fix the problem

chipweinberger, 2023-07-20:
please try 1.7.5+. reopen if you have issues.

closed 2023-07-20 by chipweinberger