chipweinberger/flutter_blue_plus #348: How do you maintain bluetooth connection across app launches?

c3ni3i, 2023-06-22:
I know that certain apps that connect to bluetooth fitness wearables maintain the connection across app launches. 

You search for your device and pair only once during onboarding. After that, the app automatically pairs with your wearable and streams data over bluetooth without you needing to scan and connect.

How do these apps achieve this behavior?

alexxgarci, 2023-06-23:
You can use the `autoConnect` flag in `connect` function.

alexxgarci, 2023-06-23:
> I know that certain apps that connect to bluetooth fitness wearables maintain the connection across app launches.
>
> You search for your device and pair only once during onboarding. After that, the [...]

You can use the `autoConnect` flag in `connect` function.

c3ni3i, 2023-06-23:
It looks like the `autoConnect` flag only sets the `androidAutoConnect` variable. How do we get auto connect behavior on iOS?

```
    /// Establishes a connection to the Bluetooth Device.
    Future<void> connect({
        Duration? timeout,
        bool autoConnect = true,
        bool shouldClearGattCache = true,
    }) async
    {
        if (Platform.isAndroid && shouldClearGattCache) {
            clearGattCache();
        }

        var request = protos.ConnectRequest.create()
            ..remoteId = id.toString()
            ..androidAutoConnect = autoConnect;
  
        var responseStream = state.where((s) => s == BluetoothDeviceState.connected);
  
        // Start listening now, before invokeMethod, to ensure we don't miss the response
        Future<BluetoothDeviceState> futureState = responseStream.first;
  
        await FlutterBluePlus.instance._channel
              .invokeMethod('connect', request.writeToBuffer());
  
        // wait for connection
        if (timeout != null) {
            await futureState.timeout(timeout, onTimeout: () {
                throw TimeoutException('Failed to connect in time.', timeout);
            });
        } else {
            await futureState;
        }
    }
 ```

FXschwartz, 2023-06-23:
As far as I know this isn’t possible with this package. When your app closes it kills the Bluetooth connection

keskink, 2023-06-24:
I cloned this package and changed codes in iOS. Now I can connect with protos w/o scan.

c3ni3i, 2023-06-28:
> I cloned this package and changed codes in iOS. Now I can connect with protos w/o scan.

Can you please explain what you changed or share the updates? Thanks

keskink, 2023-07-05:
@property(nonatomic, retain) CBPeripheral *fxPeripheral;
.
.
.
.
else if ([@"connect" isEqualToString:call.method])
    {
        FlutterStandardTypedData *data = [call arguments];
        ProtosConnectRequest *request = [[ProtosConnectRequest alloc] initWithData:[data data] error:nil];
        NSString *remoteId = [request remoteId];
        @try
        {
            CBPeripheral *peripheral = [_scannedPeripherals objectForKey:remoteId];

           if (peripheral == nil)
            {

                NSArray<CBPeripheral *> *periphs =
                    [_centralManager retrievePeripheralsWithIdentifiers:@[ [[NSUUID alloc] initWithUUIDString:remoteId] ]];

                for (CBPeripheral *p in periphs)
                {
                    NSString *uuid = [[p identifier] UUIDString];
                    p.delegate = self;
                    [_scannedPeripherals setObject:p forKey:uuid];
                }
                peripheral = [_scannedPeripherals objectForKey:remoteId];
                if (peripheral != nil) {
                    NSLog(@"retrievePeripheralsWithIdentifiers bulundu %@", [peripheral.identifier UUIDString]);
                }

            }

            if (peripheral == nil)
            {
                NSArray *periphs = [self->_centralManager
                    retrieveConnectedPeripheralsWithServices:@[ [CBUUID UUIDWithString:@"1800"] ]];
                for (CBPeripheral *p in periphs)
                {
                    NSString *uuid = [[p identifier] UUIDString];
                    p.delegate = self;
                    [_scannedPeripherals setObject:p forKey:uuid];
                }
                peripheral = [_scannedPeripherals objectForKey:remoteId];
            }

            if (peripheral == nil)
            {
                @throw [FlutterError errorWithCode:@"connect" message:@"Peripheral not found" details:nil];
            }
            // TODO: Implement Connect options (#36)
            //NSLog(@"connecting -- %@", [peripheral.identifier UUIDString]);
            self.fxPeripheral = peripheral;

            [_centralManager connectPeripheral:self.fxPeripheral options:nil];
            result(nil);
        }
        @catch (NSException *e)
        {
            result(e);
        }
    }

MrCsabaToth, 2023-07-19:
Just hypothetical: if you place your Bluetooth code into a background worker and you run things in an isolate theoretically that can stay alive even after your app quits. If it's not in a background task then the connection will be killed at app close.

WeiKuang, 2023-07-20:
on iOS,you can use 
`
- (void) centralManager:(CBCentralManager *)central willRestoreState:(NSDictionary *)state {
    NSArray *peripherals = state[CBCentralManagerRestoredStatePeripheralsKey];
    [peripherals enumerateObjectsUsingBlock:^(CBPeripheral *peripheral, NSUInteger idx, BOOL *stop) {
        peripheral.delegate = self;
        if (peripheral.state == CBPeripheralStateDisconnected) {
            [central connectPeripheral:peripheral options:nil];
        }
    }];
}
`

WeiKuang, 2023-07-20:
on iOS,you can use
`- (void) centralManager:(CBCentralManager *)central willRestoreState:(NSDictionary *)state {
    NSArray *peripherals = state[CBCentralManagerRestoredStatePeripheralsKey];
    [peripherals enumerateObjectsUsingBlock:^(CBPeripheral *peripheral, NSUInteger idx, BOOL *stop) {
        peripheral.delegate = self;
        if (peripheral.state == CBPeripheralStateDisconnected) {
            [central connectPeripheral:peripheral options:nil];
        }
    }];
}`

WeiKuang, 2023-07-20:
```objective-c
- (void) centralManager:(CBCentralManager *)central willRestoreState:(NSDictionary *)state {
    NSArray *peripherals = state[CBCentralManagerRestoredStatePeripheralsKey];
    [peripherals enumerateObjectsUsingBlock:^(CBPeripheral *peripheral, NSUInteger idx, BOOL *stop) {
        peripheral.delegate = self;
        if (peripheral.state == CBPeripheralStateDisconnected) {
            [central connectPeripheral:peripheral options:nil];
        }
    }];
}
```

chipweinberger, 2023-07-20:
as of 1.7.0, you can connect without scanning. You just need to remember and reuse the same `remote_id`.

closed 2023-07-20 by chipweinberger