chipweinberger/flutter_blue_plus #351: not to use singleton mode

ludy520, 2023-06-26:
It is best not to use singleton mode, otherwise it will be troublesome to connect multiple devices at the same time

FXschwartz, 2023-06-26:
> It is best not to use singleton mode, otherwise it will be troublesome to connect multiple devices at the same time

I don't see how using a singleton makes it troublesome to connect to multiple devices at the same time. Connecting to multiple devices at the same time is a very easy thing to do that I haven't seen any issues with.

ludy520, 2023-06-28:
> I don't see how using a singleton makes it troublesome to connect to multiple devices at the same time. Connecting to multiple devices at the same time is a very easy thing to do that I haven't seen [...]

For example, if I connect two devices at the same time and subscribe to the connection status in a single case, it is impossible to distinguish which device the current connection status belongs to.

FXschwartz, 2023-06-29:
> For example, if I connect two devices at the same time and subscribe to the connection status in a single case, it is impossible to distinguish which device the current connection status belongs to.

It is not impossible, you can use the device and device service from the connection and each will have its own unique ID.

MrCsabaToth, 2023-07-19:
Maybe @ludy520 is accidentally thinking about [QuickBlue](https://pub.dev/packages/quick_blue) which is actually mentioned at the top of the Flutter Blue Plus README. QuickBlue is sucha  singleton that cannot handle multiple connections (last time I looked some months ago). Flutter Blue Plus is definitely able to handle multiple devices and characteristics, I've been using it that way. Singleton pattern does not contradict that internally it can handle multiple machines and connections.

chipweinberger, 2023-07-20:
multiple devices works fine

closed 2023-07-20 by chipweinberger