chipweinberger/flutter_blue_plus #352: version 1.5.2 doesn't read or write on android 13

mohamedsl22, 2023-06-26:
Hey, 
I tried the app on android 13 (pixel 6) it connects to device but doesn't read or write to it, although sometimes it works ! 
with another android version it works fine! 
please help

mohamedsl22, 2023-07-02:
@FXschwartz 
only with android 13 I get this error when trying to read:
[ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: PlatformException(discover_services_error, no instance of BluetoothGatt, have you connected first?, java.lang.Exception: no instance of BluetoothGatt, have you connected first?

deverlex, 2023-07-07:
> @FXschwartz only with android 13 I get this error when trying to read: [ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: PlatformException(discover_services_error, no instance [...]

You need disconnect to blue before call to writing data

closed 2023-07-20 by chipweinberger