chipweinberger/flutter_blue_plus #356: Switching the device's Bluetooth on and off is an app crash.

kalismeras61, 2023-06-28:
The app crashes when I switch bluetooth off or on.  Because of below.  Same issue happening both IOS and Andorid. 

E/AndroidRuntime( 5702): java.lang.RuntimeException: Error receiving broadcast Intent { act=android.bluetooth.adapter.action.STATE_CHANGED flg=0x5000010 (has extras) } in com.boskokg.flutter_blue_plus.FlutterBluePlusPlugin$1@4de05ee
E/AndroidRuntime( 5702): 	at android.app.LoadedApk$ReceiverDispatcher$Args.lambda$getRunnable$0$LoadedApk$ReceiverDispatcher$Args(Unknown Source:222)
E/AndroidRuntime( 5702): 	at android.app.LoadedApk$ReceiverDispatcher$Args$$ExternalSyntheticLambda0.run(Unknown Source:2)
E/AndroidRuntime( 5702): 	at android.os.Handler.handleCallback(Unknown Source:2)
E/AndroidRuntime( 5702): 	at android.os.Handler.dispatchMessage(Unknown Source:4)
E/AndroidRuntime( 5702): 	at android.os.Looper.loopOnce(Unknown Source:176)
E/AndroidRuntime( 5702): 	at android.os.Looper.loop(Unknown Source:76)
E/AndroidRuntime( 5702): 	at android.app.ActivityThread.main(Unknown Source:138)
E/AndroidRuntime( 5702): 	at java.lang.reflect.Method.invoke(Native Method)
E/AndroidRuntime( 5702): 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(Unknown Source:11)
E/AndroidRuntime( 5702): 	at com.android.internal.os.ZygoteInit.main(Unknown Source:309)
E/AndroidRuntime( 5702): Caused by: java.lang.IllegalArgumentException: Unsupported value: 'com.boskokg.flutter_blue_plus.Protos$BluetoothState$Builder@5a2ce37' of type 'class com.boskokg.flutter_blue_plus.Protos$BluetoothState$Builder'
E/AndroidRuntime( 5702): 	at io.flutter.plugin.common.StandardMessageCodec.writeValue(StandardMessageCodec.java:297)

LvJson, 2023-06-29:
I had the same problem today！

//******************************************************///
E/AndroidRuntime( 6725): FATAL EXCEPTION: main
E/AndroidRuntime( 6725): Process: com.dc.flutter_jiachao, PID: 6725
E/AndroidRuntime( 6725): java.lang.RuntimeException: Error receiving broadcast Intent { act=android.bluetooth.adapter.action.STATE_CHANGED flg=0x15000010 (has extras) } in com.boskokg.flutter_blue_plus.FlutterBluePlusPlugin$1@e77c0b1
E/AndroidRuntime( 6725): 	at android.app.LoadedApk$ReceiverDispatcher$Args.lambda$getRunnable$0$LoadedApk$ReceiverDispatcher$Args(LoadedApk.java:1805)
E/AndroidRuntime( 6725): 	at android.app.-$$Lambda$LoadedApk$ReceiverDispatcher$Args$_BumDX2UKsnxLVrE6UJsJZkotuA.run(Unknown Source:2)
E/AndroidRuntime( 6725): 	at android.os.Handler.handleCallback(Handler.java:900)
E/AndroidRuntime( 6725): 	at android.os.Handler.dispatchMessage(Handler.java:103)
E/AndroidRuntime( 6725): 	at android.os.Looper.loop(Looper.java:219)
E/AndroidRuntime( 6725): 	at android.app.ActivityThread.main(ActivityThread.java:8673)
E/AndroidRuntime( 6725): 	at java.lang.reflect.Method.invoke(Native Method)
E/AndroidRuntime( 6725): 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:513)
E/AndroidRuntime( 6725): 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1109)
E/AndroidRuntime( 6725): Caused by: java.lang.IllegalArgumentException: Unsupported value: 'com.boskokg.flutter_blue_plus.Protos$BluetoothState$Builder@3888820' of type 'class com.boskokg.flutter_blue_plus.Protos$BluetoothState$Builder'
E/AndroidRuntime( 6725): 	at io.flutter.plugin.common.StandardMessageCodec.writeValue(StandardMessageCodec.java:297)
E/AndroidRuntime( 6725): 	at io.flutter.plugin.common.StandardMethodCodec.encodeSuccessEnvelope(StandardMethodCodec.java:61)
E/AndroidRuntime( 6725): 	at io.flutter.plugin.common.EventChannel$IncomingStreamRequestHandler$EventSinkImplementation.success(EventChannel.java:251)
E/AndroidRuntime( 6725): 	at com.boskokg.flutter_blue_plus.FlutterBluePlusPlugin$1.onReceive(FlutterBluePlusPlugin.java:1214)
E/AndroidRuntime( 6725): 	at android.app.LoadedApk$ReceiverDispatcher$Args.lambda$getRunnable$0$LoadedApk$ReceiverDispatcher$Args(LoadedApk.java:1795)
E/AndroidRuntime( 6725): 	... 8 more

Ervin2018, 2023-06-30:
I had the same problem today！My version of flutter is 3.10.5
W/om.baicycle.app( 6894): type=1400 audit(0.0:3778512): avc: denied { search } for name="miuilog" dev="dm-40" ino=447798 scontext=u:r:untrusted_app_30:s0:c182,c258,c512,c768 tcontext=u:object_r:data_log_file:s0 tclass=dir permissive=0 app=com.baicycle.app
W/OOMEventManagerFK( 6894): Failed to mkdir /data/miuilog/stability/memleak/heapdump/
I/om.baicycle.app( 6894): oneway function results for code 1599295570 on binder at 0xb4000074672468e0 will be dropped but finished with status UNKNOWN_TRANSACTION
W/BpBinder( 6894): PerfMonitor binderTransact :  time=404ms interface=android.app.IActivityManager code=7
I/Process ( 6894): Process is going to kill itself!
I/Process ( 6894): java.lang.Exception
I/Process ( 6894): 	at android.os.Process.killProcess(Process.java:1344)
I/Process ( 6894): 	at com.android.internal.os.RuntimeInit$KillApplicationHandler.uncaughtException(RuntimeInit.java:181)
I/Process ( 6894): 	at java.lang.ThreadGroup.uncaughtException(ThreadGroup.java:1073)
I/Process ( 6894): 	at java.lang.ThreadGroup.uncaughtException(ThreadGroup.java:1068)
I/Process ( 6894): 	at java.lang.Thread.dispatchUncaughtException(Thread.java:2306)
I/Process ( 6894): Sending signal. PID: 6894 SIG: 9

Ervin2018, 2023-06-30:
I think it was this code that caused it：flutterBlue.state.listen

LipJ01, 2023-07-03:
same, flutter 3.10.2. Android 11.
D/AndroidRuntime( 2750): Shutting down VM
E/AndroidRuntime( 2750): FATAL EXCEPTION: main
E/AndroidRuntime( 2750): Process: com.liteip.microZ, PID: 2750
E/AndroidRuntime( 2750): java.lang.RuntimeException: Error receiving broadcast Intent { act=android.bluetooth.adapter.action.STATE_CHANGED flg=0x5000010 (has extras) } in com.boskokg.flutter_blue_plus.FlutterBluePlusPlugin$1@f4c6542
E/AndroidRuntime( 2750):        at android.app.LoadedApk$ReceiverDispatcher$Args.lambda$getRunnable$0$LoadedApk$ReceiverDispatcher$Args(LoadedApk.java:1640)
E/AndroidRuntime( 2750):        at android.app.-$$Lambda$LoadedApk$ReceiverDispatcher$Args$_BumDX2UKsnxLVrE6UJsJZkotuA.run(Unknown Source:2)
E/AndroidRuntime( 2750):        at android.os.Handler.handleCallback(Handler.java:938)
E/AndroidRuntime( 2750):        at android.os.Handler.dispatchMessage(Handler.java:99)
E/AndroidRuntime( 2750):        at android.os.Looper.loop(Looper.java:223)
E/AndroidRuntime( 2750):        at android.app.ActivityThread.main(ActivityThread.java:7888)
E/AndroidRuntime( 2750):        at java.lang.reflect.Method.invoke(Native Method)
E/AndroidRuntime( 2750):        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:592)
E/AndroidRuntime( 2750):        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:981)
E/AndroidRuntime( 2750): Caused by: java.lang.IllegalArgumentException: Unsupported value: 'com.boskokg.flutter_blue_plus.Protos$BluetoothState$Builder@603ddc5' of type 'class com.boskokg.flutter_blue_plus.Protos$BluetoothState$Builder'
E/AndroidRuntime( 2750):        at io.flutter.plugin.common.StandardMessageCodec.writeValue(StandardMessageCodec.java:297)
E/AndroidRuntime( 2750):        at io.flutter.plugin.common.StandardMethodCodec.encodeSuccessEnvelope(StandardMethodCodec.java:61)
E/AndroidRuntime( 2750):        at io.flutter.plugin.common.EventChannel$IncomingStreamRequestHandler$EventSinkImplementation.success(EventChannel.java:251)
E/AndroidRuntime( 2750):        at com.boskokg.flutter_blue_plus.FlutterBluePlusPlugin$1.onReceive(FlutterBluePlusPlugin.java:1214)
E/AndroidRuntime( 2750):        at android.app.LoadedApk$ReceiverDispatcher$Args.lambda$getRunnable$0$LoadedApk$ReceiverDispatcher$Args(LoadedApk.java:1622)
E/AndroidRuntime( 2750):        ... 8 more
I/Process ( 2750): Sending signal. PID: 2750 SIG: 9
Lost connection to device.

chipweinberger, 2023-07-05:
If you fix it, please open a PR. I'm not sure the cause.

kalismeras61, 2023-07-06:
> E/AndroidRuntime( 2750): java.lang.RuntimeException: Error receiving broadcast Intent { act=android.bluetooth.adapter.action.STATE_CHANGED flg=0x5000010 (has extras) } in com.boskokg.flutter_blue_plus.FlutterBluePlusPlugin$1@f4c6542

Ok but not sure it is also fix below state issue ? 

E/AndroidRuntime( 2750): java.lang.RuntimeException: Error receiving broadcast Intent { act=android.bluetooth.adapter.action.STATE_CHANGED flg=0x5000010 (has extras) } in com.boskokg.flutter_blue_plus.FlutterBluePlusPlugin$1@f4c6542

chipweinberger, 2023-07-14:
try version 1.7.1

closed 2023-07-18 by chipweinberger