chipweinberger/flutter_blue_plus #361: Commit: [Protobuf] remove protobuf from iOS and MacOS. todo: android breaks Android (at least V7.0)

Staars, 2023-07-05:
https://github.com/boskokg/flutter_blue_plus/commit/1848617e6ce25385c2a7b4129aa199801bcfee71

breaks BLE on (my only Android testing device running 7.0). It throws an exception at the flame symbol below and shows BLE adapter as unavailable afterwards.

```
  /// Gets the current state of the Bluetooth module
  Stream<BluetoothState> get state async* {
    BluetoothState initialState = 🔥  await _channel
        .invokeMethod('state')
        .then((buffer) => BmBluetoothPowerState.fromJson(buffer))
        .then((s) => bmToBluetoothState(s.state));
```

Every older commit does not compile for macOS, which does not leave the option for me to just use an older version.

chipweinberger, 2023-07-05:
This is expected. Android is still _todo_ on master branch. You'll have to wait, or fix the MacOS compile errors yourself.

Staars, 2023-07-06:
Thanks for the quick answer, my experience with Android is close to zero (starting yesterday with the old testing device).

Nevertheless I tried my best to understand the compiler and after the very recent commits I did add one large try{}catch{} in 

```
    @Override
    public void onMethodCall(@NonNull MethodCall call,
                                 @NonNull Result result)
    {
        if(mBluetoothAdapter == null && !"isAvailable".equals(call.method)) {
            result.error("bluetooth_unavailable", "the device does not have bluetooth", null);
            return;
        }

        try{
          switch (call.method) {
           .... big chunk of code ...
          }
        }
        catch(Exception e) {
            result.error("bluetooth_unavailable", "undefined exception", null);
            return;
        }
    }
```

Without it I could not compile for Android.

This is the first time, that I can use all my test devices (macOs, Android an iOS) with one code base.

Anyway, thanks a ton for this project!!

chipweinberger, 2023-07-06:
i'm glad it's working for you! 

thanks for the nice follow up.

Staars, 2023-07-06:
Another follow up:
Sadly the most recent commit does throw an exception later. So Bluetooth is available, I can successfully scan, find and connect to the device (ESP32). But then discoverservices() throws.
I have no idea, if this is related to my very old Android test device.

Just for other users that may stumble across, here is my current workaround in pub spec.yaml:
```
  flutter_blue_plus:
      git: 
        url: https://github.com/boskokg/flutter_blue_plus.git
        ref: 1.6.1 # use for Android, comment out for macOS
```

I will try to educate myself a bit, maybe I can contribute something later on.

closed 2023-07-11 by Staars