chipweinberger/flutter_blue_plus #362: Unsupported value for standard codec  for iOS

JDongKhan, 2023-07-07:
It has crash when return NSException.

```  
Trapped uncaught exception 'NSInternalInconsistencyException', reason: 'Unsupported value for standard codec' 
(
	0   CoreFoundation                      0x00000001922017a8 50CF3336-313F-3A7D-9048-CB1ED8EC3368 + 1222568
	1   libobjc.A.dylib                     0x0000000191f23bcc objc_exception_throw + 56
	2   CoreFoundation                      0x000000019210482c 50CF3336-313F-3A7D-9048-CB1ED8EC3368 + 186412
	3   Foundation                          0x000000019258a584 C82F8A4F-3D0F-33DB-8D40-C1E8BEF27E56 + 873860
	4   Flutter                             0x00000001058debf4 _ZL16WriteValueOfTypePKvP8__CFData28FlutterStandardCodecObjcTypeS0_ + 768
	5   Flutter                             0x00000001058de748 -[FlutterStandardWriter writeValue:] + 52
	6   Flutter                             0x00000001058dfab0 -[FlutterStandardMethodCodec encodeSuccessEnvelope:] + 136
	7   Flutter                             0x00000001058daa08 __45-[FlutterMethodChannel setMethodCallHandler:]_block_invoke_2 + 184
	8   flutter_blue_plus                   0x0000000104c6f794 -[FlutterBluePlusPlugin handleMethodCall:result:] + 8760
	9   Flutter                             0x00000001058da72c __45-[FlutterMethodChannel setMethodCallHandler:]_block_invoke + 172
	10  Flutter                             0x0000000105334b80 ___ZN7flutter25PlatformMessageHandlerIos21HandlePlatformMessageENSt21_LIBCPP_ABI_NAMESPACE10unique_ptrINS_15PlatformMessageENS1_14default_deleteIS3_EEEE_block_invoke + 116
	11  libdispatch.dylib                   0x0000000191ec69a8 652A10DA-A0E7-3B47-9C95-AF3DFE45A66C + 371112
	12  libdispatch.dylib                   0x0000000191ec7524 652A10DA-A0E7-3B47-9C95-AF3DFE45A66C + 374052
	13  libdispatch.dylib                   0x0000000191eaa6fc 652A10DA-A0E7-3B47-9C95-AF3DFE45A66C + 255740
	14  CoreFoundation                      0x000000019217f7fc 50CF3336-313F-3A7D-9048-CB1ED8EC3368 + 690172
	
      15  CoreFoundation                      0x000000019217a6d0 50CF3336-313F-3A7D-9048-CB1ED8EC3368 + 669392
	16  CoreFoundation                      0x0000000192179ce8 CFRunLoopRunSpecific + 424
	17  GraphicsServices                    0x000000019c2c438c GSEventRunModal + 160
	18  UIKitCore                           0x00000001962a8444 UIApplicationMain + 1932
	19  Runner                              0x00000001047dd060 main + 64
	20  libdyld.dylib                       0x00000001920018f0 3D6D64B4-CB2B-3CC4-A7E9-02774DF7AE74 + 6384
)
```

[image]

JDongKhan, 2023-07-07:
```result([e description]);```  instead of  ```result(e); ```   will be solve this issue

chipweinberger, 2023-07-08:
`result([FlutterError errorWithCode:@"writeDescriptor" message:[e reason] details:NULL]);`

chipweinberger, 2023-07-14:
try version 1.7.1

JDongKhan, 2023-07-17:
@chipweinberger  thanks for fix this issue.

But,i face other issue when upgrade to 1.7.1

```
⛔ Exception: type 'Null' is not a subtype of type 'String' #0      new BmWriteDescriptorRequest.fromMap (package:flutter_blue_plus/src/bluetooth_msgs.dart:555:24)
I/flutter (26609): │ ⛔ #1      new BmWriteDescriptorResponse.fromMap (package:flutter_blue_plus/src/bluetooth_msgs.dart:572:41)
I/flutter (26609): │ ⛔ #2      BluetoothDescriptor.write.<anonymous closure>.<anonymous closure> (package:flutter_blue_plus/src/bluetooth_descriptor.dart:92:54)
I/flutter (26609): │ ⛔ #3      _MapStream._handleData (dart:async/stream_pipe.dart:213:31)
I/flutter (26609): │ ⛔ #4      _ForwardingStreamSubscription._handleData (dart:async/stream_pipe.dart:153:13)
I/flutter (26609): │ ⛔ #5      _rootRunUnary (dart:async/zone.dart:1407:47)
I/flutter (26609): │ ⛔ #6      _CustomZone.runUnary (dart:async/zone.dart:1308:19)
I/flutter (26609): │ ⛔ #7      _CustomZone.runUnaryGuarded (dart:async/zone.dart:1217:7)
I/flutter (26609): │ ⛔ #8      _BufferingStreamSubscription._sendData (dart:async/stream_impl.dart:339:11)
I/flutter (26609): │ ⛔ #9      _BufferingStreamSubscription._add (dart:async/stream_impl.dart:271:7)
I/flutter (26609): │ ⛔ #10     _ForwardingStreamSubscription._add (dart:async/stream_pipe.dart:123:11)
I/flutter (26609): │ ⛔ #11     _MapStream._handleData (dart:async/stream_pipe.dart:218:10)
I/flutter (26609): │ ⛔ #12     _ForwardingStreamSubscription._handleData (dart:async/stream_pipe.dart:153:13)
I/flutter (26609): │ ⛔ #13     _rootRunUnary (dart:async/zone.dart:1407:47)
I/flutter (26609): │ ⛔ #14     _CustomZone.runUnary (dart:async/zone.dart:1308:19)
I/flutter (26609): │ ⛔ #15     _CustomZone.runUnaryGuarded (dart:async/zone.dart:1217:7)
I/flutter (26609): │ ⛔ #16     _BufferingStreamSubscription._sendData (dart:async/stream_impl.dart:339:11)
I/flutter (26609): │ ⛔ #17     _BufferingStreamSubscription._add (dart:async/stream_impl.dart:271:7)
I/flutter (26609): │ ⛔ #18     _ForwardingStreamSubscription._add (dart:async/stream_pipe.dart:123:11)
I/flutter (26609): │ ⛔ #19     _WhereStream._handleData (dart:async/stream_pipe.dart:195:12)
I/flutter (26609): │ ⛔ #20     _ForwardingStreamSubscription._handleData (dart:async/stream_pipe.dart:153:13)
I/flutter (26609): │ ⛔ #21     _rootRunUnary (dart:async/zone.dart:1407:47)
I/flutter (26609): │ ⛔ #22     _CustomZone.runUnary (dart:async/zone.dart:1308:19)
I/flutter (26609): │ ⛔ #23     _CustomZone.runUnaryGuarded (dart:async/zone.dart:1217:7)
I/flutter (26609): │ ⛔ #24     _BufferingStreamSubscription._sendData (dart:async/stream_impl.dart:339:11)
I/flutter (26609): │ ⛔ #25     _DelayedData.perform (dart:async/stream_impl.dart:515:14)
I/flutter (26609): │ ⛔ #26     _PendingEvents.handleNext (dart:async/stream_impl.dart:620:11)
I/flutter (26609): │ ⛔ #27     _PendingEvents.schedule.<anonymous closure> (dart:async/stream_impl.dart:591:7)
I/flutter (26609): │ ⛔ #28     _rootRun (dart:async/zone.dart:1391:47)
I/flutter (26609): │ ⛔ #29     _CustomZone.run (dart:async/zone.dart:1301:19)
I/flutter (26609): │ ⛔ #30     _CustomZone.runGuarded (dart:async/zone.dart:1209:7)
I/flutter (26609): │ ⛔ #31     _CustomZone.bindCallbackGuarded.<anonymous closure> (dart:async/zone.dart:1249:23)
I/flutter (26609): │ ⛔ #32     _rootRun (dart:async/zone.dart:1399:13)
I/flutter (26609): │ ⛔ #33     _CustomZone.run (dart:async/zone.dart:1301:19)
I/flutter (26609): │ ⛔ #34     _CustomZone.runGuarded (dart:async/zone.dart:1209:7)
I/flutter (26609): │ ⛔ #35     _CustomZone.bindCallbackGuarded.<anonymous closure> (dart:async/zone.dart:1249:23)
I/flutter (26609): │ ⛔ #36     _microtaskLoop (dart:async/schedule_microtask.dart:40:21)
I/flutter (26609): │ ⛔ #37     _startMicrotaskLoop (dart:async/schedule_microtask.dart:49:5)
I/flutter (26609): │ ⛔ 
I/flutter (26609): │ ⛔  #0      BluetoothDescriptor.write.<anonymous closure> (package:flutter_blue_plus/src/bluetooth_descriptor.dart:113:7)
I/flutter (26609): │ ⛔ #1      _rootRunBinary (dart:async/zone.dart:1423:47)
I/flutter (26609): │ ⛔ #2      _CustomZone.runBinary (dart:async/zone.dart:1315:19)
I/flutter (26609): │ ⛔ #3      _FutureListener.handleError (dart:async/future_impl.dart:162:22)
I/flutter (26609): │ ⛔ #4      Future._propagateToListeners.handleError (dart:async/future_impl.dart:796:47)
I/flutter (26609): │ ⛔ #5      Future._propagateToListeners (dart:async/future_impl.dart:817:13)
I/flutter (26609): │ ⛔ #6      Future._completeError (dart:async/future_impl.dart:592:5)
I/flutter (26609): │ ⛔ #7      Future._asyncCompleteError.<anonymous closure> (dart:async/future_impl.dart:683:7)
I/flutter (26609): │ ⛔ #8      _rootRun (dart:async/zone.dart:1399:13)
I/flutter (26609): │ ⛔ #9      _CustomZone.run (dart:async/zone.dart:1301:19)
I/flutter (26609): │ ⛔ #10     _CustomZone.runGuarded (dart:async/zone.dart:1209:7)
I/flutter (26609): │ ⛔ #11     _CustomZone.bindCallbackGuarded.<anonymous closure> (dart:async/zone.dart:1249:23)
I/flutter (26609): │ ⛔ #12     _microtaskLoop (dart:async/schedule_microtask.dart:40:21)
I/flutter (26609): │ ⛔ #13     _startMicrotaskLoop (dart:async/schedule_microtask.dart:49:5)
I/flutter (26609): │ ⛔ 
```

chipweinberger, 2023-07-17:
use 1.7.2

chipweinberger, 2023-07-17:
iOS or android?

JDongKhan, 2023-07-17:
iOS
```

   // See BmWriteDescriptorRequest
    NSDictionary* request = @{
        @"remote_id":               [peripheral.identifier UUIDString],
        @"descriptor_uuid":         [descriptor.UUID fullUUIDString],
        @"service_uuid" :           [primaryService.UUID fullUUIDString],
        @"secondary_service_uuid":  secondaryService ? [secondaryService.UUID fullUUIDString] : [NSNull null],
        @"characteristic_uuid":     [descriptor.characteristic.UUID fullUUIDString],
        @"value":                   @"",
    };

    // See BmWriteDescriptorResponse
    NSDictionary* result = @{
        @"request": request,
        @"success": @(error == nil),
    };
```

android 

```
  HashMap<String, Object> request = new HashMap<>();
            request.put("remote_id", gatt.getDevice().getAddress());
            request.put("descriptor_uuid", descriptor.getUuid().toString());
            request.put("characteristic_uuid", descriptor.getCharacteristic().getUuid().toString());
            request.put("uuid", descriptor.getCharacteristic().getService().getUuid().toString());
```

dart
```
 return BmWriteDescriptorRequest(
      remoteId: json['remote_id'],
      descriptorUuid: json['descriptor_uuid'],
      serviceUuid: json['service_uuid'],
      secondaryServiceUuid: json['secondary_service_uuid'],
      characteristicUuid: json['characteristic_uuid'],
      value: _hexDecode(json['value'] ?? ""),
    );
```

They are inconsistent on Android  Devices.

chipweinberger, 2023-07-17:
should be fixed: 7236694b9429dc68b12b8ad9ec93a37e8db630e5

try master.

JDongKhan, 2023-07-17:
ok,thank you. I will try it.

JDongKhan, 2023-07-17:
miss secondary_service_uuid ? @chipweinberger

chipweinberger, 2023-07-17:
it is not necessary. thanks for asking.

JDongKhan, 2023-07-17:
```
  BmWriteDescriptorRequest({
    required this.remoteId,
    required this.descriptorUuid,
    required this.serviceUuid,
    required this.secondaryServiceUuid,
    required this.characteristicUuid,
    required this.value,
  });
```
But the secondaryServiceUuid is required on dart lib.

chipweinberger, 2023-07-17:
it is set like this:

```
  BmWriteDescriptorRequest(
    ...
    "secondaryServiceUuid": null
    ...
);
```

closed 2023-07-18 by chipweinberger