chipweinberger/flutter_blue_plus #363: java.lang.RuntimeException: Error receiving broadcast Intent { act=android.bluetooth.adapter.action.STATE_CHANGED

zinahe, 2023-07-07:
Hi,

I'm running the example code without any changes on a Nexus 4 android phone with Android version 5.1.1 and the app crashes with the following error message:

Any suggestions as to why this is happening?

Thanks.

```
The Flutter DevTools debugger and profiler on Nexus 4 is available at: http://127.0.0.1:9100?uri=http://127.0.0.1:32909/VGet0_uLTTc=/
D/BluetoothManager(10930): getConnectedDevices
D/BluetoothManager(10930): getConnectedDevices
D/BluetoothManager(10930): getConnectedDevices
D/BluetoothManager(10930): getConnectedDevices
D/BluetoothManager(10930): getConnectedDevices
D/BluetoothManager(10930): getConnectedDevices
D/BluetoothManager(10930): getConnectedDevices
D/BluetoothManager(10930): getConnectedDevices
D/BluetoothManager(10930): getConnectedDevices
D/BluetoothManager(10930): getConnectedDevices
D/BluetoothManager(10930): getConnectedDevices
D/AndroidRuntime(10930): Shutting down VM
E/AndroidRuntime(10930): FATAL EXCEPTION: main
E/AndroidRuntime(10930): Process: com.boskokg.flutter_blue_plus_example, PID: 10930
E/AndroidRuntime(10930): java.lang.RuntimeException: Error receiving broadcast Intent { act=android.bluetooth.adapter.action.STATE_CHANGED flg=0x4000010 (has extras) } in com.boskokg.flutter_blue_plus.FlutterBluePlusPlugin$1@117ad30f
E/AndroidRuntime(10930):        at android.app.LoadedApk$ReceiverDispatcher$Args.run(LoadedApk.java:876)
E/AndroidRuntime(10930):        at android.os.Handler.handleCallback(Handler.java:739)
E/AndroidRuntime(10930):        at android.os.Handler.dispatchMessage(Handler.java:95)
E/AndroidRuntime(10930):        at android.os.Looper.loop(Looper.java:135)
E/AndroidRuntime(10930):        at android.app.ActivityThread.main(ActivityThread.java:5254)
E/AndroidRuntime(10930):        at java.lang.reflect.Method.invoke(Native Method)
E/AndroidRuntime(10930):        at java.lang.reflect.Method.invoke(Method.java:372)
E/AndroidRuntime(10930):        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:903)
E/AndroidRuntime(10930):        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:698)
E/AndroidRuntime(10930): Caused by: java.lang.IllegalArgumentException: Unsupported value: 'com.boskokg.flutter_blue_plus.Protos$BluetoothState$Builder@39fcf8a6' of type 'class com.boskokg.flutter_blue_plus.Protos$BluetoothState$Builder'
E/AndroidRuntime(10930):        at io.flutter.plugin.common.StandardMessageCodec.writeValue(StandardMessageCodec.java:297)
E/AndroidRuntime(10930):        at io.flutter.plugin.common.StandardMethodCodec.encodeSuccessEnvelope(StandardMethodCodec.java:61)
E/AndroidRuntime(10930):        at io.flutter.plugin.common.EventChannel$IncomingStreamRequestHandler$EventSinkImplementation.success(EventChannel.java:251)
E/AndroidRuntime(10930):        at com.boskokg.flutter_blue_plus.FlutterBluePlusPlugin$1.onReceive(FlutterBluePlusPlugin.java:1214)
E/AndroidRuntime(10930):        at android.app.LoadedApk$ReceiverDispatcher$Args.run(LoadedApk.java:866)
E/AndroidRuntime(10930):        ... 8 more
I/Process (10930): Sending signal. PID: 10930 SIG: 9
Lost connection to device.
```

chipweinberger, 2023-07-11:
This should be fixed on master branch. We've removed protobuffers.

anwin-kp, 2023-07-13:
is this fixed?

chipweinberger, 2023-07-13:
you tell me!

chipweinberger, 2023-07-14:
try version 1.7.1

anwin-kp, 2023-07-18:
it is fixed in the new update 1.7.5 
can close this thread

closed 2023-07-18 by chipweinberger