chipweinberger/flutter_blue_plus #367: Mobile Receipt Printer not found (Bixolon SPP-R200III)

leutbounpaseuth, 2023-07-12:
HI,

I use this printer and I'm able to pair the device properly on Android device.

Trying to discover the bluetooth mobile printer but it is not listed in the scan results.

Any idea ?

chipweinberger, 2023-07-14:
please try version 1.7.0

leutbounpaseuth, 2023-07-17:
Updated with 1.7.3 but I have compile problem now.

https://github.com/boskokg/flutter_blue_plus/issues/369#issuecomment-1637532089

leutbounpaseuth, 2023-07-17:
Found the problem, seems that blue_print_pos package uses your old package.
https://github.com/boskokg/flutter_blue_plus/issues/369#issuecomment-1637709805

But my printer is still not listed :(

chipweinberger, 2023-07-25:
The issue is likely that your Printer is not discoverable. Are other BLE scanner apps from the App Store finding it?

closed 2023-07-25 by chipweinberger