chipweinberger/flutter_blue_plus #368: Turn on / off Bluetooth App Crash

cedar-yxs, 2023-07-13:
There are some error log

```
java.lang.RuntimeException: Error receiving broadcast Intent { act=android.bluetooth.adapter.action.STATE_CHANGED flg=0x1000010 (has extras) } in com.boskokg.flutter_blue_plus.FlutterBluePlusPlugin$1@38753a
                                                                                                    	at android.app.LoadedApk$ReceiverDispatcher$Args.lambda$getRunnable$0$LoadedApk$ReceiverDispatcher$Args(LoadedApk.java:1590)
                                                                                                    	at android.app.-$$Lambda$LoadedApk$ReceiverDispatcher$Args$_BumDX2UKsnxLVrE6UJsJZkotuA.run(Unknown Source:2)
                                                                                                    	at android.os.Handler.handleCallback(Handler.java:938)
                                                                                                    	at android.os.Handler.dispatchMessage(Handler.java:99)
                                                                                                    	at android.os.Looper.loop(Looper.java:236)
                                                                                                    	at android.app.ActivityThread.main(ActivityThread.java:8057)
                                                                                                    	at java.lang.reflect.Method.invoke(Native Method)
                                                                                                    	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:620)
                                                                                                    	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1011)
                                                                                                    Caused by: java.lang.IllegalArgumentException: Unsupported value: 'com.boskokg.flutter_blue_plus.Protos$BluetoothState$Builder@2f84f23' of type 'class com.boskokg.flutter_blue_plus.Protos$BluetoothState$Builder'
                                                                                                    	at io.flutter.plugin.common.StandardMessageCodec.writeValue(StandardMessageCodec.java:297)
                                                                                                    	at io.flutter.plugin.common.StandardMethodCodec.encodeSuccessEnvelope(StandardMethodCodec.java:61)
                                                                                                    	at io.flutter.plugin.common.EventChannel$IncomingStreamRequestHandler$EventSinkImplementation.success(EventChannel.java:251)
                                                                                                    	at com.boskokg.flutter_blue_plus.FlutterBluePlusPlugin$1.onReceive(FlutterBluePlusPlugin.java:1225)
                                                                                                    	at android.app.LoadedApk$ReceiverDispatcher$Args.lambda$getRunnable$0$LoadedApk$ReceiverDispatcher$Args(LoadedApk.java:1580)
                                                                                                    	at android.app.-$$Lambda$LoadedApk$ReceiverDispatcher$Args$_BumDX2UKsnxLVrE6UJsJZkotuA.run(Unknown Source:2) 
                                                                                                    	at android.os.Handler.handleCallback(Handler.java:938) 
                                                                                                    	at android.os.Handler.dispatchMessage(Handler.java:99) 
                                                                                                    	at android.os.Looper.loop(Looper.java:236) 
                                                                                                    	at android.app.ActivityThread.main(ActivityThread.java:8057) 
                                                                                                    	at java.lang.reflect.Method.invoke(Native Method) 
                                                                                                    	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:620) 
                                                                                                    	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1011) 
```

cedar-yxs, 2023-07-13:
The same problem with example app

Xu-158, 2023-07-14:
I encountered the same problem

chipweinberger, 2023-07-14:
please try version 1.7.0

closed 2023-07-18 by chipweinberger

kalismeras61, 2023-07-20:
@chipweinberger this should be opened it is crashing on IOS... 

[FBP-iOS] handleMethodCall: setNotification
*** Terminating app due to uncaught exception of class 'FlutterError'
libc++abi: terminating due to uncaught exception of type FlutterError
* thread #1, queue = 'com.apple.main-thread', stop reason = signal SIGABRT
    frame #0: 0x00000001df58d158 libsystem_kernel.dylib`__pthread_kill + 8
libsystem_kernel.dylib`:
->  0x1df58d158 <+8>:  b.lo   0x1df58d174               ; <+36>
    0x1df58d15c <+12>: stp    x29, x30, [sp, #-0x10]!
    0x1df58d160 <+16>: mov    x29, sp
    0x1df58d164 <+20>: bl     0x1df588b30               ; cerror_nocancel
Target 0: (Runner) stopped.
Lost connection to device.