chipweinberger/flutter_blue_plus #369: flutter blue_plus causes build excption on Android

talitshak, 2023-07-14:
After Pub Get, compiling and running the app fails with exception:
FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':flutter_blue_plus:generateDebugProto'.
> protoc: stdout: . stderr: C:\Users\...\AppData\Local\Pub\Cache\hosted\pub.dev\flutter_blue_plus-1.5.2\protos: warning: directory does not exist.
  Could not make proto path relative: C:\Users\...\AppData\Local\Pub\Cache\hosted\pub.dev\flutter_blue_plus-1.5.2\protos\flutterblueplus.proto: No such file or directory

chipweinberger, 2023-07-14:
please try version 1.7.0

leutbounpaseuth, 2023-07-17:
Hi,

I have the same error with version 1.7.3

```
Launching lib/main.dart on SM A025G in debug mode...
: Error: Error when reading '../../../../.pub-cache/hosted/pub.dev/flutter_blue_plus-1.7.3/lib/gen/flutterblueplus.pb.dart': No such file or directory
import 'package:flutter_blue_plus/gen/flutterblueplus.pb.dart' as proto;
       ^

: Error: Undefined name 'BluetoothDevice_Type'.
            type: proto.BluetoothDevice_Type.valueOf(selectedDevice?.type ?? 0),
                        ^^^^^^^^^^^^^^^^^^^^
: Error: Method not found: 'BluetoothDevice'.
          proto.BluetoothDevice(
                ^^^^^^^^^^^^^^^
Target kernel_snapshot failed: Exception

FAILURE: Build failed with an exception.

* Where:
Script '/Users/leutbounpaseuth/flutter/packages/flutter_tools/gradle/flutter.gradle' line: 1201

* What went wrong:
Execution failed for task ':app:compileFlutterBuildDebug'.
> Process 'command '/Users/leutbounpaseuth/flutter/bin/flutter'' finished with non-zero exit value 1

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 3s
Exception: Gradle task assembleDebug failed with exit code 1
Exited

```

chipweinberger, 2023-07-17:
Are you sure you are on 1.7.3? `flutterblueplus.pb.dart` is not found.

leutbounpaseuth, 2023-07-17:
Yes I am, here is my pubspec.yaml

```
environment:
  sdk: '>=3.0.5 <4.0.0'

dependencies:
  flutter:
    sdk: flutter

  # The following adds the Cupertino Icons font to your application.
  # Use with the CupertinoIcons class for iOS style icons.
  cupertino_icons: ^1.0.2
  permission_handler: ^10.4.2
  flutter_blue_plus: ^1.7.3
  blue_thermal_printer: ^1.2.3
  esc_pos_utils: ^1.1.0
  flutter_simple_bluetooth_printer: ^0.1.4
  bluetooth_advanced: ^0.0.83
  blue_print_pos: ^0.0.14
```

leutbounpaseuth, 2023-07-17:
Found the problem, seems that blue_print_pos package uses your old package.
I remove it and it compile now.

closed 2023-07-17 by chipweinberger