chipweinberger/flutter_blue_plus #370: PlatformException - Version 1.7.1

cedar-yxs, 2023-07-15:
When I run example app, then I click search button, I had an error
Logs as below:
```
E/flutter ( 5612): [ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: PlatformException(startScan, FlutterBluePlus requires android.permission.ACCESS_FINE_LOCATION permission, null, null)
E/flutter ( 5612): #0      StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:653:7)
E/flutter ( 5612): #1      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:315:18)
E/flutter ( 5612): <asynchronous suspension>
E/flutter ( 5612): #2      FlutterBluePlus.scan (package:flutter_blue_plus/src/flutter_blue_plus.dart:189:7)
E/flutter ( 5612): <asynchronous suspension>
```

I run it with flutter 3.7.12 and 3.10.6. I had the same error

The action button is 'TURN OFF'. I hit the button and nothing happened

chipweinberger, 2023-07-15:
perhaps we/you need to update the permissions here

https://github.com/boskokg/flutter_blue_plus/blob/master/example/android/app/src/main/AndroidManifest.xml

huayun502, 2023-07-16:
me too,I update the permissions,but not working - Android 13

chipweinberger, 2023-07-16:
it may be due to this

https://github.com/boskokg/flutter_blue_plus/commit/c0b1de652b141a5962477b48d6cd6de757fa36f0

perhaps you can add breakpoints and debug if the permission is being requested

huayun502, 2023-07-16:
E/flutter (15423): [ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: PlatformException(startScan, FlutterBluePlus requires android.permission.ACCESS_FINE_LOCATION permission, null, null)
E/flutter (15423): #0      StandardMethodCodec.decodeEnvelope
message_codecs.dart:652
E/flutter (15423): #1      MethodChannel._invokeMethod
platform_channel.dart:310
E/flutter (15423): <asynchronous suspension>
E/flutter (15423): #2      FlutterBluePlus.scan
flutter_blue_plus.dart:189
E/flutter (15423): <asynchronous suspension>

chipweinberger, 2023-07-16:
you can also try reverting this commit: https://github.com/boskokg/flutter_blue_plus/commit/8a35c6a0b6aca2fe23e5cd3730b7d3cf7903437c

chipweinberger, 2023-07-16:
or setting this permission: <uses-permission android:name="android.permission.BLUETOOTH_SCAN" android:usesPermissionFlags="neverForLocation" />

a371057600, 2023-07-17:
> you can also try reverting this commit: [8a35c6a](https://github.com/boskokg/flutter_blue_plus/commit/8a35c6a0b6aca2fe23e5cd3730b7d3cf7903437c)
> 
> or perhaps setting this permission: `<uses-permission android:name="android.permission.BLUETOOTH_SCAN" android:usesPermissionFlags="neverForLocation" />`
> 
> to be honest, I'm not sure what the fix is.
There's still no way to fix it

a371057600, 2023-07-17:
update the permissions,but not working  in Android 13

FritzMatthaeus, 2023-07-17:
I have made a pull request to fix this

FritzMatthaeus, 2023-07-18:
I just double checked it. Version 1.7.3 throws a PlatformException on Android 13 without ACCESS_FINE_LOCATION

```
 Unhandled Exception: PlatformException(startScan, FlutterBluePlus requires android.permission.ACCESS_FINE_LOCATION permission, null, null)
```

I have been using this package for more than a year and i checked my project history. I had issues with Android 13 about 6 month ago and could fix it by adding
```
<uses-permission android:name="android.permission.BLUETOOTH_SCAN"
                     android:usesPermissionFlags="neverForLocation" />
<uses-permission android:name="android.permission.NEARBY_DEVICES"/>
```
I used flutter_blue_plus Version 1.4.0 at that time and did not need the ACCESS_FINE_LOCATION permission. So this exception might be a regression.

FritzMatthaeus, 2023-07-18:
I am no Java expert, but it seems to me that the changes in "startScan" in the onMethodCall method in dynamically handling the permissions for >= Android Version 31 might be at the origin for this issue.

FritzMatthaeus, 2023-07-18:
I have just checked Version 1.4.0 and there we already had the ACCESS_FINE_LOCATION but without maxSDK Restriction. It is commented as legacy, but obviously it kept the package working on Android 13.

```
    <!-- New Bluetooth permissions in Android 12
           https://developer.android.com/about/versions/12/features/bluetooth-permissions
      -->
    <uses-permission android:name="android.permission.BLUETOOTH_SCAN" />
    <uses-permission android:name="android.permission.BLUETOOTH_CONNECT" />

    <!-- Request legacy Bluetooth permissions on older devices. -->
    <uses-permission android:name="android.permission.BLUETOOTH" />
    <uses-permission android:name="android.permission.BLUETOOTH_ADMIN" />
    <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />
```

Seems to me as if ACCESS_FINE_LOCATION always has been required.

FritzMatthaeus, 2023-07-18:
@chipweinberger: You were right on the permissions, but the package always requested this permission as the 
androidUsesFineLocation flag defaulted to true in the scan and startScan method.

I've made a pull request to fix this and updated the readMe and commentation

chipweinberger, 2023-07-18:
thanks for the high quality PR!

chipweinberger, 2023-07-18:
It looks like there is a discrepancy here? Should `android/src/main/AndroidManifest.xml` have `<uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" android:maxSdkVersion="30"/>` as well?

[image]

FritzMatthaeus, 2023-07-18:
Yeah, i did miss that. Sorry. Though it is not actually needed as the example worked fine on my device this way. But for better unterstanding, i would add it this way: 
```
<!-- remove the maxSdkVersion restriction to access location information via Bluetooth on Android Version >=31 -->
<uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" android:maxSdkVersion="30"/>
```

chipweinberger, 2023-07-18:
would be great if you could update your branch. easier than me doing it.  no worries either way

FritzMatthaeus, 2023-07-18:
No problem, i added it to my pull request

chipweinberger, 2023-07-20:
Please try 1.7.7+. reopen a new issue if you still have problems.

closed 2023-07-20 by chipweinberger