chipweinberger/flutter_blue_plus #374: characteristic.write throws exception

mdikcinar, 2023-07-16:
Hi, i got a problem when writing data. This example code thows exception even the result is success.

```dart
  Future<bool> writeWithResponse({required BluetoothCharacteristic characteristic, required List<int> bytes}) async {
    try {
      await characteristic.write(bytes);
      return true;
    } catch (e) {
      return false;
    }
  } 
```

Here is the excepiton

```
_Exception (Exception: type 'Null' is not a subtype of type 'int' in type cast #0      new BmWriteCharacteristicRequest.fromMap (package:flutter_blue_plus/src/bluetooth_msgs.dart:490:56)
#1      new BmWriteCharacteristicResponse.fromMap (package:flutter_blue_plus/src/bluetooth_msgs.dart:513:45)
#2      BluetoothCharacteristic.write.<anonymous closure>.<anonymous closure> (package:flutter_blue_plus/src/bluetooth_characteristic.dart:161:60)
#3      _MapStream._handleData (dart:async/stream_pipe.dart:213:31)
#4      _ForwardingStreamSubscription._handleData (dart:async/stream_pipe.dart:153:13)
#5      _RootZone.runUnaryGuarded (dart:async/zone.dart:1594:10)
#6      _BufferingStreamSubscription._sendData (dart:async/stream_impl.dart:339:11)
#7      _BufferingStreamSubscription._add (dart:async/stream_impl.dart:271:7)
#8      _ForwardingStreamSubscription._add (dart:async/stream_pipe.dart:123:11)
#9      _MapStream._handleData (dart:async/stream_pipe.dart:218:10)
#10     _ForwardingStreamSubscription._handleData (dart:async/stream_pipe.dart:153:13)
#11     _RootZone.runUnaryGuarded (dart:async/zone.dart:1594:10)
#12     _BufferingStreamSubscription._sendData (dart:async/stream_impl.dart:339:11)
#13     _BufferingStreamSubscription._add (dart:async/stream_impl.dart:271:7)
#14     _ForwardingStreamSubscription._add (dart:async/stream_pipe.dart:123:11)
#15     _WhereStream._handleData (dart:async/stream_pipe.dart:195:12)
#16     _ForwardingStreamSubscription._handleData (dart:async/stream_pipe.dart:153:13)
#17     _RootZone.runUnaryGuarded (dart:async/zone.dart:1594:10)
#18     _BufferingStreamSubscription._sendData (dart:async/stream_impl.dart:339:11)
#19     _DelayedData.perform (dart:async/stream_impl.dart:515:14)
#20     _PendingEvents.handleNext (dart:async/stream_impl.dart:620:11)
#21     _PendingEvents.schedule.<anonymous closure> (dart:async/stream_impl.dart:591:7)
#22     _microtaskLoop (dart:async/schedule_microtask.dart:40:21)
#23     _startMicrotaskLoop (dart:async/schedule_microtask.dart:49:5)
)
```

mdikcinar, 2023-07-16:
pr
https://github.com/boskokg/flutter_blue_plus/pull/375

chipweinberger, 2023-07-16:
ios or android?

mdikcinar, 2023-07-16:
Android

chipweinberger, 2023-07-16:
fixed: f561cb196f8ea2b967325f0cca4524f15ef6948a

closed 2023-07-18 by chipweinberger