chipweinberger/flutter_blue_plus #376: ACCESS_FINE_LOCATION can not work in 1.7.3

a371057600, 2023-07-17:
```
<manifest xmlns:android="http://schemas.android.com/apk/res/android"
    package="com.fit.monster">
    <uses-feature
        android:name="android.hardware.bluetooth_le"
        android:required="true" /> 
   
   <application
        android:label="@string/app_name"
        android:icon="@mipmap/tergasy"
        android:name="${applicationName}"
        android:usesCleartextTraffic="true">
       
        <activity
            android:name=".MainActivity"
            android:exported="true"
            android:launchMode="singleTop"
            android:theme="@style/LaunchTheme"
            android:configChanges="orientation|keyboardHidden|keyboard|screenSize|smallestScreenSize|locale|layoutDirection|fontScale|screenLayout|density|uiMode"
            android:hardwareAccelerated="true"
            android:windowSoftInputMode="adjustResize">
            <!-- Specifies an Android theme to apply to this Activity as soon as
                 the Android process has started. This theme is visible to the user
                 while the Flutter UI initializes. After that, this theme continues
                 to determine the Window background behind the Flutter UI. -->
            <meta-data
              android:name="io.flutter.embedding.android.NormalTheme"
              android:resource="@style/NormalTheme"
              />
                <!-- <meta-data
              android:name="io.flutter.embedding.android.SplashScreenDrawable"
              android:resource="@drawable/launch_background"
              /> -->
            <intent-filter>
                <action android:name="android.intent.action.MAIN"/>
                <category android:name="android.intent.category.LAUNCHER"/>
            </intent-filter>
        </activity>
        
        <!-- Don't delete the meta-data below.
             This is used by the Flutter tool to generate GeneratedPluginRegistrant.java -->
        <meta-data
            android:name="flutterEmbedding"
            android:value="2" />
    </application>
        <uses-permission android:name="android.permission.CAMERA"/>
    <uses-permission android:name="android.permission.WRITE_EXTERNAL_STORAGE"/>
    <uses-permission android:name="android.permission.READ_EXTERNAL_STORAGE" />
    <uses-permission android:name="android.permission.REORDER_TASKS" />
    <uses-permission android:name="android.permission.WAKE_LOCK" />
    <uses-permission android:name="android.permission.INTERNET"/>
    <uses-permission android:name="android.permission.BLUETOOTH_SCAN" android:usesPermissionFlags="neverForLocation" />
     <uses-permission android:name="android.permission.BLUETOOTH"  android:maxSdkVersion="30"/>  
    <uses-permission android:name="android.permission.BLUETOOTH_CONNECT" />
	 <uses-permission android:name="android.permission.BLUETOOTH_ADMIN"  android:maxSdkVersion="30"/>  
	 <uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION"  android:maxSdkVersion="28"/> 
   <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION"  android:maxSdkVersion="30"/>
</manifest>
```
```
Error starting scan.
E/flutter (16616): [ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: PlatformException(startScan, FlutterBluePlus requires android.permission.ACCESS_FINE_LOCATION permission, null, null)
E/flutter (16616): #0      StandardMethodCodec.decodeEnvelope
message_codecs.dart:652
E/flutter (16616): #1      MethodChannel._invokeMethod
platform_channel.dart:310
E/flutter (16616): <asynchronous suspension>
E/flutter (16616): #2      FlutterBluePlus.scan
flutter_blue_plus.dart:189
```
i add all uses-permission  to the app,but i still not work;i can open the permission

AymenXD, 2023-07-17:
same issue :/

chipweinberger, 2023-07-18:
should be fixed in 1.7.4

closed 2023-07-18 by a371057600