chipweinberger/flutter_blue_plus #378: StreamBuilder<BluetoothState> is not updated by BT on/off changes

talitshak, 2023-07-17:
flutter_blue_plus 1.7.4
 (tested on android)
 
 Followed the example code, but getting no changes:
 
 return Scaffold(
        appBar: AppBar(
          title: const Text('Bluetooth'),
        ),
        body: StreamBuilder<BluetoothState>(
          stream: FlutterBluePlus.instance.state,
          initialData: BluetoothState.unknown,
          builder: (context, snapshot) {
            final state = snapshot.data;
            if (state == BluetoothState.on) {
              return BluetothOnScreen()
              .........
              
              
              added needed permission:
  
    <uses-feature android:required="true" android:name="android.hardware.bluetooth_le"/>
    <uses-permission android:name="android.permission.BLUETOOTH" android:maxSdkVersion="30"/>
    <uses-permission android:name="android.permission.BLUETOOTH_ADMIN" android:maxSdkVersion="30"/>
    <uses-permission android:name="android.permission.BLUETOOTH_SCAN"/>
    <uses-permission android:name="android.permission.BLUETOOTH_ADVERTISE"/>
    <uses-permission android:name="android.permission.BLUETOOTH_CONNECT"/>
    <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />
              
              what might be the reason?
              
              thanks

chipweinberger, 2023-07-17:
Just pushed a fix. try master branch. 709ab810dc70a53a1a2405df7c38530c22f48a10

talitshak, 2023-07-18:
Thanks, now it does work but device.connect does not for some reason

chipweinberger, 2023-07-18:
please open a new issue

closed 2023-07-18 by chipweinberger