chipweinberger/flutter_blue_plus #380: import 'package:flutter_blue_plus/gen/flutterblueplus.pbserver.dart' as protos;

Jbz797, 2023-07-18:
Is there any documentation for migrating from version 1.5.2 to 1.7.0?

This file has been deleted:
`import 'package:flutter_blue_plus/gen/flutterblueplus.pbserver.dart' as protos;`

As a result, I can no longer use this method:

  ```
BluetoothCharacteristic _phoneToVmi(DeviceIdentifier remoteId) {
    final protos.BluetoothCharacteristic phoneToVmiCharac = protos.BluetoothCharacteristic.create();
    phoneToVmiCharac.remoteId = remoteId.toString();
    phoneToVmiCharac.serviceUuid = 'XXXXX';
    phoneToVmiCharac.uuid = 'XXXXX';
    return BluetoothCharacteristic.fromProto(phoneToVmiCharac);
  }
```

mohamedsl22, 2023-07-18:
I also was using 
import 'package:flutter_blue_plus/gen/flutterblueplus.pb.dart' as ProtoBluetoothDevice;
and now it’s deleted !!! Please fix

chipweinberger, 2023-07-18:
Use `BmBluetoothCharacteristic` instead.

import 'package:flutter_blue_plus/flutter_blue_plus.dart';

BluetoothCharacteristic _phoneToVmi(DeviceIdentifier remoteId) {
  final BmBluetoothCharacteristic phoneToVmiCharac = BmBluetoothCharacteristic(
      "remoteId": remoteId.toString(),
      "serviceUuid": 'XXXXX',
      "uuid" = 'XXXXX',
        ....
  return BluetoothCharacteristic.fromProto(phoneToVmiCharac);
}

Jbz797, 2023-07-19:
> @Jbz797 This is fascinating! It looks like you are augmenting a characteristic to your own liking, essentially acting as a Bluetooth peripheral? Does that work? I was looking at plugins ([TrackMyIndoorWorkout/TrackMyIndoorWorkout#389](https://github.com/TrackMyIndoorWorkout/TrackMyIndoorWorkout/issues/389)), I tried three of them but none of them worked.

I didn't think it was so extraordinary !
Yes, it's been working for a long time now. I'm forced to work this way because I connect to a particular Bluetooth device and I've never been able to use the standard methods. 

> Use `BmBluetoothCharacteristic` instead.
>
> ```
> import 'package:flutter_blue_plus/flutter_blue_plus.dart';
>
> BluetoothCharacteristic _phoneToVmi(DeviceIdentifier remoteId) {
>   final [...]

Its work, thanks a lot.

closed 2023-07-19 by chipweinberger