chipweinberger/flutter_blue_plus #381: 【Android】"characteristic.value.listen((value)" does't have any callback.

wanxiao, 2023-07-18:
c1.setNotify(true);
c1.value.listen((value){
print("callback");
});

c2.write(data);

On Android , it does't have any callback.

**BUT not every characteristic has the same problem**.

I have two characteristics for receiving notifications, one gets notifications, one doesn't.

This problem occurs in v1.7.x and does not occur in v1.6.1, which was forked earlier.
However on v1.6.1, Android 12 cannot return scan results.
Is there no release for 1.6.x?

chipweinberger, 2023-07-18:
please open a PR if you fix it, and we'll update 1.7.x

shawsa-artc, 2023-07-19:
I was facing the same problem, managed to solved it.
It occurs when you are trying to subscribing to multiple characteristics at same time.
You need to surround each with a try catch

          try {
            await c.setNotifyValue(true);
            // set listener for changes
            c.value.listen((value) {
              //print('2a37: ${value.toString()}');
              getHRM = value[1];
            });
          } catch (error) {
            // Handle the error
            //print('Error occurred while setting notify value: $error');
          }

MrCsabaToth, 2023-07-19:
"does't have any callback" means that the listener callback function is not called, so the second listener is mute?

chipweinberger, 2023-07-20:
seems like same issue: https://github.com/boskokg/flutter_blue_plus/issues/329

chipweinberger, 2023-07-25:
Note, calling `write()` DOES NOT cause `lastValueStream`, `lastValue`, or `onValueReceived` to be updated. This is by design. 

Reads are treated very separately from writes in many BLE protocols, so we cannot treat a write as "setting" the read value.

chipweinberger, 2023-07-25:
please open a new if it still occurs on the latest releases.

closed 2023-07-25 by chipweinberger