chipweinberger/flutter_blue_plus #385: Unhandled Exception: PlatformException(write_characteristic_error, writeCharacteristic failed, null, null)

faiyaz-shaikh, 2023-07-19:
In iOS it's working fine and able to get data but in android getting below error.

```
E/flutter (19543): [ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: PlatformException(set_notification_error, error when writing the descriptor, null, null)
E/flutter (19543): #0      StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:652:7)
E/flutter (19543): #1      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:310:18)
E/flutter (19543): <asynchronous suspension>
E/flutter (19543): #2      BluetoothCharacteristic.setNotifyValue (package:flutter_blue_plus/src/bluetooth_characteristic.dart:208:9)
E/flutter (19543): <asynchronous suspension>
E/flutter (19543): 
E/flutter (19543): [ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: PlatformException(set_notification_error, error when writing the descriptor, null, null)
E/flutter (19543): #0      StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:652:7)
E/flutter (19543): #1      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:310:18)
E/flutter (19543): <asynchronous suspension>
E/flutter (19543): #2      BluetoothCharacteristic.setNotifyValue (package:flutter_blue_plus/src/bluetooth_characteristic.dart:208:9)
E/flutter (19543): <asynchronous suspension>
E/flutter (19543): 
E/flutter (19543): [ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: PlatformException(write_characteristic_error, writeCharacteristic failed, null, null)
E/flutter (19543): #0      StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:652:7)
E/flutter (19543): #1      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:310:18)
E/flutter (19543): <asynchronous suspension>
E/flutter (19543): #2      _Mutex.synchronized (package:flutter_blue_plus/src/utils.dart:314:13)
E/flutter (19543): <asynchronous suspension>
E/flutter (19543): #3      BluetoothCharacteristic.write (package:flutter_blue_plus/src/bluetooth_characteristic.dart:141:9)
E/flutter (19543): <asynchronous suspension>
E/flutter (19543): 
I/chatty  (19543): uid=10690(io.blocship.blinq) 1.ui identical 4 lines
E/flutter (19543): [ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: PlatformException(write_characteristic_error, writeCharacteristic failed, null, null)
E/flutter (19543): #0      StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:652:7)
E/flutter (19543): #1      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:310:18)
E/flutter (19543): <asynchronous suspension>
E/flutter (19543): #2      _Mutex.synchronized (package:flutter_blue_plus/src/utils.dart:314:13)
E/flutter (19543): <asynchronous suspension>
E/flutter (19543): #3      BluetoothCharacteristic.write (package:flutter_blue_plus/src/bluetooth_characteristic.dart:141:9)
E/flutter (19543): <asynchronous suspension>
E/flutter (19543): 
```

And we are not able to get data and it's stuck on onCharacteristicWrite. In iOS it's working perfect.

```
I/flutter (18139): Received Data!!!! [16, 48]
I/flutter (18139): Current State: States.waitForAckBeginHandshake1AsMaster
I/flutter (18139): Got ACK! - sending data [16, 1, 67, 84, 79, 48, 48, 49, 48, 48, 48, 49, 82, 82, 48, 49, 76, 48, 49, 16, 3, 134, 37]
D/FlutterBluePlugin(18139): [onCharacteristicWrite] uuid: 0000ffe1-0000-1000-8000-00805f9b34fb status: 0
```

chipweinberger, 2023-07-19:
there are valid reasons for writeCharacteristic to fail. 

you should catch the error and handle it.

MrCsabaToth, 2023-07-19:
Is this in a code which worked fine with pre 1.7.0 versions by any chance?

Ervin2018, 2023-07-20:
The same，my flutter version is 3.10.5，flutter_blue_plus: ^1.7.5

chipweinberger, 2023-07-20:
in 1.4.0, write_characteristic errors were often ignored (accidentally). 

in 1.5.0 we make sure to throw an exception when they fail.

shwetachauhan-simform, 2023-07-20:
@chipweinberger : Crash we can handle but we are not able to write data in characteristic in Android. Is there anything in package issue ?

chipweinberger, 2023-07-20:
no issues that I am aware of.

shwetachauhan-simform, 2023-07-20:
Ok, we will investigate from our side.

chipweinberger, 2023-07-20:
see some of these linked issues and see if the suggestions help^

chipweinberger, 2023-07-20:
try 1.7.8

shwetachauhan-simform, 2023-07-20:
Exception: PlatformException(write_characteristic_error, The WRITE_NO_RESPONSE property is not supported by this BLE characteristic, null, null) #0      StandardMethodCodec.decodeEnvelope

Now, getting this error

shwetachauhan-simform, 2023-07-20:
> Exception: PlatformException(write_characteristic_error, The WRITE_NO_RESPONSE property is not supported by this BLE characteristic, null, null) #0 StandardMethodCodec.decodeEnvelope
> 
> Now, getting this error

After setting 'withoutResponse = false' Getting below error:

```
[3:55 pm] Faiyaz Shaikh

D/FlutterBluePlugin(21849): [onCharacteristicWrite] uuid: 0000ffe1-0000-1000-8000-00805f9b34fb status: 3

I/flutter (21849): ----------------FIREBASE CRASHLYTICS----------------

I/flutter (21849): Exception: Exception: Failed to write the characteristic #0      BluetoothCharacteristic.write.<anonymous closure> (package:flutter_blue_plus/src/bluetooth_characteristic.dart:168:11)

I/flutter (21849): <asynchronous suspension>

I/flutter (21849): #1      _Mutex.synchronized (package:flutter_blue_plus/src/utils.dart:311:7)

I/flutter (21849): <asynchronous suspension>

I/flutter (21849): #2      BluetoothCharacteristic.write (package:flutter_blue_plus/src/bluetooth_characteristic.dart:138:5)

I/flutter (21849): <asynchronous suspension>

I/flutter (21849): #3      _DexScreenState._onWriteData (package:chq_flutter/dex/device_screen.dart:302:5)

I/flutter (21849): <asynchronous suspension>

I/flutter (21849): #4      _DexScreenState.getServices (package:chq_flutter/dex/device_screen.dart:84:5)

I/flutter (21849): <asynchronous suspension>

I/flutter (21849): #5      _DexScreenState.startScan.<anonymous closure> (package:chq_flutter/dex/device_screen.dart:60:11)

I/flutter (21849): <asynchronous suspension>

I/flutter (21849): #0      BluetoothCharacteristic.write.<anonymous closure> (package:flutter_blue_plus/src/bluetooth_characteristic.dart:177:7)

I/flutter (21849): #1      _RootZone.runBinary (dart:async/zone.dart:1666:54)

I/flutter (21849): #2      _FutureListener.handleError (dart:async/future_impl.dart:162:22)

I/flutter (21849): #3      Future._propagateToListeners.handleError (dart:async/future_impl.dart:796:47)

I/flutter (21849): #4      Future._propagateToListeners (dart:async/future_impl.dart:817:13)

I/flutter (21849): #5      Future._completeError (dart:async/future_impl.dart:592:5)

I/flutter (21849): #6      Future._asyncCompleteError.<anonymous closure> (dart:async/future_impl.dart:683:7)

I/flutter (21849): #7      _microtaskLoop (dart:async/schedule_microtask.dart:40:21)

I/flutter (21849): #8      _startMicrotaskLoop (dart:async/schedule_microtask.dart:49:5)

```

chipweinberger, 2023-07-20:
Great! at least you know why now!

Eashubh128, 2023-07-20:
> Exception: PlatformException(write_characteristic_error, The WRITE_NO_RESPONSE property is not supported by this BLE characteristic, null, null) #0 StandardMethodCodec.decodeEnvelope
> 
> Now, getting this error

> Exception: PlatformException(write_characteristic_error, The WRITE_NO_RESPONSE property is not supported by this BLE characteristic, null, null) #0 StandardMethodCodec.decodeEnvelope
> 
> Now, getting this error

Hey , @shwetachauhan-simform stuck with the same issue ,

faiyaz-shaikh, 2023-07-20:
After setting true getting this error in Android. In iOS it's working.

Exception: PlatformException(write_characteristic_error, The WRITE_NO_RESPONSE property is not supported by this BLE characteristic, null, null) #0 StandardMethodCodec.decodeEnvelope

Eashubh128, 2023-07-20:
> After setting true getting this error in Android. In iOS it's working.
> 
> Exception: PlatformException(write_characteristic_error, The WRITE_NO_RESPONSE property is not supported by this BLE characteristic, null, null) #0 StandardMethodCodec.decodeEnvelope

Setting the write with response property wont help , because i guess it depends upon your hardware too , example the ble device you are trying to communicate with should also support write with response then only it will work

Eashubh128, 2023-07-20:
@shwetachauhan-simform , @faiyaz-shaikh  
Dont know about you guys but I encountered another very serious issue 
when i get the connected devices say in a list using connectedDevices() method , and then try to discoverServices() of the device present in the list it also throws an exception . Please check this also you might encounter the same .
[](https://github.com/boskokg/flutter_blue_plus/issues/390)

chipweinberger, 2023-07-20:
try 1.8.1. 

this was the bug: https://github.com/boskokg/flutter_blue_plus/commit/f1bed64f3993d744f9d288ddec18fa9c5941c0ce

chipweinberger, 2023-07-20:
Yes

Eashubh128, 2023-07-20:
@chipweinberger Sorry to address it here , but can you please look at  #390
this error also came with the recent versions

shwetachauhan-simform, 2023-07-20:
> try 1.8.1.
> 
> this was the bug: [f1bed64](https://github.com/boskokg/flutter_blue_plus/commit/f1bed64f3993d744f9d288ddec18fa9c5941c0ce)

PlatformException(write_characteristic_error, The WRITE property is not supported by this BLE characteristic ) ? This issue fixed but not able to write data in android. It's working in iOS. 

Is is possible that the hardware device is not working with Android??

cyclone-pk, 2023-07-20:
> try 1.8.2.
> 
> 1.8.1 fixed that WRITE property issue 1.8.2 added more details for write errors

Exception: PlatformException(write_characteristic_error, gattServer.writeCharacteristic() returned false, null, null) I am getting this in 1.8.2. does it work for you?

chipweinberger, 2023-07-20:
improved status codes require sdk 33+

```
                    // Version 33 
                    if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.TIRAMISU) {

                        int rv = gattServer.writeCharacteristic(characteristic, hexToBytes(value), writeType);

                        if (rv != BluetoothStatusCodes.SUCCESS) {
                            String s = "gattServer.writeCharacteristic() returned " + rv + " : " + bluetoothStatusString(rv);
                            result.error("write_characteristic_error", s, null);
                            return;
                        }

                    }
```

shwetachauhan-simform, 2023-07-20:
> > try 1.8.2.
> > 1.8.1 fixed that WRITE property issue 1.8.2 added more details for write errors
> 
> Exception: PlatformException(write_characteristic_error, gattServer.writeCharacteristic() returned false, null, null) I am getting this in 1.8.2. does it work for you? working on iOS as expected

We are changing, let you know if it works or not

samirasalah, 2023-07-20:
[ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: PlatformException(androidException, java.lang.Exception: locateGatt: BluetoothDeviceCache is null, have you connected first?, java.lang.Exception: locateGatt: BluetoothDeviceCache is null, have you connected first?

Same for me on Android 10

chipweinberger, 2023-07-20:
@samirasalah , you'll have to debug that issue yourself. I am not sure the cause. See the troubleshooting guide in the readme.

shwetachauhan-simform, 2023-07-20:
@chipweinberger: Finally @faiyaz-shaikh found issue in our code. In Android it's because of request larger size.

So, we put below line for OS Android and it's working as expected. 
`await device.requestMtu(512);`

Thanks for all your help.

chipweinberger, 2023-07-20:
great, glad you found the problem. I added MTU checks in 1.8.3 to make this easier.

closed 2023-07-20 by chipweinberger

chipweinberger, 2023-07-28:
@shwetachauhan-simform, were you doing a writeWithoutResponse or a normal write (with response)?