chipweinberger/flutter_blue_plus #386: import 'package:flutter_blue_plus/gen/flutterblueplus.pb.dart' as ProtoBluetoothDevice

mohamedsl22, 2023-07-19:
Hello;
in version 1.4.0 I was using the way to reconnect to a saved device automatically, in this way:

```
import 'package:flutter_blue_plus/gen/flutterblueplus.pb.dart' as ProtoBluetoothDevice

var protoBt = ProtoBluetoothDevice.BluetoothDevice(
            remoteId: relevantCollar.collarScanId
            );
var device = BluetoothDevice.fromProto(protoBt);
```

and then I connect the device and it worked, but with version 1.7.x I can't import this 'package:flutter_blue_plus/gen/flutterblueplus.pb.dart' as ProtoBluetoothDevice to do it. 

I tried this way, but after I close the app and reopen it doesn't connect 

```
  final BmBluetoothDevice phone = BmBluetoothDevice(remoteId: relevantScanId, name: 'Relax 2E', type: BmBluetoothSpecEnum.classic);
  var device = BluetoothDevice.fromProto(phone);
```
and this is the error that i'am getting :
Cancelling connection for unused peripheral <CBPeripheral: 0x2835c4750, identifier = 6F06023D-DD5D-DA22-53EF-9B68DF15A95D, name = 2E, mtu = 0, state = connecting>, Did you forget to keep a reference to it?
[FBP-iOS] handleMethodCall: deviceState

please help :)))

chipweinberger, 2023-07-19:
my suggestion is to look at how the FBP ios connection code changed between 1.4.0 and now, and play around with it.

MrCsabaToth, 2023-07-19:
Ref #380

chipweinberger, 2023-07-22:
there should be enough code here to help you update your code.

closed 2023-07-22 by chipweinberger