chipweinberger/flutter_blue_plus #388: Bluetooth permission ask on start up in ios

divyaEncodework, 2023-07-20:
@boskokg 
when I start my application in ios Bluetooth permission is asked on startup but I need Bluetooth on only one screen how can I manage it?
if anyone has a solution please update me
thank you

chipweinberger, 2023-07-20:
just make sure not to call any FlutterBluePlus code until needed and it should work i think.

chipweinberger, 2023-07-20:
not sure. try looking at the device logs using a Macbook and opening Console.app.

divyaEncodework, 2023-07-20:
i cant find any Bluetooth services in logs and I am not using it in start up

chipweinberger, 2023-07-20:
perhaps you permissions plist has issues? try removing the bluetooth permission and see what happens.

divyaEncodework, 2023-07-20:
but I need Bluetooth permissions on another screen

chipweinberger, 2023-07-20:
I know. But temporarily remove the permission to help troubleshoot. Perhaps you need to specify something else.

divyaEncodework, 2023-07-20:
if I remove permission then my app crashes on my iPhone with the message that app is trying to access data without permission

chipweinberger, 2023-07-22:
let me know if you fix it.

This might happen when we instantiate the CBCentralManager.

Try searching the code and try commenting those lines out and seeing what happens.

chipweinberger, 2023-07-25:
did you figure it out?

divyaEncodework, 2023-07-25:
no,
precisely that's because of CBCentralManager.
but I don't know which file I have to change.

chipweinberger, 2023-07-25:
The CBCentralManager is only invoked if you interact with FlutterBluePlus.

If you see any logs like:

`2023-07-24 21:29:14.845 macos[72103:5766776] [FBP-iOS] handleMethodCall: XXXXXXXX`

Then the problem is in your code. You should not be calling any FlutterBluePlus functions at all.

divyaEncodework, 2023-07-25:
no, I am not using any flutterBluePlus functions and not receiving logs like this

chipweinberger, 2023-07-25:
Have you tried commenting out all of your flutterBluePlus code?

Try temporarily deleting all flutterBluePlus code for your app.

divyaEncodework, 2023-07-25:
I can't do this 
it's a live project
we need it for device configurations.

chipweinberger, 2023-07-25:
?  I am not asking for you to save the changes.

divyaEncodework, 2023-07-25:
I comment out all the code and run the application
Bluetooth permission dialogue still occurs.

divyaEncodework, 2023-07-25:
ok, I found that this dialogue is not because of your plugin.
thank you.

chipweinberger, 2023-07-25:
glad you figured it out!

closed 2023-07-25 by chipweinberger