chipweinberger/flutter_blue_plus #390: java.lang.Exception: no instance of BluetoothGatt, have you connected first?, java.lang.Exception: no instance of BluetoothGatt, have you connected first?

Eashubh128, 2023-07-20:
I am using the flutter_blue_plus version ^1.7.5 . And getting this error while trying to discover services.
This problem specifically occurs when I try this
`
_connDevices= await FlutterBluePlus.instance.connectedDevices();
if(connDevices.isnotEmpty)
{
    _services = _connDevices.elementAt(0).discoverServices();  ////This causes the error
}
`
If I i connect to a device and then discover the services it works fine , but getting connected devices using connectedDevices() and then discovering services causes the error.

[image]

chipweinberger, 2023-07-20:
I think you are mistaken. This can never work without first calling connect().

[image]

chipweinberger, 2023-07-20:
[image]

Eashubh128, 2023-07-21:
I am sorry , i assumed this would work , because i changed my dependency from flutter_blue to flutter_blue_plus. 
And this worked in flutter blue without any exception 

> I am using the flutter_blue_plus version ^1.7.5 . And getting this error while trying to discover services. This problem specifically occurs when I try this `_connDevices= await [...]

chipweinberger, 2023-07-22:
i'm going to close this issue.

If you think this can actually be solved, please try to implement it and open a PR.

closed 2023-07-22 by chipweinberger