chipweinberger/flutter_blue_plus #391: The READ property is not supported by this BLE characteristic

iresolveservices, 2023-07-20:
"We obtained the data from the BLE device, but when I clicked on the characteristic with the value 0x2A52, it threw an error."

D/FlutterBluePlugin(13128): [onReadRemoteRssi] rssi: -51 status: 0
I/flutter (13128): remoteId: F4:B8:98:45:0C:E8characteristicUuid: 00002a52-0000-1000-8000-00805f9b34fbserviceUuid: 00000000-0000-1000-1000-000000000000
E/flutter (13128): [ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: Exception: PlatformException(read_characteristic_error, [image], null, null) #0 StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:652:7)
E/flutter (13128): https://github.com/boskokg/flutter_blue_plus/pull/1 MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:310:18)
E/flutter (13128):
E/flutter (13128): https://github.com/boskokg/flutter_blue_plus/issues/2 BluetoothCharacteristic.read. (package:flutter_blue_plus/src/bluetooth_characteristic.dart:111:7)
E/flutter (13128):
E/flutter (13128): https://github.com/boskokg/flutter_blue_plus/issues/3 _Mutex.synchronized (package:flutter_blue_plus/src/utils.dart:311:7)
E/flutter (13128):
E/flutter (13128): https://github.com/boskokg/flutter_blue_plus/issues/4 BluetoothCharacteristic.read (package:flutter_blue_plus/src/bluetooth_characteristic.dart:85:5)
E/flutter (13128):
E/flutter (13128):
E/flutter (13128): #0 BluetoothCharacteristic.read. (package:flutter_blue_plus/src/bluetooth_characteristic.dart:124:7)
E/flutter (13128): https://github.com/boskokg/flutter_blue_plus/pull/1 _RootZone.runBinary (dart:async/zone.dart:1666:54)
E/flutter (13128): https://github.com/boskokg/flutter_blue_plus/issues/2 _FutureListener.handleError (dart:async/future_impl.dart:162:22)
E/flutter (13128): https://github.com/boskokg/flutter_blue_plus/issues/3 Future._propagateToListeners.handleError (dart:async/future_impl.dart:796:47)
E/flutter (13128): https://github.com/boskokg/flutter_blue_plus/issues/4 Future._propagateToListeners (dart:async/future_impl.dart:817:13)
E/flutter (13128): https://github.com/boskokg/flutter_blue_plus/issues/5 Future._completeError (dart:async/future_impl.dart:592:5)
E/flutter (13128): https://github.com/boskokg/flutter_blue_plus/issues/6 Future._asyncCompleteError. (dart:async/future_impl.dart:683:7)
E/flutter (13128): https://github.com/boskokg/flutter_blue_plus/issues/7 _microtaskLoop (dart:async/schedule_microtask.dart:40:21)
E/flutter (13128): https://github.com/boskokg/flutter_blue_plus/issues/8 _startMicrotaskLoop (dart:async/schedule_microtask.dart:49:5)
E/flutter (13128):

iresolveservices, 2023-07-20:
[image]

iresolveservices, 2023-07-20:
"We use the Accu-Chek Instant glucometer, and it supports my sugar app using Bluetooth to display readings in the same app. However, our scenario is that the READ property is not supported by this BLE characteristic. Can you tell me what is wrong with us?

If my glucometer device does not support BLE services, could you suggest which glucometer would properly work with Flutter Blue Pulse?"

iresolveservices, 2023-07-20:
> Update to 1.8.1 That will help

onReadRemoteRssi] rssi: -46 status: 0
I/flutter (29904): remoteId: F4:B8:98:45:0C:E8characteristicUuid: 00002a52-0000-1000-8000-00805f9b34fbserviceUuid: 00000000-0000-1000-1000-000000000000
E/flutter (29904): [ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: Exception: PlatformException(read_characteristic_error, The READ property is not supported by this BLE characteristic, null, null) #0      StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:652:7)
E/flutter (29904): #1      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:310:18)
E/flutter (29904): <asynchronous suspension>
E/flutter (29904): #2      BluetoothCharacteristic.read.<anonymous closure> (package:flutter_blue_plus/src/bluetooth_characteristic.dart:111:7)
E/flutter (29904): <asynchronous suspension>
E/flutter (29904): #3      _Mutex.synchronized (package:flutter_blue_plus/src/utils.dart:311:7)
E/flutter (29904): <asynchronous suspension>
E/flutter (29904): #4      BluetoothCharacteristic.read (package:flutter_blue_plus/src/bluetooth_characteristic.dart:85:5)
E/flutter (29904): <asynchronous suspension>
E/flutter (29904): 
E/flutter (29904): #0      BluetoothCharacteristic.read.<anonymous closure> (package:flutter_blue_plus/src/bluetooth_characteristic.dart:128:7)
E/flutter (29904): #1      _RootZone.runBinary (dart:async/zone.dart:1666:54)
E/flutter (29904): #2      _FutureListener.handleError (dart:async/future_impl.dart:162:22)
E/flutter (29904): #3      Future._propagateToListeners.handleError (dart:async/future_impl.dart:796:47)
E/flutter (29904): #4      Future._propagateToListeners (dart:async/future_impl.dart:817:13)
E/flutter (29904): #5      Future._completeError (dart:async/future_impl.dart:592:5)
E/flutter (29904): #6      Future._asyncCompleteError.<anonymous closure> (dart:async/future_impl.dart:683:7)
E/flutter (29904): #7      _microtaskLoop (dart:async/schedule_microtask.dart:40:21)
E/flutter (29904): #8      _startMicrotaskLoop (dart:async/schedule_microtask.dart:49:5)
E/flutter (29904):

iresolveservices, 2023-07-20:
getting same

chipweinberger, 2023-07-20:
This is expected.

00002a52-0000-1000-8000-00805f9b34fb is the Record Access Control Point (RACP) characteristic.

`The Record Access Control Point (RACP) characteristic is generally specified to support the "Write" and "Indicate" properties.`

closed 2023-07-20 by chipweinberger

iresolveservices, 2023-07-20:
could you suggest which glucometer would properly work with Flutter Blue Pulse?"

samirasalah, 2023-07-20:
`[ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: Exception: PlatformException(write_characteristic_error, The WRITE_NO_RESPONSE property is not supported by this BLE characteristic, null, null) #0      StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:652:7)
E/flutter (17994): #1      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:310:18)
E/flutter (17994): <asynchronous suspension>
E/flutter (17994): #2      _Mutex.synchronized (package:flutter_blue_plus/src/utils.dart:311:7)
E/flutter (17994): <asynchronous suspension>
E/flutter (17994): #3      BluetoothCharacteristic.write (package:flutter_blue_plus/src/bluetooth_characteristic.dart:142:5)
E/flutter (17994): <asynchronous suspension>`

Same issue, any solution pls ?