chipweinberger/flutter_blue_plus #397: Peripheral not getting cleaned up when Android app is backgrounded

sqcsabbey, 2023-07-20:
When I swipe my app away, a peripheral is staying connected, but my app is losing access to it; here's what I'm seeing:

When backgrounding, I see these log lines from this plugin:

> D/FlutterBluePlugin( 4445): onDetachedFromActivity
> D/FlutterBluePlugin( 4445): onDetachedFromEngine
> D/FlutterBluePlugin( 4445): teardown

Then I start seeing lots of log lines like:

> W/FlutterBluePlugin(18881): Tried to call OnCharacteristicChanged on closed channel

We were using version 1.3.7 of this plugin, but I upgraded to 1.8.2 and continue to see this behavior. 

What should happen here— should the app be responsible for disconnecting when it's being closed, or is this cleanup this plugin could do? Or something else altogether?

chipweinberger, 2023-07-20:
to be honest I'm not sure. If you figure it out, please report back.

chipweinberger, 2023-07-20:
[image]

chipweinberger, 2023-07-20:
should be fixed: a103824339d20377a02f41983d9e6212f8618b28

sqcsabbey, 2023-07-20:
Thank you, @chipweinberger!

chipweinberger, 2023-07-20:
let me know if it works

sqcsabbey, 2023-07-20:
> let me know if it works

@chipweinberger Yes, it worked for me; the peripheral disconnects a few seconds after the app is backgrounded. I get a couple of the 'on closed channel' messages before the cleanup happens, but I don't suspect that would have bad effects.

chipweinberger, 2023-07-20:
great!

closed 2023-07-20 by chipweinberger