chipweinberger/flutter_blue_plus #404: [Android] connectionState stream does not emit any BluetoothConnectionState

BreadKey, 2023-07-21:
Version is 1.8.7.

connectionState stream does not emit events when the Bluetooth device is connected or disconnected.

# Flutter Doctor
```
Doctor summary (to see all details, run flutter doctor -v):
[√] Flutter (Channel stable, 3.10.6, on Microsoft Windows [Version 10.0.22621.1992], locale ko-KR)
[√] Windows Version (Installed version of Windows is version 10 or higher)
[√] Android toolchain - develop for Android devices (Android SDK version 33.0.0)
[√] Chrome - develop for the web
[√] Visual Studio - develop for Windows (Visual Studio Community 2022 17.2.6)
[!] Android Studio (version 2021.2)
    X Unable to find bundled Java version.
[√] Android Studio (version 2022.2)
[√] VS Code (version 1.80.1)
[√] Connected device (4 available)
[√] Network resources
```

# Android Device Info
Model Name: Galaxy Tab A
Mdeol Number: SM-T515N
Android Version: 11

kalismeras61, 2023-07-21:
I can confirm this latest build is having issue 1.8.4 works fine.

victor-butler, 2023-07-21:
I confirm the same.

kalismeras61, 2023-07-21:
this PR https://github.com/boskokg/flutter_blue_plus/pull/399 fix this issue

chipweinberger, 2023-07-21:
merged. but that is adapterState, this issue is about connectionState

kalismeras61, 2023-07-21:
I think there is two issue one of them connect the other one is adapterState. 

adapter is fixed but connection is still same issue i guess.

I/flutter (22077): DevicesBloc : Error connecting to device TimeoutException after 0:00:20.000000: Failed to connect in time., #0      BluetoothDevice.connect.<anonymous closure>

chipweinberger, 2023-07-21:
fixed here: https://github.com/boskokg/flutter_blue_plus/commit/9df6d2283eb9f7ada7d210c1d6686acbfd36c0d8

try 1.8.8

kalismeras61, 2023-07-21:
@chipweinberger i am using latest 1.8.8, first time there is no issue with the connect but if i close and re-open BLE from the device it gives me timeout exception. It doesn't connect to device.

kalismeras61, 2023-07-21:
When switch off device ble i am getting below error also 

[log] PlatformException(set_notification_error, The device is not connected, null, null)
[log] #0      StandardMethodCodec.decodeEnvelope
message_codecs.dart:652
#1      MethodChannel._invokeMethod
platform_channel.dart:310
        <asynchronous suspension>
#2      BluetoothCharacteristic.setNotifyValue
bluetooth_characteristic.dart:213
        <asynchronous suspension>

MrCsabaToth, 2023-07-21:
> When switch off device ble i am getting below error also
> 
> [log] PlatformException(set_notification_error, The device is not connected, null, null) [log] #0 StandardMethodCodec.decodeEnvelope message_codecs.dart:652 #1 MethodChannel._invokeMethod platform_channel.dart:310 #2 BluetoothCharacteristic.setNotifyValue bluetooth_characteristic.dart:213

To understand better, when you say "if i close and re-open BLE from the device" means that you have a BLE gadget, and you manually turn off Bluetooth on that gadget? That'll cut off the connection, so actually the exception is not that surprising. Did this work differently with pre 1.7.0 before? Would you expect maybe some behavior when the FlutterBluePlus layer emits connection change stream instead of the exception?

chipweinberger, 2023-07-22:
please open a new Issue if you still have problems. thanks!

closed 2023-07-22 by chipweinberger