chipweinberger/flutter_blue_plus #405: 00002a51-0000-1000-8000-00805f9b34fbserviceUuid: 00001808-0000-1000-8000-00805f9b34fb E/flutter (21919): [ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: Exception: PlatformException(read_characteristic_error, The device is not connected,

iresolveservices, 2023-07-21:
characteristicUuid: 00002a51-0000-1000-8000-00805f9b34fbserviceUuid: 00001808-0000-1000-8000-00805f9b34fb
E/flutter (21919): [ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: Exception: PlatformException(read_characteristic_error, The device is not connected, null, null) #0      StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:652:7)
E/flutter (21919): #1      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:310:18)
E/flutter (21919): <asynchronous suspension>
E/flutter (21919): #2      BluetoothCharacteristic.read.<anonymous closure> (package:flutter_blue_plus/src/bluetooth_characteristic.dart:117:7)
E/flutter (21919): <asynchronous suspension>
E/flutter (21919): #3      _Mutex.synchronized (package:flutter_blue_plus/src/utils.dart:311:7)
E/flutter (21919): <asynchronous suspension>
E/flutter (21919): #4      BluetoothCharacteristic.read (package:flutter_blue_plus/src/bluetooth_characteristic.dart:91:5)
E/flutter (21919): <asynchronous suspension>
E/flutter (21919): 
E/flutter (21919): #0      BluetoothCharacteristic.read.<anonymous closure> (package:flutter_blue_plus/src/bluetooth_characteristic.dart:134:7)
E/flutter (21919): #1      _RootZone.runBinary (dart:async/zone.dart:1666:54)
E/flutter (21919): #2      _FutureListener.handleError (dart:async/future_impl.dart:162:22)
E/flutter (21919): #3      Future._propagateToListeners.handleError (dart:async/future_impl.dart:796:47)
E/flutter (21919): #4      Future._propagateToListeners (dart:async/future_impl.dart:817:13)
E/flutter (21919): #5      Future._completeError (dart:async/future_impl.dart:592:5)
E/flutter (21919): #6      Future._asyncCompleteError.<anonymous closure> (dart:async/future_impl.dart:683:7)
E/flutter (21919): #7      _microtaskLoop (dart:async/schedule_microtask.dart:40:21)
E/flutter (21919): #8      _startMicrotaskLoop (dart:async/schedule_microtask.dart:49:5)

"We integrated the same BLE glucometer device with Android native, and it is working fine. 

00002a18-0000-1000-8000-00805f9b34fb
00002a52-0000-1000-8000-00805f9b34fb

above both services getting in android native 

I think you have a problem with Bluetooth GATT services."

[image]

D/BluetoothGatt: registerApp()
D/BluetoothGatt: registerApp() - UUID=8d1fe9a1-9009-4469-a422-d31b06205c43
D/BluetoothGatt: onClientRegistered() - status=0 clientIf=13
D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=13 device=F4:B8:98:45:0C:E8
D/BluetoothGatt: discoverServices() - device: F4:B8:98:45:0C:E8
D/BluetoothGatt: onSearchComplete() = Device=F4:B8:98:45:0C:E8 Status=0
D/BluetoothGatt: setCharacteristicNotification() - uuid: 00002a18-0000-1000-8000-00805f9b34fb enable: true
D/BluetoothGatt: setCharacteristicNotification() - uuid: 00002a52-0000-1000-8000-00805f9b34fb enable: true
D/CompatibilityChangeReporter: Compat change id reported: 171228096; UID 10828; state: ENABLED
D/BluetoothAdapter: isLeEnabled(): ON

for more ref  https://github.com/Pekwerike/SugaIOT

iresolveservices, 2023-07-21:
https://www.youtube.com/watch?v=BSC11n1Fvz4

chipweinberger, 2023-07-21:
it says the error right here: "The device is not connected"

You need to connect first.

closed 2023-07-21 by chipweinberger

iresolveservices, 2023-07-21:
it happens because of adpter state will open right issue in some time