chipweinberger/flutter_blue_plus #406: characteristicUuid: 00002a18-0000-1000-8000-00805f9b34fbserviceUuid: 00001808-0000-1000-8000-00805f9b34fb E/flutter (32044): [ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: Exception: PlatformException(read_characteristic_error, The READ property is not supported by this BLE characteristic, null, null) #0

iresolveservices, 2023-07-21:
characteristicUuid: 00002a18-0000-1000-8000-00805f9b34fbserviceUuid: 00001808-0000-1000-8000-00805f9b34fb
E/flutter (32044): [ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: Exception: PlatformException(read_characteristic_error, The READ property is not supported by this BLE characteristic, null, null) #0      StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:652:7)
E/flutter (32044): #1      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:310:18)
E/flutter (32044): <asynchronous suspension>
E/flutter (32044): #2      BluetoothCharacteristic.read.<anonymous closure> (package:flutter_blue_plus/src/bluetooth_characteristic.dart:117:7)
E/flutter (32044): <asynchronous suspension>
E/flutter (32044): #3      _Mutex.synchronized (package:flutter_blue_plus/src/utils.dart:311:7)
E/flutter (32044): <asynchronous suspension>
E/flutter (32044): #4      BluetoothCharacteristic.read (package:flutter_blue_plus/src/bluetooth_characteristic.dart:91:5)
E/flutter (32044): <asynchronous suspension>
E/flutter (32044): 
E/flutter (32044): #0      BluetoothCharacteristic.read.<anonymous closure> (package:flutter_blue_plus/src/bluetooth_characteristic.dart:134:7)
E/flutter (32044): #1      _RootZone.runBinary (dart:async/zone.dart:1666:54)
E/flutter (32044): #2      _FutureListener.handleError (dart:async/future_impl.dart:162:22)
E/flutter (32044): #3      Future._propagateToListeners.handleError (dart:async/future_impl.dart:796:47)
E/flutter (32044): #4      Future._propagateToListeners (dart:async/future_impl.dart:817:13)
E/flutter (32044): #5      Future._completeError (dart:async/future_impl.dart:592:5)
E/flutter (32044): #6      Future._asyncCompleteError.<anonymous closure> (dart:async/future_impl.dart:683:7)
E/flutter (32044): #7      _microtaskLoop (dart:async/schedule_microtask.dart:40:21)
E/flutter (32044): #8      _startMicrotaskLoop (dart:async/schedule_microtask.dart:49:5)
E/flutter (32044):

iresolveservices, 2023-07-21:
"We integrated the same BLE glucometer device with Android native, and it is working fine.

00002a18-0000-1000-8000-00805f9b34fb
00002a52-0000-1000-8000-00805f9b34fb

above both services getting in android native 

I think you have a problem with Bluetooth GATT services."

iresolveservices, 2023-07-21:
D/BluetoothGatt: registerApp()
D/BluetoothGatt: registerApp() - UUID=8d1fe9a1-9009-4469-a422-d31b06205c43
D/BluetoothGatt: onClientRegistered() - status=0 clientIf=13
D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=13 device=F4:B8:98:45:0C:E8
D/BluetoothGatt: discoverServices() - device: F4:B8:98:45:0C:E8
D/BluetoothGatt: onSearchComplete() = Device=F4:B8:98:45:0C:E8 Status=0
D/BluetoothGatt: setCharacteristicNotification() - uuid: 00002a18-0000-1000-8000-00805f9b34fb enable: true
D/BluetoothGatt: setCharacteristicNotification() - uuid: 00002a52-0000-1000-8000-00805f9b34fb enable: true
D/CompatibilityChangeReporter: Compat change id reported: 171228096; UID 10828; state: ENABLED
D/BluetoothAdapter: isLeEnabled(): ON

for more ref https://github.com/Pekwerike/SugaIOT

https://www.youtube.com/watch?v=BSC11n1Fvz4

iresolveservices, 2023-07-21:
override fun onReceive(context: Context?, intent: Intent?) {
        intent?.let {
            when (intent.action) {
                BLUETOOTH_LE_GATT_ACTION_CONNECTED_TO_DEVICE -> {
                    val connectedDevice =
                        intent.getParcelableExtra<BluetoothDevice>(DEVICE_CONNECTED_TO_EXTRA)!!
                    bluetoothGattStateInformationCallback.connectedToAGattServer(connectedDevice = connectedDevice)
                }
                BLUETOOTH_LE_GATT_ACTION_DISCONNECTED_FROM_DEVICE -> {
                    bluetoothGattStateInformationCallback.disconnectedFromAGattServer()
                }
                BLUETOOTH_LE_GATT_ACTION_GLUCOSE_MEASUREMENT_RECORD_AVAILABLE -> {
                    synchronized(this) {
                        intent.getParcelableExtra<GlucoseMeasurementRecord>(
                            BLUETOOTH_LE_GATT_GLUCOSE_MEASUREMENT_RECORD_EXTRA
                        )?.let { glucoseMeasurementRecord ->
                            // send the new glucoseMeasurementRecord to the main activity to display it to the user
                            bluetoothGattStateInformationCallback.glucoseMeasurementRecordAvailable(
                                glucoseMeasurementRecord
                            )
                        }
                    }
                }
                RECORDS_SENT_COMPLETE -> {
                    bluetoothGattStateInformationCallback.recordsSentComplete()
                }
                else -> {

                }
            }
        }

MrCsabaToth, 2023-07-21:
Flutter Blue Plus is going through some major refactoring lately. To rule out that your error would be connected to any of the recent changes could you test your code with pre 1.7.0 Flutter Blue Plus as well (let's say 1.5.2)?

chipweinberger, 2023-07-22:
The Glucose Measurement characteristic (UUID: 0x2A18) supports the following properties according to the Bluetooth SIG specifications:

- **Indicate**: This property means that the characteristic is capable of sending indications of its value to a connected device. An indication is similar to a notification, but it also requires a response from the receiver acknowledging that it got the data.

The Glucose Measurement characteristic does not support properties like "read", "write", "notify", or "broadcast".

you'll need to use `setNotifications` and `onValueReceived`. you cannot use `read()`

closed 2023-07-22 by chipweinberger