chipweinberger/flutter_blue_plus #408: Characteristic write never returns on iOS

RubenKid, 2023-07-21:
I am doing characteristic.write and it ends because its timeout but never got an answer from the device. This work perfect on Android though with the same Bluetooth device

Attached Logs:

2023-07-21 16:01:02.921469+0200 Runner[472:8143] [FBP-iOS] handleMethodCall: getConnectedDevices
2023-07-21 16:01:02.922586+0200 Runner[472:8143] [FBP-iOS] handleMethodCall: writeCharacteristic
2023-07-21 16:01:17.928 +0200: {BleNotifier} {} {Exception: TimeoutException after 0:00:15.000000: Future not completed } {INFO}

I have been digging into the code and it seems didWriteValueForCharacteristic is never called when doing peripheral.write. Is strange because according to Apple documentation it should be called always if peripheral has delegate associate, and this is the case. I would appreciate any help

Environment:
- iPhone 13 (iOS 16.1.1)

[✓] Flutter (Channel stable, 3.10.6, on macOS 13.3.1 22E261 darwin-arm64, locale en-ES)
[✓] Android toolchain - develop for Android devices (Android SDK version 32.1.0-rc1)
[✓] Xcode - develop for iOS and macOS (Xcode 14.3)
[✓] Chrome - develop for the web
[✓] Android Studio (version 2022.2)
[✓] Connected device (4 available)
[✓] Network resources

MrCsabaToth, 2023-07-21:
Which Flutter Blue Plus version did you use? Was it the latest (1.8.8 @ 2023/07/21 16:11PM PDT)?

chipweinberger, 2023-07-22:
> I have been digging into the code and it seems didWriteValueForCharacteristic is never called when doing peripheral.write. Is strange because according to Apple documentation it should be called always if peripheral has delegate associate, and this is the case. I would appreciate any help

Hmm. that is strange.  It could be a problem with your bluetooth device or bluetooth radio interference. 

Does the device receive the write?

RubenKid, 2023-07-24:
@MrCsabaToth yes I am using latest version, tried with old ones also just in case but still same thing

@chipweinberger I am asking other team mates to test with different device and I will let you know. But, I am checking with the ioS RnF app and I see the device disconnects after writing on characteristic, this could be a reason I believe

[image]

chipweinberger, 2023-07-24:
yes that would cause the issue you see. closing this issue.

closed 2023-07-24 by chipweinberger