chipweinberger/flutter_blue_plus #409: Version 1.7.0 and above breaks location permission in Android

shubhamsinghshubham777, 2023-07-21:
Up until version `1.5.2`, my app used to show its dependency on the `location` permission just fine. I need this permission in order to be able to scan for BLE devices and get their location. In the image below, we can see that it is present in the list of permissions:

[image]

But with version `1.7.0` and above, I am not able to see the location permission:

[image]

I can see that the library's `AndroidManifest.xml` has changed, i.e.

### v1.5.2
[image]

### v1.7.0
[image]

We might need to remove `android:usesPermissionFlags="neverForLocation"` because in some cases, people might need to use the location services in their apps. Or is there some other alternative using which I could replace this AndroidManifest.xml with my own AndroidManifest?

MrCsabaToth, 2023-07-21:
The location permissions related to Bluetooth has a history outside of Flutter Blue Plus: https://www.androidpolice.com/2021/05/19/android-12-apps-wont-ask-for-location-permissions-when-all-they-want-is-bluetooth-scanning-which-yes-was-a-thing/

Android required location permission for BLE only because they originally targeted the Bluetooth tracker scenarios (such as [Tile](https://www.tile.com/), [Nut](https://www.nutfind.com/), [AirTag](https://www.apple.com/airtag/)). I have many privacy savvy users of my app and it's hard to explain every time why my app would need location permission if it's only a Bluetooth gadget handling app.

So starting from Android 12 the location permission is no longer needed for BLE, and I think Android 13 changed a bit as well. Flutter Blue Plus went through some iterations (justa  few examples):
* https://github.com/boskokg/flutter_blue_plus/issues/370
* https://github.com/boskokg/flutter_blue_plus/commit/f70249ec27b9b2e8ff0472cb4ba6bf3df9c8175f
* https://github.com/boskokg/flutter_blue_plus/commit/df51fb22d45b5b395ba2cba32013d77ede36547f

If you need extra permissions you can always override the merged permissions merged together from all the manifests of your plugins which have native platform sides. With that being said there could be use-cases which require some adjustment, but any adjustment can break other use-cases. If the `android:usesPermissionFlags="neverForLocation"` was removed then privacy savvy users would complain why the app requires location permission on Android 12 or 13.

shubhamsinghshubham777, 2023-07-21:
Thank you for the explanation. Although I mentioned in the description above that I have already tried adding `ACCESS_COARSE_LOCATION` into my `AndroidManifest.xml` but it didn't work, I'll still try to see how I can make my Manifest override all other manifests (from dependencies). It would be awesome if you could point me to some resources regarding the same 🌟

MrCsabaToth, 2023-07-21:
I didn't want to squash your issue in any way, close it when you find the solution and post it here.

I know how to remove a permission: with the 
https://stackoverflow.com/questions/27962789/android-gradle-manifest-merger-override-uses-permission-attribute-coming-from
I think adding one into your manifest should simply unify the added permissions to the merged ones. However I don't see how to modify one, and I opened a Stack Overflow entry about that: https://stackoverflow.com/questions/76741214/how-to-modify-android-manifest-permission-coming-form-the-merged-manifest

chipweinberger, 2023-07-26:
If you fix this issue, please let me know what your manifest looks like so we can add it to the readme.

chipweinberger, 2023-07-26:
Seems like you need to add:

```
      <!-- remove plugin permission which sets neverForLocation -->
    <uses-permission
        android:name="android.permission.BLUETOOTH_SCAN"
        tools:node="remove"
        tools:selector="com.boskokg.flutter_blue_plus">
    </uses-permission>
 
      <!-- add back without neverForLocation -->
    <uses-permission android:name="android.permission.BLUETOOTH_SCAN"/>
```

chipweinberger, 2023-07-26:
I've updated the readme

### Add permissions for Bluetooth

We need to add the permission to use Bluetooth and access location:

#### **Android (No Location)**

In the **android/app/src/main/AndroidManifest.xml** add:

```xml
<uses-permission android:name="android.permission.BLUETOOTH_SCAN" android:usesPermissionFlags="neverForLocation" />
<uses-permission android:name="android.permission.BLUETOOTH_CONNECT" />
<uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" android:maxSdkVersion="30"/>
```

#### **Android (With Fine Location)**

```xml
<!-- remove the plugin permission which sets neverForLocation -->
<uses-permission
    android:name="android.permission.BLUETOOTH_SCAN"
    tools:node="remove"
    tools:selector="com.boskokg.flutter_blue_plus">
</uses-permission>

<!-- add back without neverForLocation -->
<uses-permission android:name="android.permission.BLUETOOTH_SCAN" />
<uses-permission android:name="android.permission.BLUETOOTH_CONNECT" />
<uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" android:maxSdkVersion="30"/>
```

```dart
// Start scanning
flutterBlue.startScan(timeout: Duration(seconds: 4), androidUsesFineLocation: true);
```

shubhamsinghmutualmobile, 2023-07-26:
This is great @chipweinberger, although I had to add another remove block for the location permission:
```
<uses-permission
        android:name="android.permission.ACCESS_FINE_LOCATION"
        tools:node="remove"
        tools:selector="com.boskokg.flutter_blue_plus">
    </uses-permission>
```

It now works as expected! Thanks for the solution. We can close this issue now ✅

shubhamsinghshubham777, 2023-07-26:
This is great @chipweinberger, although I had to add another remove block for the location permission:
```
<uses-permission
        android:name="android.permission.ACCESS_FINE_LOCATION"
        tools:node="remove"
        tools:selector="com.boskokg.flutter_blue_plus">
    </uses-permission>
```
And then add the ACCESS_FINE_LOCATION permission again:
```
<uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />
```

It now works as expected! Thanks for the solution. We can close this issue now ✅

closed 2023-07-26 by shubhamsinghshubham777