chipweinberger/flutter_blue_plus #410: Macos is carsh

fanchou, 2023-07-21:
`*** Terminating app due to uncaught exception 'NSInvalidArgumentException', reason: '*** -[__NSPlaceholderDictionary initWithObjects:forKeys:count:]: attempt to insert nil object from objects[2]'
*** First throw call stack:
(
	0   CoreFoundation                      0x00007fff20a98bdb __exceptionPreprocess + 242
	1   libobjc.A.dylib                     0x00007fff207d1d92 objc_exception_throw + 48
	2   CoreFoundation                      0x00007fff20b4efea _CFThrowFormattedException + 202
	3   CoreFoundation                      0x00007fff20b5aad1 -[__NSPlaceholderDictionary initWithCapacity:].cold.1 + 0
	4   CoreFoundation                      0x00007fff209a8dee -[__NSPlaceholderDictionary initWithObjects:forKeys:count:] + 251
	5   CoreFoundation                      0x00007fff209a8cd9 +[NSDictionary dictionaryWithObjects:forKeys:count:] + 49
	6   flutter_blue_plus                   0x00000001022fcdc6 -[FlutterBluePlusPlugin toCharacteristicProto:characteristic:] + 3398
	7   flutter_blue_plus                   0x00000001022fbb97 -[FlutterBluePlusPlugin toServiceProto:service:] + 439
	8   flutter_blue_plus                   0x00000001022f6737 -[FlutterBluePlusPlugin peripheral:didDiscoverDescriptorsForCharacteristic:error:] + 711
	9   CoreBluetooth                       0x00007fff30ec3caa -[CBPeripheral handleAttributeEvent:args:attributeSelector:delegateSelector:delegateFlag:] + 219
	10  CoreBluetooth                       0x00007fff30ec3de1 -[CBPeripheral handleCharacteristicEvent:characteristicSelector:delegateSelector:delegateFlag:] + 113
	11  CoreBluetooth                       0x00007fff30ec04f6 -[CBPeripheral handleMsg:args:] + 291
	12  CoreBluetooth                       0x00007fff30eb5dda -[CBCentralManager handleMsg:args:] + 189
	13  CoreBluetooth                       0x00007fff30ea8037 -[CBManager xpcConnectionDidReceiveMsg:args:] + 177
	14  CoreBluetooth                       0x00007fff30ea7f73 __30-[CBXpcConnection _handleMsg:]_block_invoke + 66
	15  libdispatch.dylib                   0x00007fff2077b623 _dispatch_call_block_and_release + 12
	16  libdispatch.dylib                   0x00007fff2077c806 _dispatch_client_callout + 8
	17  libdispatch.dylib                   0x00007fff207825ea _dispatch_lane_serial_drain + 606
	18  libdispatch.dylib                   0x00007fff207830e0 _dispatch_lane_invoke + 417
	19  libdispatch.dylib                   0x00007fff20788a08 _dispatch_main_queue_callback_4CF + 613
	20  CoreFoundation                      0x00007fff20a5b8d8 __CFRUNLOOP_IS_SERVICING_THE_MAIN_DISPATCH_QUEUE__ + 9
	21  CoreFoundation                      0x00007fff20a1db32 __CFRunLoopRun + 2755
	22  CoreFoundation                      0x00007fff20a1c9ac CFRunLoopRunSpecific + 563
	23  HIToolbox                           0x00007fff28c681f3 RunCurrentEventLoopInMode + 292
	24  HIToolbox                           0x00007fff28c67f55 ReceiveNextEventCommon + 587
	25  HIToolbox                           0x00007fff28c67cf3 _BlockUntilNextEventMatchingListInModeWithFilter + 70
	26  AppKit                              0x00007fff23226ad2 _DPSNextEvent + 864
	27  AppKit                              0x00007fff232252a5 -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 1364
	28  AppKit                              0x00007fff232175c9 -[NSApplication run] + 586
	29  AppKit                              0x00007fff231eb7cc NSApplicationMain + 816

	31  libdyld.dylib                       0x00007fff20941f3d start + 1
)
libc++abi: terminating with uncaught exception of type NSException`

At the same time, I think the version of MacOS is too high.

MrCsabaToth, 2023-07-21:
What's the related Flutter/Dart and Swift/Object-C source?

fanchou, 2023-07-22:
> What's the related Flutter/Dart and Swift/Object-C source?

may be is here, no check for null

[image]

chipweinberger, 2023-07-22:
It says it crashes here `[FlutterBluePlusPlugin toCharacteristicProto:characteristic:]`

but im not sure where in that function exactly.

fanchou, 2023-07-22:
[FlutterBluePlusPlugin peripheral:didDiscoverDescriptorsForCharacteristic:error:] 

{uuid: 49535343-6daa-4d02-abf6-19569aca69fe, deviceId: C4:7F:0E:11:66:12, serviceUuid: 49535343-fe7d-4ae5-8fa9-9fafd205e455, secondaryServiceUuid: null, properties: CharacteristicProperties{broadcast: false, read: true, writeWithoutResponse: true, write: true, notify: false, indicate: false, authenticatedSignedWrites: false, extendedProperties: false, notifyEncryptionRequired: false, indicateEncryptionRequired: false}, descriptors: [], value: []}

It may be that the descriptors are empty causing this problem.

chipweinberger, 2023-07-22:
"This crash is happening because an attempt is being made to initialize an NSDictionary with an object or a key that is nil."

This is what ChatGPT says ^

chipweinberger, 2023-07-22:
So it could be a problem here

```
    // See BmBluetoothCharacteristic
    return @{
        @"characteristic_uuid":    [characteristic.UUID fullUUIDString],
        @"remote_id":              [peripheral.identifier UUIDString],
        @"service_uuid":           [pair.primary.UUID fullUUIDString],
        @"secondary_service_uuid": pair.secondary ? [pair.secondary.UUID fullUUIDString] : [NSNull null],
        @"descriptors":            descriptorProtos,
        @"properties":             propsMap,
        @"value":                  [self convertDataToHex:[characteristic value]],
    };
```
    
   Potentially this new function is causing pair.primary to be null?
   
```
   - (ServicePair *)getServicePair:(CBPeripheral *)peripheral
                 characteristic:(CBCharacteristic *)characteristic
```

chipweinberger, 2023-07-22:
Perhaps you can add logs to see which values are null, or put a breakpoint in XCode

Staars, 2023-07-22:
🤔 , the wave of the last updates did break basically everything and ATM I am a bit lost.
I am not familar enough with the codebase to fix this myself. Whatever I tried to catch variables with value null simply ended with functions that do nothing (nothing read/write/notitified).
@chipweinberger  Maybe the easiest thing would be, if you just go through the example app, that does not work for me now.
Expected behavior:
1. Scanning works.
2. After pressing CONNECT we get an exception `ArgumentError (Invalid argument: Instance of 'DeviceIdentifier')` on `      yield await device.readRssi();`in line 372.
3. If we comment out this line we probably see readable values on the characteristics, but if we reads them again, the become empty.
4. Trying to subscribe will raise an exception in line 227 of main.dart: `FlutterBluePlusException (FlutterBluePlusException: name:setNotifyValueFail errorCode:-1, errorString:notifications were not updated)`

Slightly off topic: Maybe you can add macOS to the example app in the repo, which was pretty automatic, although not completely successful for now :)

chipweinberger, 2023-07-22:
I've fixed the `DeviceIdentifier` error in 1.9.2.

> notifications were not updated

It's possible the check I added for this is not correct. It's strange for that check to be failing. This needs more investigation.

Staars, 2023-07-23:
Thanks, for the additions.
At least one throwing call is left here (with my test device):
`    await FlutterBluePlus.instance._channel
        .invokeMethod('requestMtu', request.toMap());`

Will try to test more later.

chipweinberger, 2023-07-23:
android or iOS?

iOS does not support requestMtu

Staars, 2023-07-23:
Only macOS (perhaps the same on iOS).
But even if it is not supported, it should gracefully fail and not throw an exception and stop.
But besides - could it make sense to use `- (NSUInteger)maximumWriteValueLengthForType:(CBCharacteristicWriteType)type;` to emulate requestMTU functionality?

Android seems to look better now. I was not able to crash the demo app within 60 seconds, but I will continue to do more tests.

chipweinberger, 2023-07-23:
`maximumWriteValueLengthForType` is not a request to change the value. It just returns the current value.

Throwing an exception is the only reasonable solution here IMO. Failing silently is terrible and makes things very hrd to debug. 

You should just add a `if(Platform.android) {}` to your code if you want to use Android specific features.

Staars, 2023-07-23:
Then please add this behavior to the example app too, which is easily crashable by just pressing a button on the screen.

I have no problem with the platform check. The reason for my confusion here was, that I did not see some info text in the exception on my console, which made me think, this is some totally unexpected behavior.

My rough idea about the "emulation" would have been to indeed present the `maximumWriteValueLengthForType` on request or use the value from the other device if it is smaller. Then it should be possible to use this limit in the Dart part of flutter_blue_plus and the iOS/macOS part will already get the correct (maybe truncated) length.
I can test this myself locally. Maybe it's a bad idea, IDK ...

Staars, 2023-07-23:
Nice update to the example app, Thanks 👍 !

Staars, 2023-07-23:
And finally my little flutter BLE app works again with your latest changes on macOS too (after refactoring for the small API changes 😄).
Thanks for your work!!

chipweinberger, 2023-07-23:
glad to hear it!

the refactors are done, so stability is going to increase, and i'll be testing and squashing every bug.

chipweinberger, 2023-07-24:
@fanchou ,  is the original crash still happening?

chipweinberger, 2023-07-25:
I think these problems have been fixed. We made fixes in bluetooth_messages.dart and related code.

Please reopen a new issue if you hit the problem again.

fanchou, 2023-07-25:
The latest version still crashes.

`return @{
        @"remote_id":              [peripheral.identifier UUIDString],
        @"service_uuid":           [pair.primary.UUID fullUUIDString],
        @"secondary_service_uuid": pair.secondary ? [pair.secondary.UUID fullUUIDString] : [NSNull null],
        @"characteristic_uuid":    [characteristic.UUID fullUUIDString],
        @"descriptors":            descriptorProtos,
        @"properties":             propsMap,
        @"value":                  [self convertDataToHex:characteristic.value],
    };`

` (NSString *)convertDataToHex:(NSData *)data
{
    if (data == nil) {
        return @"";
    }
    const unsigned char *bytes = (const unsigned char *)[data bytes];
    NSMutableString *hexString = [NSMutableString new];

    for (NSInteger i = 0; i < data.length; i++) {
        [hexString appendFormat:@"%02x", bytes[i]];
    }

    return [hexString copy];
}`

maybe cash here, the data is nil.

chipweinberger, 2023-07-25:
Can you try doing this?

 return @{
        @"remote_id":              @"00000000-0000-0000-0000-000000000000",
        @"service_uuid":          @"00000000-0000-0000-0000-000000000000",
        @"secondary_service_uuid": @"00000000-0000-0000-0000-000000000000",
        @"characteristic_uuid":    @"00000000-0000-0000-0000-000000000000",
        @"descriptors":            [NSMutableArray new],
        @"properties":             [NSMutableArray new],
        @"value":                  [self convertDataToHex:characteristic.value],
    };

david675566, 2023-07-25:
> Can you try doing this?
>
> ```
>  return @{
>         @"remote_id":              @"00000000-0000-0000-0000-000000000000",
>         @"service_uuid":          @"00000000-0000-0000-0000-000000000000", [...]

Hi, I'm facing the same write issue after I upgraded the package version (as well as pods) too,
I did changed the return data as you mentioned within ```toCharacteristicProto``` method
but now it's gave me following error:
```
[VERBOSE-2:dart_vm_initializer.cc(41)] Unhandled Exception: type 'List<Object?>' is not a subtype of type 'Map<dynamic, dynamic>'
#0      new BmBluetoothCharacteristic.fromMap
bluetooth_msgs.dart:318
#1      new BmBluetoothService.fromMap
bluetooth_msgs.dart:265
#2      new BmDiscoverServicesResult.fromMap.<anonymous closure>
bluetooth_msgs.dart:413
#3      MappedListIterable.elementAt (dart:_internal/iterable.dart:415:31)
#4      ListIterator.moveNext (dart:_internal/iterable.dart:344:26)
#5      new _GrowableList._ofEfficientLengthIterable (dart:core-patch/growable_array.dart:189:27)
#6      new _GrowableList.of (dart:core-patch/growable_array.dart:150:28)
#7      new List.of (dart:core-patch/array_patch.dart:47:28)
#8      ListIterable.toList (dart:_internal/iterable.dart:214:7)
#9      new BmDiscoverServicesResult.fromMap
bluetooth_msgs.dart:414
#10     BluetoothDevice.discoverServices.<anonymous closure>
#11     _MapStream._handleData (dart:async/stream_pipe.dart:213:31)
#12     _ForwardingStreamSubscription._handleData (dart:async/stream_pipe.dart:153:13)
#13     _RootZone.runUnaryGuarded (dart:async/zone.dart:1594:10)
#14     _BufferingStreamSubscription._sendData (dart:async/stream_impl.dart:339:11)
#15     _BufferingStreamSubscription._add (dart:async/stream_impl.dart:271:7)
#16     _ForwardingStreamSubscription._add (dart:async/stream_pipe.dart:123:11)
#17     _MapStream._handleData (dart:async/stream_pipe.dart:218:10)
#18     _ForwardingStreamSubscription._handleData (dart:async/stream_pipe.dart:153:13)
#19     _RootZone.runUnaryGuarded (dart:async/zone.dart:1594:10)
#20     _BufferingStreamSubscription._sendData (dart:async/stream_impl.dart:339:11)
#21     _BufferingStreamSubscription._add (dart:async/stream_impl.dart:271:7)
#22     _ForwardingStreamSubscription._add (dart:async/stream_pipe.dart:123:11)
#23     _WhereStream._handleData (dart:async/stream_pipe.dart:195:12)
#24     _ForwardingStreamSubscription._handleData (dart:async/stream_pipe.dart:153:13)
#25     _RootZone.runUnaryGuarded (dart:async/zone.dart:1594:10)
#26     _BufferingStreamSubscription._sendData (dart:async/stream_impl.dart:339:11)
#27     _DelayedData.perform (dart:async/stream_impl.dart:515:14)
#28     _PendingEvents.handleNext (dart:async/stream_impl.dart:620:11)
#29     _PendingEvents.schedule.<anonymous closure> (dart:async/stream_impl.dart:591:7)
#30     _microtaskLoop (dart:async/schedule_microtask.dart:40:21)
#31     _startMicrotaskLoop (dart:async/schedule_microtask.dart:49:5)
```

chipweinberger, 2023-07-25:
Okay, thats not as helpful as I thought. Instead try

```
NSObject* a = [peripheral.identifier UUIDString];
NSObject* b = [pair.primary.UUID fullUUIDString];
NSObject* c = pair.secondary ? [pair.secondary.UUID fullUUIDString] : [NSNull null];
NSObject* d = [characteristic.UUID fullUUIDString];
NSObject* e = descriptorProtos;
NSObject* f = propsMap;
NSObject* g = [self convertDataToHex:characteristic.value];

if (a == nil) {NSLog(@"a is nil");}
if (b == nil) {NSLog(@"b is nil");}
if (c == nil) {NSLog(@"c is nil");}
if (d == nil) {NSLog(@"d is nil");}
if (e == nil) {NSLog(@"e is nil");}
if (f == nil) {NSLog(@"f is nil");}
if (g == nil) {NSLog(@"g is nil");}

 return @{
        @"remote_id":              a,
        @"service_uuid":           b,
        @"secondary_service_uuid": c,
        @"characteristic_uuid":    d,
        @"descriptors":            e,
        @"properties":             f,
        @"value":                  g,
    };

```

david675566, 2023-07-26:
> Okay, thats not as helpful as I thought. Instead try
>
> ```
> NSObject* a = [peripheral.identifier UUIDString];
> NSObject* b = [pair.primary.UUID fullUUIDString];
> NSObject* c = pair.secondary ? [...]

Hi, thanks for the reply and sorry if I wasn't able to reply you asap,
yesterday when I try this one, the error stays the same as PlatformException data != nil problem like before,
but after I updated the package today, the issue seems resolved at my end.

chipweinberger, 2023-08-02:
closing. Please open if you still have issues.

Also please add `if (a == nil) {NSLog(@"a is nil");}` style lines everywhere so we can determine what is nil.

closed 2023-08-02 by chipweinberger