chipweinberger/flutter_blue_plus #411: The latest version does not work

fanchou, 2023-07-21:
{secondary_service_uuid: null, descriptors: [{service_uuid: 49535343-fe7d-4ae5-8fa9-9fafd205e455, remote_id: C4:7F:0E:11:66:12, descriptor_uuid: 00002902-0000-1000-8000-00805f9b34fb, characteristic_uuid: 49535343-1e4d-4bd9-ba61-23c647249616}], service_uuid: 49535343-fe7d-4ae5-8fa9-9fafd205e455, remote_id: C4:7F:0E:11:66:12, characteristic_uuid: 49535343-1e4d-4bd9-ba61-23c647249616, properties: {broadcast: 0, write_without_response: 0, notify_encryption_required: 0, read: 0, authenticated_signed_writes: 0, extended_properties: 0, indicate: 0, indicate_encryption_required: 0, write: 0, notify: 1}}

secondary_service_uuid may be null, which can cause a lot of problems.

 return BmOnCharacteristicResponse(
      type: bmOnCharacteristicResponseTypeParse(json['type']),
      remoteId: json['remote_id'],
      serviceUuid: Guid(json['service_uuid']),
      secondaryServiceUuid: Guid(json['secondary_service_uuid']),
      characteristicUuid: Guid(json['characteristic_uuid']),
      value: _hexDecode(json['value']),
      success: json['success'] != 0,
      errorCode: json['error_code'],
      errorString: json['error_string'],
    );

value may be null.

 return BmBluetoothService(
      serviceUuid: json['service_uuid'],
      remoteId: json['remote_id'],
      isPrimary: json['is_primary'] != 0,
      characteristics: chrs,
      includedServices: svcs,
    );

i think serviceUuid is Guid.

My bluetooth printer still doesn't work on the latest version, although I found some bugs.

It works on version 1.7.4，but the version of MacOS is too high, so my application cannot be packaged on macOS.

MrCsabaToth, 2023-07-21:
What do you mean by "the version of MacOS is too high"?
Were you able to overcome #410 ?

MrCsabaToth, 2023-07-22:
Actually I think I ran into this. At the line of
```secondaryServiceUuid: Guid(json['secondary_service_uuid']),```

```
type 'Null' is not a subtype of type 'String'
I/flutter ( 3095): #0      new BmBluetoothCharacteristic.fromMap (package:flutter_blue_plus/src/bluetooth_msgs.dart:272:38)
I/flutter ( 3095): #1      new BmBluetoothService.fromMap (package:flutter_blue_plus/src/bluetooth_msgs.dart:222:42)
I/flutter ( 3095): #2      new BmDiscoverServicesResult.fromMap.<anonymous closure> (package:flutter_blue_plus/src/bluetooth_msgs.dart:370:42)
I/flutter ( 3095): #3      MappedListIterable.elementAt (dart:_internal/iterable.dart:415:31)
I/flutter ( 3095): #4      ListIterator.moveNext (dart:_internal/iterable.dart:344:26)
I/flutter ( 3095): #5      new _GrowableList._ofEfficientLengthIterable (dart:core-patch/growable_array.dart:189:27)
I/flutter ( 3095): #6      new _GrowableList.of (dart:core-patch/growable_array.dart:150:28)
I/flutter ( 3095): #7      new List.of (dart:core-patch/array_patch.dart:47:28)
I/flutter ( 3095): #8      ListIterable.toList (dart:_internal/iterable.dart:214:7)
I/flutter ( 3095): #9      new BmDiscoverServicesResult.fromMap (package:flutter_blue_plus/src/bluetooth_msgs.dart:371:12)
I/flutter ( 3095): #10     BluetoothDevice.discoverServices.<anonymous closure> (package:flutter_blue_plus/src/bluetooth_device.dart:104:51)
I/flutter ( 3095): #11     _MapStream._handleData (dart:async/stream_pipe.dart:213:31)
I/flutter ( 3095): #12     _ForwardingStreamSubscription._handleData (dart:async/stream_pipe.dart:153:13)
I/flutter ( 3095): #13     _rootRunUnary (dart:async/zone.dart:1407:47)
I/flutter ( 3095): #14     _CustomZone.runUnary (dart:async/zone.dart:1308:19)
I/flutter ( 3095): #15     _CustomZone.runUnaryGuarded (dart:async/zone.dart:1217:7)
I/flutter ( 3095): #16     _BufferingStreamSubscription._sendData (dart:async/stream_impl.dart:339:11)
I/flutter ( 3095): #17     _BufferingStreamSubscription._add (dart:async/stream_impl.dart:271:7)
I/flutter ( 3095): #18     _ForwardingStreamSubscription._add (dart:async/stream_pipe.dart:123:11)
I/flutter ( 3095): #19     _MapStream._handleData (dart:async/stream_pipe.dart:218:10)
I/flutter ( 3095): #20     _ForwardingStreamSubscription._handleData (dart:async/stream_pipe.dart:153:13)
I/flutter ( 3095): #21     _rootRunUnary (dart:async/zone.dart:1407:47)
I/flutter ( 3095): #22     _CustomZone.runUnary (dart:async/zone.dart:1308:19)
I/flutter ( 3095): #23     _CustomZone.runUnaryGuarded (dart:async/zone.dart:1217:7)
I/flutter ( 3095): #24     _BufferingStreamSubscription._sendData (dart:async/stream_impl.dart:339:11)
I/flutter ( 3095): #25     _BufferingStreamSubscription._add (dart:async/stream_impl.dart:271:7)
I/flutter ( 3095): #26     _ForwardingStreamSubscription._add (dart:async/stream_pipe.dart:123:11)
I/flutter ( 3095): #27     _WhereStream._handleData (dart:async/stream_pipe.dart:195:12)
I/flutter ( 3095): #28     _ForwardingStreamSubscription._handleData (dart:async/stream_pipe.dart:153:13)
I/flutter ( 3095): #29     _rootRunUnary (dart:async/zone.dart:1407:47)
I/flutter ( 3095): #30     _CustomZone.runUnary (dart:async/zone.dart:1308:19)
I/flutter ( 3095): #31     _CustomZone.runUnaryGuarded (dart:async/zone.dart:1217:7)
I/flutter ( 3095): #32     _BufferingStreamSubscription._sendData (dart:async/stream_impl.dart:339:11)
I/flutter ( 3095): #33     _DelayedData.perform (dart:async/stream_impl.dart:515:14)
I/flutter ( 3095): #34     _PendingEvents.handleNext (dart:async/stream_impl.dart:620:11)
I/flutter ( 3095): #35     _PendingEvents.schedule.<anonymous closure> (dart:async/stream_impl.dart:591:7)
I/flutter ( 3095): #36     _rootRun (dart:async/zone.dart:1391:47)
I/flutter ( 3095): #37     _CustomZone.run (dart:async/zone.dart:1301:19)
I/flutter ( 3095): #38     _CustomZone.runGuarded (dart:async/zone.dart:1209:7)
I/flutter ( 3095): #39     _CustomZone.bindCallbackGuarded.<anonymous closure> (dart:async/zone.dart:1249:23)
I/flutter ( 3095): #40     _rootRun (dart:async/zone.dart:1399:13)
I/flutter ( 3095): #41     _CustomZone.run (dart:async/zone.dart:1301:19)
I/flutter ( 3095): #42     _CustomZone.runGuarded (dart:async/zone.dart:1209:7)
I/flutter ( 3095): #43     _CustomZone.bindCallbackGuarded.<anonymous closure> (dart:async/zone.dart:1249:23)
I/flutter ( 3095): #44     _microtaskLoop (dart:async/schedule_microtask.dart:40:21)
I/flutter ( 3095): #45     _startMicrotaskLoop (dart:async/schedule_microtask.dart:49:5)
I/flutter ( 3095): ; #0      new BmBluetoothCharacteristic.fromMap (package:flutter_blue_plus/src/bluetooth_msgs.dart:272:38)
I/flutter ( 3095): #1      new BmBluetoothService.fromMap (package:flutter_blue_plus/src/bluetooth_msgs.dart:222:42)
I/flutter ( 3095): #2      new BmDiscoverServicesResult.fromMap.<anonymous closure> (package:flutter_blue_plus/src/bluetooth_msgs.dart:370:42)
I/flutter ( 3095): #3      MappedListIterable.elementAt (dart:_internal/iterable.dart:415:31)
I/flutter ( 3095): #4      ListIterator.moveNext (dart:_internal/iterable.dart:344:26)
I/flutter ( 3095): #5      new _GrowableList._ofEfficientLengthIterable (dart:core-patch/growable_array.dart:189:27)
I/flutter ( 3095): #6      new _GrowableList.of (dart:core-patch/growable_array.dart:150:28)
I/flutter ( 3095): #7      new List.of (dart:core-patch/array_patch.dart:47:28)
I/flutter ( 3095): #8      ListIterable.toList (dart:_internal/iterable.dart:214:7)
[... cut ...]
I/flutter ( 3095): #40     _rootRun (dart:async/zone.dart:1399:13)
I/flutter ( 3095): #41     _CustomZone.run (dart:async/zone.dart:1301:19)
I/flutter ( 3095): #42     _CustomZone.runGuarded (dart:async/zone.dart:1209:7)
I/flutter ( 3095): #43     _CustomZone.bindCallbackGuarded.<anonymous closure> (dart:async/zone.dart:1249:23)
I/flutter ( 3095): #44     _microtaskLoop (dart:async/schedule_microtask.dart:40:21)
I/flutter ( 3095): #45     _startMicrotaskLoop (dart:async/schedule_microtask.dart:49:5)
```

fanchou, 2023-07-22:
[image]

12 is too high for some device.

but it's a different issue. These three bugs I found on Android, have nothing to do with MacOS.

MrCsabaToth, 2023-07-22:
1. I changed `Guid(json['secondary_service_uuid'])` to `json['secondary_service_uuid'] != null ? Guid(json['secondary_service_uuid']) : null`.
2. Then I there was a type mismatch (Dart compiler doesn't pick it up due to the non type safe nature of the dictionary which replaced ProtoBuf). It is `BmBluetoothService(` construction's `serviceUuid: json['service_uuid']`. It expects a `Guid` instead of a `String` so it should be `Guid(json['service_uuid'])`
3. After that I needed to change a few `value: _hexDecode(json['value'])` to `value: _hexDecode(json['value'] ?? "")` as it was at three other instances already.

I have a fork, but it has modifications for #412 and some other changes, so it's not 100% clean for only this issue: https://github.com/boskokg/flutter_blue_plus/compare/master...CsabaConsulting:flutter_blue_plus:master

chipweinberger, 2023-07-22:
it looks like you are using Master Branch? 

yes master branch is needs some work still. I'm in the middle of some refactors to simplify the code and make it more reliable.

MrCsabaToth, 2023-07-22:
> @MrCsabaToth I incorporated your changes.

Take them with a grain of salt. I'm reacting to the exceptions and applying to similar code sections with pattern search.

chipweinberger, 2023-07-22:
please open a new issue if you still have problems

closed 2023-07-22 by chipweinberger