chipweinberger/flutter_blue_plus #412: Is the -3 needed when checking for MTU size?

MrCsabaToth, 2023-07-21:
https://github.com/boskokg/flutter_blue_plus/commit/625ccd1a687395ca215dd7306f9f4a54ea523ee4 introduced MTU size checks. I have a device which requires everything to be chopped up to 20 bytes and I'm receiving errors now complaining about an MTU size of 17 (`20 - 3 = 17`).

MrCsabaToth, 2023-07-21:
It can happen that my particular device is whacky, because it has some other antics too, so I'm not 100% confident if this ticket is relevant.

chipweinberger, 2023-07-22:
perhaps not. is this iOS or android?

chipweinberger, 2023-07-22:
I pushed some changes

MrCsabaToth, 2023-07-22:
I'm working with the Android port of the plugin and my app.

chipweinberger, 2023-07-22:
let me know if the changes I pushed fixed the problem. I had a different solution than you. 

Android and iOS use a different values for MTU.

AFAICT, the MTU that iOS returns already has -3 done to it, but on Android it does not. 

Therefore mtu(20) on ios is equivalent to mtu(23) on Android.

MrCsabaToth, 2023-07-23:
Ref https://github.com/boskokg/flutter_blue_plus/commit/067cc8bd4c640dc32d700462b225f0ef077c70b7

MrCsabaToth, 2023-07-23:
I tested on Android and it worked with the machine.
Does the 20 come from some standard? Now I'm curious that other machines I worked with - how they'll fare, because some of the FTMS packets are longer than 20 bytes and in the past I received them as a whole. Did the old Flutter Blue also had this 20 byte default MTU just was not checking for it?

chipweinberger, 2023-07-23:
23 the the minimum MTU (and initial default) from the BLE standard.

20 is the payload size. (3 bytes are used by the BLE protocol itself)

closed 2023-07-24 by chipweinberger

jasaw, 2023-09-13:
@chipweinberger Can you please clarify this when I call `bluetoothDevice.mtu.first` to get the MTU size on iOS and Android.
On iOS, the call will return 20 and I don't need to subtract 3.
On Android, the call will return 23, and I have to subtract 3.
Is that correct?

chipweinberger, 2023-09-15:
they should both return 23

if android and ios do not agree, that is a bug in FBP and should be reported.