chipweinberger/flutter_blue_plus #413: API changed without major version number update

pvaibhav, 2023-07-23:
Just wanted to mention that the API changed significantly between v1.7.x and 1.9.x.

Isn't it generally accepted to update the major version number when making breaking changes to the APIs? I'd have assumed the version would be bumped to v2.x.x. Or am I misunderstanding something?

chipweinberger, 2023-07-23:
What API changed? There are no change AFAIK.

mrRedSun, 2023-07-24:
Renames are breaking changes
The BluetoothConnectionState, method renames are all breaking changes and should bump the major

chipweinberger, 2023-07-24:
doing an automated copy paste replacement is not a breaking change.

mrRedSun, 2023-07-24:
1. Any change in the package that requires users to do anything in the project is a breaking change
2. No, they don't
3. Classes were renamed so that at least is a breaking change

mrRedSun, 2023-07-24:
[image]
Changelogs seem to acknowledge the breaking changes

chipweinberger, 2023-07-24:
yes those are fair points. 1.10.0 should have been 2.0 using the dart versioning rules.

chipweinberger, 2023-07-24:
> Also moving methods from instance to static, why was it even done? Instance is way more testable

The change was made because multiple instances will not actually work. The plugin uses named platform channels that can only exist once, and Android and iOS only support a single Bluetooth adapter.

mrRedSun, 2023-07-25:
It's not really about multiple instances, the thing is that single instance classes can be mocked  – static classes can not

MrCsabaToth, 2023-07-26:
#407

jfacoustic, 2023-07-27:
@chipweinberger The only way that you'd be able to mock it is by creating a wrapper around Bluetooth calls.  If we were to update to the latest version of the package, we would have to change a significant amount of code because we would no longer be able to directly inject flutter blue plus as a dependency.  

It makes more sense to have an accessible instance getter rather than accessing Bluetooth calls through static methods.

chipweinberger, 2023-08-01:
@jfacoustic yes you're correct. apologies for that.

closed 2023-08-01 by chipweinberger