chipweinberger/flutter_blue_plus #414: Exception: Exception: Failed to write the characteristic

itlws, 2023-07-23:
`
[ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: Exception: Exception: Failed to write the characteristic #0      BluetoothCharacteristic.write.<anonymous closure> (package:flutter_blue_plus/src/bluetooth_characteristic.dart:176:21)
 <asynchronous suspension>
 #1      _Mutex.synchronized (package:flutter_blue_plus/src/utils.dart:316:13)
 <asynchronous suspension>
 #2      BluetoothCharacteristic.write (package:flutter_blue_plus/src/bluetooth_characteristic.dart:143:9)
 <asynchronous suspension>
 
 #0      BluetoothCharacteristic.write.<anonymous closure> (package:flutter_blue_plus/src/bluetooth_characteristic.dart:187:13)
 #1      _RootZone.runBinary (dart:async/zone.dart:1665:54)
 #2      _FutureListener.handleError (dart:async/future_impl.dart:162:22)
 #3      Future._propagateToListeners.handleError (dart:async/future_impl.dart:779:47)
 #4      Future._propagateToListeners (dart:async/future_impl.dart:800:13)
 #5      Future._completeError (dart:async/future_impl.dart:575:5)
 #6      Future._asyncCompleteError.<anonymous closure> (dart:async/future_impl.dart:666:7)
 #7      _microtaskLoop (dart:async/schedule_microtask.dart:40:21)
 #8      _startMicrotaskLoop (dart:async/schedule_microtask.dart:49:5)
`

chipweinberger, 2023-07-23:
update to 1.9.4.

ios or android?

it should tell you the reason it fails.

itlws, 2023-07-24:
Thank you! After upgrading to 1.9.4, I received an error message. The problem has now been resolved.

closed 2023-07-24 by itlws