chipweinberger/flutter_blue_plus #417: cannot connect without scanning on iOS/MacOS : PlatformException(connect, Peripheral not found, null, null)

mocheffendi, 2023-07-24:
BluetoothDevice.fromId(Uuid)
use Uuid or macAddress for iOS/MacOS ?

2023-07-24 15:25:16.112 macos[44629:7566851] [FBP-iOS] handleMethodCall: connect
2023-07-24 15:25:16.309 macos[44629:7566851] [FBP-iOS] centralManagerDidUpdateState 5
2023-07-24 15:25:16.387 macos[44629:7566851] [FBP-iOS] handleMethodCall: getConnectionState
[ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: PlatformException(connect, Peripheral not found, null, null)

chipweinberger, 2023-07-24:
> use Uuid or macAddress for iOS/MacOS ?

You should use a the UUID of the remote device. iOS generates this UUID randomly per device. It changes every so often, so if the peripheral cannot be found, the UUID may have expired and you'll need to scan again.

let me know if you fix the problem

mocheffendi, 2023-07-24:
I did another scan and the remoteId is still the same, did I miss something? it works on Android. 

please give an example function code to scan without showing it on the page.

chipweinberger, 2023-07-24:
To be clear, you are trying to connect without having to first scan every time right?

I suggest you put some logs in this function to see why it is not found.

```
- (CBPeripheral *)findPeripheral:(NSString *)remoteId
{
    NSUUID *uuid = [[NSUUID alloc] initWithUUIDString:remoteId];

    NSArray<CBPeripheral *> *peripherals = [_centralManager retrievePeripheralsWithIdentifiers:@[uuid]];

    CBPeripheral *peripheral;
    for (CBPeripheral *p in peripherals)
    {
        if ([[p.identifier UUIDString] isEqualToString:remoteId])
        {
            peripheral = p;
            break;
        }
    }
    return peripheral;
}
```

mocheffendi, 2023-07-25:
I try this and solve, thanks for your great library.

scanconnect() async {
    try {
      // Start scanning for Bluetooth devices.
      FlutterBluePlus.scan(timeout: Duration(seconds: 5))
          .listen((scanResult) async {
        // Check if the desired device is found using its deviceId.
        if (scanResult.device.remoteId.toString() == macAddress) {
          targetdevice = scanResult.device;
          print(targetdevice.toString());

          try {
            // Connect to the desired device.
            await targetdevice.connect(
                autoConnect:
                    false); // Set autoConnect to true if you want auto-reconnect.
            print('Connected to device: ${targetdevice.localName}');

            // You can now discover services and characteristics to communicate with the device.
            discoverServices();
          } catch (e) {
            print('Error connecting to device: $e');
          }
        }
      });
    } catch (e) {
      print('Error scanning for devices: $e');
    }
}

chipweinberger, 2023-07-25:
what was the fix?

mocheffendi, 2023-07-25:
stopped at stopscan

2023-07-25 07:18:18.621 macos[35640:7922083] [FBP-iOS] handleMethodCall: setLogLevel
2023-07-25 07:18:18.622 macos[35640:7922083] [FBP-iOS] handleMethodCall: startScan
2023-07-25 07:18:18.749 macos[35640:7922083] [FBP-iOS] centralManagerDidUpdateState 5
2023-07-25 07:18:23.624 macos[35640:7922083] [FBP-iOS] handleMethodCall: stopScan

but when click the restart button at debug toolbar it can go

2023-07-25 08:39:38.077 macos[40460:7939719] [FBP-iOS] handleMethodCall: setLogLevel
2023-07-25 08:39:38.078 macos[40460:7939719] [FBP-iOS] handleMethodCall: startScan
flutter: BluetoothDevice{remoteId: 38D9D5EA-2A41-0D32-0AB6-43A1B4CF0806, localName: ESP32 Test, type: BluetoothDeviceType.le, isDiscoveringServices: false, _services: []}
2023-07-25 08:39:38.287 macos[40460:7939719] [FBP-iOS] handleMethodCall: connect
2023-07-25 08:39:38.290 macos[40460:7939719] [FBP-iOS] handleMethodCall: getConnectionState
flutter: BluetoothDevice{remoteId: 38D9D5EA-2A41-0D32-0AB6-43A1B4CF0806, localName: ESP32 Test, type: BluetoothDeviceType.le, isDiscoveringServices: false, _services: []}

chipweinberger, 2023-07-25:
I don't follow. What was the fix? what lines of code did you need to add?

If I can improve our code to be better, I would like to.

WeiKuang, 2023-07-25:
[image]

chipweinberger, 2023-07-25:
@WeiKuang why would that fix anything?

WeiKuang, 2023-07-25:
@chipweinberger 
I think it's keeping a reference to the peripheral. Of course, this is just my guess, not a perfect solution.

chipweinberger, 2023-07-25:
I've researched the problem more and released 1.10.5

JDongKhan, 2023-08-01:
same issue when upgrade 1.12.8.

```
 NSArray<CBPeripheral *> *peripherals = [_centralManager retrievePeripheralsWithIdentifiers:@[uuid]];
```
peripherals is empty array.

JDongKhan, 2023-08-01:
only call connect.
[image]

after call  scan  and then call connect
[image]

chipweinberger, 2023-08-01:
thanks for debugging a bit more.

If `retrievePeripheralsWithIdentifiers` is empty, I'm not sure what else we can do. 

Maybe try reading the iOS documentation more. Perhaps this is expected?

JDongKhan, 2023-08-01:
Maybe, I have reviewed many blog and get same response.

chipweinberger, 2023-08-01:
ok. tell me if you find a fix

chipweinberger, 2023-08-01:
ChatGPT:
- This method only returns peripherals that have been discovered during a prior scan by your app and that are currently in range. If the peripheral is not in range or turned off, the method may not return that peripheral.

Seems like this is expected behavior. Your app needs to scan at least once.

closed 2023-08-01 by chipweinberger