chipweinberger/flutter_blue_plus #422: Bluetooth write and read issue in IOS

Narayanhegde14, 2023-07-25:
I was trying the same functionality as Android for ios mobile app. When the app gets connected I get a connected response, after that when I start writing to the device, my read character listen is not returning any data.

flutter_blue_plus version is 1.8.1

console looking like this 
[FBP-iOS] handleMethodCall: writeCharacteristic
[FBP-iOS] didWriteValueForCharacteristic
[FBP-iOS] handleMethodCall: writeCharacteristic
[FBP-iOS] didWriteValueForCharacteristic
[FBP-iOS] handleMethodCall: isOn
[FBP-iOS] handleMethodCall: writeCharacteristic
[FBP-iOS] didWriteValueForCharacteristic
[FBP-iOS] handleMethodCall: writeCharacteristic
[FBP-iOS] didWriteValueForCharacteristic
[FBP-iOS] handleMethodCall: writeCharacteristic
[FBP-iOS] didWriteValueForCharacteristic
[FBP-iOS] handleMethodCall: writeCharacteristic
[FBP-iOS] didWriteValueForCharacteristic

chipweinberger, 2023-07-25:
update to 1.10.6+

closed 2023-07-25 by chipweinberger

Narayanhegde14, 2023-07-27:
In flutter_blue_plus version is 1.8.1 my Android functionalities are working fine and when I try to run the same thing in ios Guid("6e400003-b5a3-f393-e0a9-e50e24dcca9e") uuid listen is not giving any responses. 

ie. Guid("6e400002-b5a3-f393-e0a9-e50e24dcca9e") in this UUID I'm writing to Bluetooth and in Guid("6e400003-b5a3-f393-e0a9-e50e24dcca9e") UUID I'm trying to listen to the character returned from Bluetooth. Its working fine in android, Issue is in IOS.

Narayanhegde14, 2023-07-27:
> > when I start writing to the device, my read character listen is not returning any data.
> 
> if you want to read, you need to call `read()`.
> 
> ```
>   // this stream is updated:
>   //   - after read() is called
>   //   - when a notification arrives
>   Stream<List<int>> get onValueReceived => FlutterBluePlus._methodStream.stream
> ```
> 
> You should also update to 1.10.6+

In flutter_blue_plus version is 1.8.1 my Android functionalities are working fine and when I try to run the same thing in ios Guid("6e400003-b5a3-f393-e0a9-e50e24dcca9e") UUID listen is not giving any responses.

ie. Guid("6e400002-b5a3-f393-e0a9-e50e24dcca9e") in this UUID I'm writing to Bluetooth and in Guid("6e400003-b5a3-f393-e0a9-e50e24dcca9e") UUID I'm trying to listen to the character returned from Bluetooth. It's working fine in Android, Issue is in IOS.