chipweinberger/flutter_blue_plus #424: Is there any way to connect to already connect device where we have a mac address?

Snofkin, 2023-07-25:
Nn flutter blue there was a 'flutterblue.pd.dart but i could find any smilar class like this where we can connect to devices

`import 'package:flutter_blue/flutter_blue.dart';
import 'package:flutter_blue/gen/flutterblue.pb.dart' as proto;

...

 BluetoothDevice toBluetoothDevice() {
   proto.BluetoothDevice p = proto.BluetoothDevice.create();

   p.name = name;
   p.type = proto.BluetoothDevice_Type.LE;
   p.remoteId = bleno;

   return BluetoothDevice.fromProto(p);
 }`

chipweinberger, 2023-07-25:
Yes, this is still possible. 

Just construct the BluetoothDevice anyway you want (`remoteId` is the only important variable), and then call device.connect()

Snofkin, 2023-07-26:
Thanks... But i could not find any connect() method with BmBluetoothDevice...
[image]

Snofkin, 2023-07-26:
How can we connect since, we don't have connect() method with BMBluetoothDevice.  @chipweinberger

chipweinberger, 2023-07-26:
use

```
  BluetoothDevice.fromProto(BmBluetoothDevice p)
      : remoteId = DeviceIdentifier(p.remoteId),
        localName = p.localName ?? "",
        type = bmToBluetoothDeviceType(p.type);

Then call connect()
```