chipweinberger/flutter_blue_plus #426: iOS writeCharacteristic sent in disorder

fgendrecapacites, 2023-07-25:
Hello,
i've recently upgraded from flutter_blue to flutter_plus and have some problems on iOS only.
Everything worked perfectly on both android and iOS for the past 8 months. The code logic hasn't changed.

For the context, this is a firmware update by Bluetooth, it consists in sending the firmware packet by packet with a specific command.
Those chunks are sent synchronously using mutex protection. 
The problem only happens after a while when lots of data are being sent. 

Here is an example of one of the many packets sent (packet n° 449): 

PFWUD-0449-0-0512-30303036653635303030303064303030303030353036313030303032660D0A3A3130646439303030373337333030303036313637303030303635323030303030373636353030303037350D0A3A3130646461303030373237333030303032303664303030303666363430303030363532303030303061390D0A3A3130646462303030343535343030303034353439303030303465353430303030306430303030303038640D0A3A3130646463303030353036313030303037333733303030303631363730303030363532303030303036660D0A3A3130646464303030373636353030303037323733303030303230366430303030366636343030303032330D0A[CR]

[CR] indicates the end of the string 

This string is chunked in 4 pieces : 

1. PFWUD-0449-0-0512-30303036653635303030303064303030303030353036313030303032660D0A3A3130646439303030373337333030303036313637303030303635323030
2. 303030373636353030303037350D0A3A3130646461303030373237333030303032303664303030303666363430303030363532303030303061390D0A3A313064646230303034
3. 3535343030303034353439303030303465353430303030306430303030303038640D0A3A31306464633030303530363130303030373337333030303036313637303030303635
4. 32303030303036660D0A3A3130646464303030373636353030303037323733303030303230366430303030366636343030303032330D0A[CR]

Here is the log from one of all the commands sent correctly before the problem : 

```
flutter: PFWUD-0448-0-0512-3430303030306430303030303032640D0A3A3130646433303030353036313030303037333733303030303631363730303030363532303030303066660D
[FBP-iOS] handleMethodCall: writeCharacteristic
[FBP-iOS] peripheralIsReadyToSendWriteWithoutResponse

flutter: 0A3A3130646434303030373636353030303037323733303030303230353030303030346635373030303066640D0A3A3130646435303030343535323030303032303434303030
[FBP-iOS] handleMethodCall: writeCharacteristic
[FBP-iOS] peripheralIsReadyToSendWriteWithoutResponse

flutter: 303466353730303030346530643030303063370D0A3A3130646436303030303034353030303037383734303030303635366530303030373336393030303064330D0A3A313064
[FBP-iOS] handleMethodCall: writeCharacteristic
[FBP-iOS] peripheralIsReadyToSendWriteWithoutResponse

flutter: 6437303030366636653030303032303639303030303665363930303030373432303030303064320D0A3A31306464383030303634366630[CR]
[FBP-iOS] handleMethodCall: writeCharacteristic
[FBP-iOS] peripheralIsReadyToSendWriteWithoutResponse

flutter: $< OK
```

As you can see, one chunk is sent after another, with alternation between "writeCharacteristic" and "peripheralIsReadyToSendWriteWithoutResponse".

But after a while, here is what happens for another  command : 

```
flutter: PFWUD-0449-0-0512-30303036653635303030303064303030303030353036313030303032660D0A3A3130646439303030373337333030303036313637303030303635323030
[FBP-iOS] handleMethodCall: writeCharacteristic

flutter: 303030373636353030303037350D0A3A3130646461303030373237333030303032303664303030303666363430303030363532303030303061390D0A3A313064646230303034
[FBP-iOS] handleMethodCall: writeCharacteristic

flutter: 3535343030303034353439303030303465353430303030306430303030303038640D0A3A31306464633030303530363130303030373337333030303036313637303030303635
[FBP-iOS] handleMethodCall: writeCharacteristic

flutter: 32303030303036660D0A3A3130646464303030373636353030303037323733303030303230366430303030366636343030303032330D0A[CR]
[FBP-iOS] handleMethodCall: writeCharacteristic

[FBP-iOS] peripheralIsReadyToSendWriteWithoutResponse
[FBP-iOS] peripheralIsReadyToSendWriteWithoutResponse
flutter: $< NO-FRAME
```

As you can see, `peripheralIsReadyToSendWriteWithoutResponse` happens later and only twice. 
I was able to trace with Wireshark and it seems that only the first and last chunks are sent (since the last chunk contains [CR] the communication stops). 

There is a chance that other chunks get sent in disorder and that it only crashes here because the randomness happens to send the last chunk before it should. 

Adding delay betweend writes doesn't seem to solve the problem.

Do you have any idea why this happens ?

chipweinberger, 2023-07-25:
```
            // device not ready?
            if (type == CBCharacteristicWriteWithoutResponse && !peripheral.canSendWriteWithoutResponse) {
                // canSendWriteWithoutResponse is the current readiness of the peripheral to accept
                // more write requests. If the peripheral isn't ready, we queue the request for later.
                [_dataWaitingToWriteWithoutResponse setObject:args forKey:remoteId];
                result(@(true));
                return;
            } 
```

This code will be problematic for you. We only queue the most recent write. Perhaps we should add a enum flag to configure behavior.  

```
enum QueueFullBehavior {
    fail
    skip
    queueAll
    queueLast
}
```

chipweinberger, 2023-07-25:
> On iOS there is peripheralIsReadyToSendWriteWithoutResponse. What is the behavior on android when you call writeCharacteristic a lot without response?
> 
> ChatGPT
> Android does not have a similar API as iOS's peripheralIsReadyToSendWriteWithoutResponse. When you issue a lot of Write Without Response requests on Android, you must manage the pacing yourself.
> 
> In Android's BLE implementation, a Write Without Response request is issued with the BluetoothGattCharacteristic.WRITE_TYPE_NO_RESPONSE write type. Unlike a regular write request, this request does not generate a callback to the onCharacteristicWrite method in the BluetoothGattCallback instance.
> 
> However, Android does not queue these write requests for you, and the BLE specification itself mandates a minimum delay of 150 microseconds between each packet. Most devices use a longer delay for robustness.
> 
> If you write too many characteristics too quickly, the BluetoothGatt instance will start returning false from the writeCharacteristic method, indicating that the write operation could not be initiated.
> 
> To avoid this, you should wait for the onCharacteristicWrite callback with a status of BluetoothGatt.GATT_SUCCESS before issuing the next write request. This can be used as a form of flow control to ensure that you do not overwhelm the BLE stack with too many simultaneous write operations.
> 
> Please note that Write Without Response can be faster than regular writes due to its fire-and-forget nature, but it also lacks the reliability of regular writes because it does not offer any form of acknowledgment or error-checking. You should only use it when speed is more important than reliability.

chipweinberger, 2023-07-25:
Prehaps we should add a `isReadyToSendWriteWithoutResponse` stream to BluetoothCharacteristic. & On android we can use onCharacteristicWrite callback it sounds like to  listen to the stream.

chipweinberger, 2023-07-25:
Then your code would be updated to

await chr.isReadyToSendWriteWithoutResponse
await chr.write(withoutResponse)
await chr.isReadyToSendWriteWithoutResponse
await chr.write(withoutResponse)
await chr.isReadyToSendWriteWithoutResponse
await chr.write(withoutResponse)
await chr.isReadyToSendWriteWithoutResponse
await chr.write(withoutResponse)

fgendrecapacites, 2023-07-25:
Thanks a lot for your help, you surely pointed the problem here. 
I tried to use methodChannels to call isReadyToSendWriteWithoutResponse from the ios bluetooth core api without much success. 

I'll try to find a temporary workaround until there is an official fix.

Thanks again!

chipweinberger, 2023-07-26:
try 1.11.3

* dart: writeWithoutResponse should wait for completion, to prevent dropped packets
* 
https://github.com/boskokg/flutter_blue_plus/commit/a15bcf5dc3c256b1ef32ad88a2edd32e59e25fc1

chipweinberger, 2023-07-26:
let me know if it works for you^

fgendrecapacites, 2023-07-26:
It works, thank you so much !!
You'd deserve so much pints at any pub ahah thanks a lot mate

chipweinberger, 2023-07-26:
glad it worked!!

closed 2023-07-26 by chipweinberger

chipweinberger, 2023-07-26:
the latest versions have some more reliability improvements around this. (writeWithoutResponse)