chipweinberger/flutter_blue_plus #427: Write to characteristics throws exception for singed int

DmiSok, 2023-07-25:
We are using 32bit singed int as a characteristic value (little endian)

If we write to characteristic with:
`await characteristic.write([-0x0f, 0x00, 0x00, 0x00]);`

everything works fine, but as soon as we use 2 digit negative hex:
`await characteristic.write([-0x10, 0x00, 0x00, 0x00]);`

We got:
```
[ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: PlatformException(androidException, java.lang.StringIndexOutOfBoundsException: length=9; index=9, java.lang.StringIndexOutOfBoundsException: length=9; index=9 at java.lang.String.charAt(Native Method)
at com.boskokg.flutter_blue_plus.FlutterBluePlusPlugin.hexToBytes(FlutterBluePlusPlugin.java:1968)
at com.boskokg.flutter_blue_plus.FlutterBluePlusPlugin.onMethodCall(FlutterBluePlusPlugin.java:722)
at io.flutter.plugin.common.MethodChannel$IncomingMethodCallHandler.onMessage(MethodChannel.java:258)
at io.flutter.embedding.engine.dart.DartMessenger.invokeHandler(DartMessenger.java:295)
at io.flutter.embedding.engine.dart.DartMessenger.lambda$dispatchMessageToQueue$0$io-flutter-embedding-engine-dart-DartMessenger(DartMessenger.java:322)
at io.flutter.embedding.engine.dart.DartMessenger$$ExternalSyntheticLambda0.run(Unknown Source:12)
at android.os.Handler.handleCallback(Handler.java:942)
at android.os.Handler.dispatchMessage(Handler.java:99)
at android.os.Looper.loopOnce(Looper.java:201)
at android.os.Looper.loop(Looper.java:288)

#0      StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:652:7)
#1      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:310:18)
#2      BluetoothCharacteristic.write.<anonymous closure> (package:flutter_blue_plus/src/bluetooth_characteristic.dart:142:9)
#3      _Mutex.synchronized (package:flutter_blue_plus/src/utils.dart:371:7)
#4      BluetoothCharacteristic.write (package:flutter_blue_plus/src/bluetooth_characteristic.dart:118:5)
```

- Exception is reproducible on iOS and Android
- tested with `flutter_blue_plus: ^1.10.9`
- before migration `flutter_blue_plus: ^1.5.2` was not reproducible

chipweinberger, 2023-07-25:
Why would you want to write negative values?

Do you want us to cast it to a positive number for you?

```
// takes an int of any value, and return 0-255
int toUnsignedByte(int n) {
  return n & 0xFF;
}
```

Is that what you want to happen?

chipweinberger, 2023-07-25:
I've pushed 44536a87a8749888b945bf8a0b1282d430b9480a

closed 2023-07-25 by chipweinberger