chipweinberger/flutter_blue_plus #428: Platform Exception when trying to scan devices on android on 1.10.9

ryangardner, 2023-07-25:
I updated to the latest, and my app stopped being able to scan. To troubleshoot the issue, I build the example app and tried doing a scan there. 

I get the same error: 

```
D/[FBP-Android]( 2755): [FBP-Android] onMethodCall: getConnectedDevices
D/[FBP-Android]( 2755): [FBP-Android] onMethodCall: getConnectionState
D/[FBP-Android]( 2755): [FBP-Android] onMethodCall: discoverServices
E/flutter ( 2755): [ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: PlatformException(androidException, java.lang.Exception: locateGatt: BluetoothDeviceCache is null, have you connected first?, java.lang.Exception: locateGatt: BluetoothDeviceCache is null, have you connected first?
E/flutter ( 2755): 	at com.boskokg.flutter_blue_plus.FlutterBluePlusPlugin.locateGatt(FlutterBluePlusPlugin.java:1153)
E/flutter ( 2755): 	at com.boskokg.flutter_blue_plus.FlutterBluePlusPlugin.onMethodCall(FlutterBluePlusPlugin.java:594)
E/flutter ( 2755): 	at io.flutter.plugin.common.MethodChannel$IncomingMethodCallHandler.onMessage(MethodChannel.java:258)
E/flutter ( 2755): 	at io.flutter.embedding.engine.dart.DartMessenger.invokeHandler(DartMessenger.java:295)
E/flutter ( 2755): 	at io.flutter.embedding.engine.dart.DartMessenger.lambda$dispatchMessageToQueue$0$io-flutter-embedding-engine-dart-DartMessenger(DartMessenger.java:322)
E/flutter ( 2755): 	at io.flutter.embedding.engine.dart.DartMessenger$$ExternalSyntheticLambda0.run(Unknown Source:12)
E/flutter ( 2755): 	at android.os.Handler.handleCallback(Handler.java:942)
E/flutter ( 2755): 	at android.os.Handler.dispatchMessage(Handler.java:99)
E/flutter ( 2755): 	at android.os.Looper.loopOnce(Looper.java:201)
E/flutter ( 2755): 	at android.os.Looper.loop(Looper.java:288)
E/flutter ( 2755): 	at android.app.ActivityThread.main(ActivityThread.java:7918)
E/flutter ( 2755): 	at java.lang.reflect.Method.invoke(Native Method)
E/flutter ( 2755): 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:548)
E/flutter ( 2755): 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:936)
E/flutter ( 2755): , null)
E/flutter ( 2755): #0      StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:653:7)
E/flutter ( 2755): #1      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:315:18)
E/flutter ( 2755): <asynchronous suspension>
E/flutter ( 2755): #2      BluetoothDevice.discoverServices (package:flutter_blue_plus/src/bluetooth_device.dart:121:5)
E/flutter ( 2755): <asynchronous suspension>
E/flutter ( 2755): #3      BluetoothDevice.servicesList (package:flutter_blue_plus/src/bluetooth_device.dart:51:34)
E/flutter ( 2755): <asynchronous suspension>
E/flutter ( 2755): #4      _getConnectedScanResults (package:flutter_blue_plus/src/flutter_blue_plus.dart:411:35)
E/flutter ( 2755): <asynchronous suspension>
E/flutter ( 2755): #5      FlutterBluePlus.scan (package:flutter_blue_plus/src/flutter_blue_plus.dart:159:28)
E/flutter ( 2755): <asynchronous suspension>
E/flutter ( 2755): 
```

This is running the code from: 
 https://pub.dev/packages/flutter_blue_plus/example  or from the 
https://raw.githubusercontent.com/boskokg/flutter_blue_plus/master/example/lib/main.dart

It worked previously when I was on a 1.0.9 version

ryangardner, 2023-07-25:
I got it to perform one scan by turning of bluetooth and then going to the app and using the "turn on" button on the bluetooth off screen to get it to turn on... but it only worked one time that I tried it and I can't get it to do it again. It fails with the same platform exception.

chipweinberger, 2023-07-25:
this must be due to the new includeConnectedDevices feature. Thanks for reporting.

chipweinberger, 2023-07-26:
this should be fixed in 1.10.10

closed 2023-07-26 by chipweinberger

ryangardner, 2023-07-26:
Yes, upgrading to 1.10.10 fixed the issue. Thanks for the quick turnaround!