chipweinberger/flutter_blue_plus #429: Android pairing popup appears twice

mitchellilr, 2023-07-26:
Hi,

When connecting to a BLE device which requires bonding, the "Tap to pair" action notification appears twice in a row.

Thanks

chipweinberger, 2023-07-26:
Please share the logs.

mitchellilr, 2023-07-26:
```
2x D/[FBP-Android]( 7264): [FBP-Android] onMethodCall: getConnectedDevices
D/[FBP-Android]( 7264): [FBP-Android] onMethodCall: getAdapterState
D/InputConnectionAdaptor( 7264): The input method toggled cursor monitoring off
D/[FBP-Android]( 7264): [FBP-Android] onMethodCall: startScan
D/BluetoothAdapter( 7264): isLeEnabled(): ON
D/BluetoothLeScanner( 7264): onScannerRegistered() - status=0 scannerId=10 mScannerId=0
12x D/[FBP-Android]( 7264): [FBP-Android] onMethodCall: getAdapterState
D/[FBP-Android]( 7264): [FBP-Android] onMethodCall: getConnectionState
D/[FBP-Android]( 7264): [FBP-Android] onMethodCall: stopScan
D/BluetoothAdapter( 7264): isLeEnabled(): ON
D/[FBP-Android]( 7264): [FBP-Android] onMethodCall: stopScan
D/BluetoothAdapter( 7264): isLeEnabled(): ON
D/BluetoothLeScanner( 7264): could not find callback wrapper
D/[FBP-Android]( 7264): [FBP-Android] onMethodCall: connect
D/BluetoothGatt( 7264): connect() - device: 91:38:0B:A9:C5:BA, auto: false
D/BluetoothGatt( 7264): registerApp()
D/BluetoothGatt( 7264): registerApp() - UUID=4417a33d-3ab6-4d04-92d5-4245a2df0719
D/[FBP-Android]( 7264): [FBP-Android] onMethodCall: getConnectionState
D/BluetoothGatt( 7264): onClientRegistered() - status=0 clientIf=10
D/BluetoothGatt( 7264): onClientConnectionState() - status=0 clientIf=10 device=91:38:0B:A9:C5:BA
D/[FBP-Android]( 7264): [FBP-Android] onConnectionStateChange: status: 0 newState: 2
D/[FBP-Android]( 7264): [FBP-Android] onMethodCall: clearGattCache
W/.[app name]( 7264): Accessing hidden method Landroid/bluetooth/BluetoothGatt;->refresh()Z (unsupported, reflection, allowed)
D/BluetoothGatt( 7264): refresh() - device: 91:38:0B:A9:C5:BA
D/[FBP-Android]( 7264): [FBP-Android] onMethodCall: getConnectionState
D/[FBP-Android]( 7264): [FBP-Android] onMethodCall: discoverServices
D/BluetoothGatt( 7264): discoverServices() - device: 91:38:0B:A9:C5:BA
D/BluetoothGatt( 7264): onConnectionUpdated() - Device=91:38:0B:A9:C5:BA interval=6 latency=0 timeout=500 status=0
D/BluetoothGatt( 7264): onSearchComplete() = Device=91:38:0B:A9:C5:BA Status=0
D/[FBP-Android]( 7264): [FBP-Android] onServicesDiscovered: count: 3 status: 0
D/[FBP-Android]( 7264): [FBP-Android] onMethodCall: setNotification
D/BluetoothGatt( 7264): setCharacteristicNotification() - uuid: 0000abf2-0000-1000-8000-00805f9b34fb enable: true
D/[FBP-Android]( 7264): [FBP-Android] onMethodCall: getConnectionState
D/[FBP-Android]( 7264): [FBP-Android] onMethodCall: getConnectedDevices
D/[FBP-Android]( 7264): [FBP-Android] onDescriptorWrite: uuid: 00002902-0000-1000-8000-00805f9b34fb status: 0
D/BluetoothGatt( 7264): onConnectionUpdated() - Device=91:38:0B:A9:C5:BA interval=36 latency=0 timeout=500 status=0
D/[FBP-Android]( 7264): [FBP-Android] onMethodCall: requestMtu
D/BluetoothGatt( 7264): configureMTU() - device: 91:38:0B:A9:C5:BA mtu: 512
2x D/[FBP-Android]( 7264): [FBP-Android] onMethodCall: getMtu
D/[FBP-Android]( 7264): [FBP-Android] onMethodCall: requestMtu
D/BluetoothGatt( 7264): configureMTU() - device: 91:38:0B:A9:C5:BA mtu: 512
D/[FBP-Android]( 7264): [FBP-Android] onMethodCall: getMtu
D/[FBP-Android]( 7264): [FBP-Android] onMethodCall: requestMtu
D/BluetoothGatt( 7264): configureMTU() - device: 91:38:0B:A9:C5:BA mtu: 512
D/[FBP-Android]( 7264): [FBP-Android] onMethodCall: getMtu
D/[FBP-Android]( 7264): [FBP-Android] onMethodCall: requestMtu
D/BluetoothGatt( 7264): configureMTU() - device: 91:38:0B:A9:C5:BA mtu: 512
D/BluetoothGatt( 7264): onConfigureMTU() - Device=91:38:0B:A9:C5:BA mtu=512 status=0
D/[FBP-Android]( 7264): [FBP-Android] onMtuChanged: mtu: 512 status: 0
D/[FBP-Android]( 7264): [FBP-Android] onMethodCall: getMtu
```

The popups occur after this last line in the log and produce no new logs of their own.

chipweinberger, 2023-07-26:
what does the popup look like?

I don't see any calls to 'pair' or 'createBond' in your logs.

chipweinberger, 2023-07-26:
https://github.com/PhilipsHue/flutter_reactive_ble/issues/507

chipweinberger, 2023-08-01:
i'll keep this open so it does not get refiled

ericaig, 2023-08-30:
Is there a way to connect to a device without asking to pair/bond with it?

DukeJeon, 2023-11-03:
same issue in Android 13

VUNO-wonwoohan, 2023-12-11:
In my experience working with Android native code, it was helpful to call createBond() before linking method calls.
I think it can probably be solved by Android native code in the plug-in.

samnbd, 2024-01-02:
You can call createBond() yourself just after connecting and this will resolve the issue

Neuroquila-n8fall, 2024-01-10:
I double that what @samnbd suggested. Not only because "it just works" but because it will automatically circumvent any weirdness that is built into the OS flavor. At the end of the day it's about UX.