chipweinberger/flutter_blue_plus #431: androidException, java.lang.Exception: locateGatt failed.

LvJson, 2023-07-26:
flutter_blue_plus: ^1.10.10
///************************************************///

PlatformException(androidException, java.lang.Exception: locateGatt failed. have you connected first?, java.lang.Exception: locateGatt failed. have you connected first?

at com.boskokg.flutter_blue_plus.FlutterBluePlusPlugin.locateGatt(FlutterBluePlusPlugin.java:1061)
at com.boskokg.flutter_blue_plus.FlutterBluePlusPlugin.onMethodCall(FlutterBluePlusPlugin.java:506)
at io.flutter.plugin.common.MethodChannel$IncomingMethodCallHandler.onMessage(MethodChannel.java:258)
at io.flutter.embedding.engine.dart.DartMessenger.invokeHandler(DartMessenger.java:295)
at io.flutter.embedding.engine.dart.DartMessenger.lambda$dispatchMessageToQueue$0$io-flutter-embedding-engine-dart-DartMessenger(DartMessenger.java:322)
at io.flutter.embedding.engine.dart.DartMessenger$$ExternalSyntheticLambda0.run(Unknown Source:12)
at android.os.Handler.handleCallback(Handler.java:900)
at android.os.Handler.dispatchMessage(Handler.java:103)
at android.os.Looper.loop(Looper.java:219)
at android.app.ActivityThread.main(ActivityThread.java:8673)
at java.lang.reflect.Method.invoke(Native Method)
at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:513)
at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1109)
, null)

chipweinberger, 2023-07-26:
have you connected first?

chipweinberger, 2023-07-26:
try setting `connect(shouldClearGattCache: false)`

closed 2023-07-26 by LvJson

chipweinberger, 2023-07-26:
it worked?

LvJson, 2023-07-26:
yes